espressif/arduino-esp32 #2575: Crash when writing to EEPROM while using RadioHead's RH_ASK lib: Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)

simfero, 2019-03-13:
### Hardware:
Board:							Node32s
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    don't know
Upload Speed:				       921600
Computer OS:                                        Windows 10

### Description:
I'm trying to use EEPROM together with RadioHead's RH_ASK library (which is based on VirtualWire library). The program crashes with the following error when I try to store values into the EEPROM after initializing RH_ASK driver.

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400ea260: bad00bad bad00bad bad00bad

I have tried using IRAM_ATTR directive on the EEPROM function in my complete code without any luck. Attached, is the condensed code snippet the produces this crash.

### Sketch:
```cpp

#include <EEPROM.h>
#include <SPI.h>
#include <RH_ASK.h>

RH_ASK _rfDriver(2000, 5, 9, 0);

void setup() {
  EEPROM.begin(512);
  _rfDriver.init();
  EEPROM.put(0, 123456);
  EEPROM.commit();
}

void loop() {
}
```

### Debug Messages:
```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400ea260: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400ea260: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400ea260: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400ea260: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400ea260: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400ea260: bad00bad bad00bad bad00bad
```

lbernstone, 2019-03-13:
put() is not what you want there.  It is probably interpreting that as a pointer location.  If you are storing a number, use writeInt.

simfero, 2019-03-13:
Writing an integer was just an example. I am also writing structures and that's why I use put() instead of write(). I believe commit() causes the actual crash, not put().

Adrianotiger, 2019-03-19:
I have the same problem with this Core 1 panic'ed. I am debugging the entire day to find out what it can be - but still didn't found why.

**Cache disabled but cached memory region accessed**
If I understand it correctly, if you use a timer (even with IRAM_ATTR), this will disable the SPI to write into the EEPROM/Parameter/NVS/SPIFFS.

In my case, I am using a timer to write to the ADC. The problem is not the SPI itself but the interrupt. 
So, in your case, you can write EEPROM.put, but if this happens during an interrupt call, the interrupt-code will crash your esp.

I have a dacWrite (analog output on pin 25), if this interrupt occurs during a SPIFFS access (for example with the WiFiManager library), the dacWrite will crash my ESP32. The [dacWrite function from Espressif-library](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-dac.c) has the IRAM_ATTR but has no access to the registry at that point.

So, I don't know if this is just a bug or if an interrupt function can't write to a port when you use the SPI-port.

Adrianotiger, 2019-03-20:
Maybe this is something that you need to post in the other library (RH_ASK):
https://github.com/espressif/arduino-esp32/issues/2486

I can put another example (my case), probably also in this case the crash is caused by another library:

```
#include "esp32-hal-timer.h"
#include <WiFiManager.h>

hw_timer_t *timer = NULL;
WiFiManager wifiManager;

void IRAM_ATTR _onTimer() {
  dacWrite(25, random(0,255));
}

void setup() {
  Serial.begin(115200);
  
  timer = timerBegin(0, 80, true);          // 800 = 80MHz / 80 = 1000kHz (1us pulse)
  timerAttachInterrupt(timer, &_onTimer, true);
  timerAlarmWrite(timer, 10, true);       // every 1ms
  timerAlarmEnable(timer);
  Serial.println("Timer started");
  
  wifiManager.autoConnect("Test", "test1234"); // <-- Crash!
  Serial.println("WiFi started");
}

void loop() {
  delay(1000);
}

```
The library is from https://github.com/tzapu/WiFiManager
But the debug console just shows this:

```
Timer started
*WM: [1] AutoConnect 
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
...
```

So I really don't know what I can do or why it crashes...

lbernstone, 2019-03-20:
@Adrianotiger : You are trying to execute something in an interrupt that will not complete before it gets triggered again (or complete in a timely manner).  Don't interact directly with hardware in an interrupt.  Set a semaphore.  In your particular situation, you can use the Ticker library instead of a hardware timer, which executes your callback as a new task instead of an ISR:
```
#include <Ticker.h>
#include <Preferences.h>

Ticker tkDac;
Preferences preferences;

void ICACHE_RAM_ATTR _onTimer() {
  dacWrite(25, random(0,255));
}

void setup() {
  Serial.begin(115200);
  tkDac.attach_ms(1, _onTimer);
  Serial.println("Timer started");
}

void loop() {
  delay(1000);
  String output = "Simple answer, get param: ";
  preferences.begin("test", false);
  output += preferences.getString("testParam", "void");
  preferences.end();
  Serial.println(output);
}
```

Adrianotiger, 2019-03-20:
### Don't interact directly with hardware in an interrupt.
This is something I need to remember. Thanks!
Even if it will be hard to never set an output pin in an interrupt :P

Ticker library accepts only ms (I need 0.1ms in my audio-project). But in that library I found how I can use the timer. Thank you really much for your help!

atanisoft, 2019-03-20:
Using digitalWrite/digitalRead in an ISR is fine, but dacWrite/dacRead would not be as they do more than just toggle the state of the pin.

closed 2019-08-15 by stale[bot]