espressif/arduino-esp32 #2576: WiFiClient - Is there an esp32 equivelent code for esp8266's setSync

gilmaimon, 2019-03-13:
In one of my projects I make many and frequent read/writes using a WiFiClient. Because of the async (default) nature of WiFiClient the data will sometimes not be sent/receivied properly. To my understanding this is somekind of buffer overflow in the internals of WiFiClient.

The problem does not occur when I add delay between operations.

With a ESP8266 I had the same problem, but there I fixed the issue by adding `client.setSync(true)` which is not available in the ESP32 interface. 

Bottom line, I would like to know if there is a esp32 equivelent code for esp8266's `setSync` in the WiFiClient class.

Problem was first described [here](https://github.com/gilmaimon/ArduinoWebsockets/issues/2).

gilmaimon, 2019-03-26:
Bumping this issue. @me-no-dev

gilmaimon, 2019-04-25:
Still relevant

gilmaimon, 2019-08-01:
Further activity

closed 2019-10-14 by stale[bot]

krukhlis, 2020-10-17:
Guys, do you have any solution? I'm porting my code from ESP8266 to ESP32 and I'm frustrated that WiFiClient.setSync() was removed from ESP32 implementation of the WiFiClient class. Do you have any alternatives or workarounds to be used to mimic setSync() behavior under ESP32?

Miq1, 2020-10-20:
Same issue here! 👍 

> Do you have any alternatives or workarounds to be used to mimic setSync() behavior under ESP32?

Wish to have that as well.