espressif/arduino-esp32 #2577: ESP32 Serial Bluetooth loses data while receiving

JoaquinPine, 2019-03-14:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.0
IDE name:						Arduino IDE? Platform.io IDF component
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 7

### Description:
Hi. Im working in a project. I have designed an App in my phone with Processing APDE. It only sends bytes using Bluetooth. The esp32 that im using is supposed to receive this bytes but there are a lot of them that are lost. Im sending an array that contains 5220 bytes from my Phone. Using the example code of SerialBluetooth didn't work so I started to change things in order to fix this problem but i couldn't. This array contains the green color value of an image of 180 x 29 pixels.  

When I change the delay time also changes the received data. In the code I put a counter to know how many bytes is receiving de BT and then print the data received. This value changes when the delay is modified. I also tried to use timer functions in order to not block de program but didn't worked.

Anyone knows how to fix this? I have been reading lots of Issues and it seems that there was a problem with some congestion event, but it was fixed some time ago. I think that i have de lastest version of the Arduino Core for ESP32.

Thanks in advance.

The code

#include "BluetoothSerial.h" //Header File for Serial Bluetooth, will be added by default into Arduino
#include "vfs_api.h"
BluetoothSerial ESP_BT; //Object for Bluetooth

unsigned char incoming[5220];
int i = 0;
int n =0 ;
//hw_timer_t * timer = NULL;
int espera = 1;

/*void IRAM_ATTR onTimer(){
  espera=1;
}*/

void setup() {
  Serial.begin(9600); //Start Serial monitor in 9600
  ESP_BT.begin("Nuclex Holographic"); //Name of your Bluetooth Signal
  Serial.println("Bluetooth Device is Ready to Pair");
  /*timer = timerBegin(0, 80, true);
  timerAttachInterrupt(timer, &onTimer, true); // edge (not level) triggered 
  timerAlarmWrite(timer, 5, true);
  timerAlarmEnable(timer); // enable*/
  
}

void loop() {
  
  if (ESP_BT.available()) //Check if we receive anything from Bluetooth
  { 

    incoming[i] = ESP_BT.read();

   i++; //counter bytes
    delayMicroseconds(5);}
    /*if(espera==1) espera=0;
    else return;*/

  }else  if ( i != 0){
  
   Serial.print("bytes:"); 
   Serial.print(i); 
   Serial.print("\n"); 
   n=0;
  for(n=0;n<i;n++){
   Serial.println(incoming[n]);

   incoming[n]=0;
   }
   //Serial.println(incoming[5219]);
   i=0;
  }
}

Serial Monitor Arduino with 5us of delay
bytes:2498
4
4
4
4
4
...
Serial Monitor Arduino with 3us of delay
bytes:1895
4
4
4
4
4
...
Serial Monitor Arduino with 6us of delay
bytes:2141
4
4
4
4
4
...

copercini, 2019-03-23:
Try update your core version, the flow control was implemented here: https://github.com/espressif/arduino-esp32/commit/96822d783f3ab6a56a69b227ba4d1a1a36c66268#diff-59df1af5e9b4ee1961cbd5beadc4b84d and as far I remember it was after v1.0.0

JoaquinPine, 2019-03-28:
Hi mate! Thank you for answering. I think i have the lastest version of the core. To check this i only have to see in the Boards Manager from Tools in the Arduino IDE and see if there is any update, or i have to update the core in a different way? If the first option is ok, Arduino is not offering me any update.

Thanks.

closed 2019-08-15 by stale[bot]

TiSiAi, 2019-09-13:
what max buffer size can we use ?
I must receive 4ko data and I lose data with default buffer size
I try a bigger buffer, it's ok but is it possible to use 4000 octets buffer ?

replace
#define RX_QUEUE_SIZE 512
with
#define RX_QUEUE_SIZE 4000

thank

copercini, 2019-09-13:
@TiSiAi you can use any size, the only limit is the ESP memory, using 4k here you can't use this memory for other applications

TiSiAi, 2019-09-13:
> @TiSiAi you can use any size, the only limit is the ESP memory, using 4k here you can't use this memory for other applications

Thank, esp32 has enough memory, so I can use.
possible to add the size in another begin constructor  or to set the size with a function before call begin function ?
If we change this value in the original file, it will lose on the next update esp32 ...

bigLucas, 2020-04-11:
Hi guys, I'm working with the same issue, but in my case i decided to change the app code, I'm developing for android. In send_data_activity I put a delay and divided my data into two arrays holding 225 bytes in each (about).

**I don't know if this is the best way to do this, but it worked for me.**

[image]