espressif/arduino-esp32 #2578: ESP32 reboots when connecting to SoftAP

mattncsu, 2019-03-14:
### Hardware:
Board:							ESP32-WROOM module on custom pcb
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I am trying to build a AP fallback into the start wifi function, however when I connect to the AP from my laptop, the ESP reboots.  When I was testing in AP+STA mode while connected to a station, I could connect to the softAP just fine.  I'm trying to test with an invalid SSID right now.

### Sketch: 
```cpp

void WIFIstart() {       // Connect to Wi-Fi
  if (Vbat > BATT_THRESHOLD) {
    Serial.print("Stored SSID: ");  Serial.println(wifiSSID);
    WiFi.mode(WIFI_AP_STA); 
    char APname[7]="ESPTap";
    IPAddress Ip(192, 168, 1, 1);
    IPAddress NMask(255, 255, 255, 0);
    WiFi.softAPConfig(Ip, Ip, NMask);
    WiFi.begin(wifiSSID.c_str(), wifiPassword.c_str());//Try connecting to 
    uint32_t timer=millis();
    while (WiFi.status() != WL_CONNECTED) {
      delay(1000);
      Serial.println("Connecting to WiFi..");
      if(millis()-timer>8000){
        Serial.println("Giving up on " + wifiSSID + " after 8 seconds of trying.");
        WiFi.mode(WIFI_OFF);
        delay(100);
        WiFi.mode(WIFI_AP);
        int a=WiFi.softAP(APname,"1234567890");
        Serial.print("softap returned: ");Serial.println(a);
        break;
      }
    }
        Serial.println("AP Started");
        Serial.print("MAC Address: ");
        Serial.println(WiFi.macAddress());
        Serial.print("AP SSID:");Serial.println(APname);
        Serial.print("AP IPv4: ");
        Serial.println(WiFi.softAPIP());
    if (WiFi.status()){
      WIFIon=true;
      Serial.println(WiFi.localIP());
    }  
    server.begin();
  }
}
```

### Debug Messages:
Serial (verbose):
```
00:40:23.563 -> Stored SSID: none
00:40:23.665 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
00:40:23.665 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
00:40:23.665 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
00:40:24.721 -> Connecting to WiFi..
00:40:25.710 -> Connecting to WiFi..
00:40:26.590 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
00:40:26.590 -> [W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
00:40:26.590 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
00:40:26.590 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
00:40:26.724 -> Connecting to WiFi..
00:40:27.707 -> Connecting to WiFi..
00:40:28.727 -> Connecting to WiFi..
00:40:29.405 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
00:40:29.405 -> [W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
00:40:29.405 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
00:40:29.405 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
00:40:29.714 -> Connecting to WiFi..
00:40:30.735 -> Connecting to WiFi..
00:40:31.714 -> Connecting to WiFi..
00:40:32.257 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
00:40:32.257 -> [W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
00:40:32.257 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
00:40:32.257 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
00:40:32.738 -> Connecting to WiFi..
00:40:32.738 -> Giving up on none after 8 seconds of trying.
00:40:32.738 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 14 - AP_STOP
00:40:32.738 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 14 - AP_STOP
00:40:32.873 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
00:40:32.873 -> softap returned: [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
00:40:32.873 -> 80:7D:3A:CA:A4:14[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 3 - STA_STOP
00:40:32.873 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 3 - STA_STOP
00:40:32.873 -> .168.1.1
00:40:32.873 -> 0.0.0.0

00:40:42.708 -> E (28979) event: mismatch or invalid event, id=63
00:40:42.708 -> E (28979) event: default event handler failed!
00:40:42.708 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
00:40:42.708 -> Core 1 register dump:
00:40:42.708 -> PC      : 0x400014dc  PS      : 0x00060730  A0      : 0x800f90ec  A1      : 0x3ffcc470  
00:40:42.708 -> A2      : 0xffff78ff  A3      : 0xffff78fb  A4      : 0x000000ff  A5      : 0x0000ff00  
00:40:42.708 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
00:40:42.708 -> A10     : 0x00000000  A11     : 0x3ffcc672  A12     : 0x00000000  A13     : 0x00000000  
00:40:42.791 -> A14     : 0x60000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
00:40:42.791 -> EXCVADDR: 0xffff78ff  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  
00:40:42.791 -> 
00:40:42.791 -> Backtrace: 0x400014dc:0x3ffcc470 0x400f90e9:0x3ffcc480 0x40100642:0x3ffcc790 0x4010067e:0x3ffcc820 0x400e11fe:0x3ffcc860 0x400d6692:0x3ffcc8c0 0x400d69ed:0x3ffcc9e0 0x40094fd5:0x3ffcca10
00:40:42.791 -> 
00:40:42.791 -> Rebooting...

```
Decoded:
```
PC: 0x400014dc
EXCVADDR: 0xffff78ff

Decoding stack results
0x400f90e9: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1529
0x40100642: _vsnprintf_r at ../../../.././newlib/libc/stdio/vsnprintf.c line 72
0x4010067e: vsnprintf at ../../../.././newlib/libc/stdio/vsnprintf.c line 41
0x400e11fe: log_printf at C:\Users\Matt\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\cores\esp32\esp32-hal-uart.c line 473
0x400d6692: WiFiGenericClass::_eventCallback(void*, system_event_t*) at C:\Users\Matt\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\WiFi\src\WiFiGeneric.cpp line 342
0x400d69ed: _network_event_task(void*) at C:\Users\Matt\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\WiFi\src\WiFiGeneric.cpp line 67
0x40094fd5: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```

mattncsu, 2019-03-19:
Any clues where to start? Looking at the decoded stack results, the references seem to point to a problem with printing to the serial port.  If I set debug output to None, then it doesn't reboot but I still get these messages over serial when connecting to the AP:
```
22:40:14.264 -> E (24393) event: mismatch or invalid event, id=63
22:40:14.264 -> E (24393) event: default event handler failed!
```
Despite the messages, i can still connect to the AP and the ESP serves up a webpage using the async server just fine.  I'm going to _assume_ that means the problem is with the core and not my code but if someone has some suggestions, let me know.

atanisoft, 2019-03-19:
> Any clues where to start? Looking at the decoded stack results, the references seem to point to a problem with printing to the serial port. If I set debug output to None, then it doesn't reboot but I still get these messages over serial when connecting to the AP:

Give v1.0.2-rc1 a try, the error you are hitting may have been fixed in that version.

closed 2019-08-15 by stale[bot]