espressif/arduino-esp32 #2584: ESP32 crash on pthread_cond_wait()

zoliweiss, 2019-03-14:
### Hardware:
Board:							ESP32 DoIt module - Chip is ESP32D0WDQ6 (revision 1)
Core Installation version:			1.0.2-rc1
IDE name:						Arduino IDE 1.8.8 - tried 1.8.5
Flash Frequency:					80Mhz
PSRAM enabled:                                    IDK
Upload Speed:					921600
Computer OS:                                        Win 7

### Description:
I tried to implement a simple network packet consumer-producer system for my server but I get core panic when the main thread is waiting for the condition variable to be set. The code is tested on linux and works OK.
```
***ERROR*** A stack overflow in task pthread has been detected.
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40085adf  PS      : 0x00060a30  A0      : 0x80085c4c  A1      : 0x3ffb0e90  
A2      : 0xffffffff  A3      : 0xffffffff  A4      : 0x00000004  A5      : 0x3ffc989c  
A6      : 0x00060820  A7      : 0x00000000  A8      : 0x8008f675  A9      : 0x3ffb0e80  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffc98ec  A13     : 0x00000002  
A14     : 0x3ffc989c  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffffffff  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40085adf:0x3ffb0e90 0x40085c49:0x3ffb0ec0 0x4012448e:0x3ffb0ee0 0x4012452d:0x3ffb0f20 0x400d2936:0x3ffb0f40 0x400d4a43:0x3ffb0f60 0x400d68c9:0x3ffb1fb0 0x4008eeb1:0x3ffb1fd0
```
DECODED:
```
PC: 0x40085adf: lock_acquire_generic at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/locks.c line 127
EXCVADDR: 0xffffffff

Decoding stack results
0x40085adf: lock_acquire_generic at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/locks.c line 127
0x40085c49: _lock_acquire_recursive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/locks.c line 171
0x4012448e: pthread_cond_timedwait at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/pthread/pthread_cond_var.c line 131
0x4012452d: pthread_cond_wait at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/pthread/pthread_cond_var.c line 86
0x400d2936: get_packet_from_queue(thread_data_t*, packet_t*) at D:\git\super-simple-sip-server\ssss/ssss.ino line 231
0x400d4a43: loop() at D:\git\super-simple-sip-server\ssss/ssss.ino line 977
0x400d68c9: loopTask(void*) at D:\Progik\arduino\arduino-1.8.8\hardware\espressif\esp32\cores\esp32\main.cpp line 25
0x4008eeb1: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
### Sketch:  
https://gitlab.com/zoliweiss/super-simple-sip-server/commit/7d3916e0c14e4a12a64e215ffd72a04cdf95e99c

### Debug Messages:
-

stickbreaker, 2019-03-14:
>***ERROR*** A stack overflow in task pthread has been detected.

by default the `loop()` task in Arduino only has a 8,192 byte STACK depth.
Look at your local variables in the call sequence.
Chuck.

zoliweiss, 2019-03-14:
I tried to increase the stack size in main.cpp to 81920, the stack overflow error disappeared, but the rest is the same.

stickbreaker, 2019-03-14:
@zoliweiss in line 955 of your source you have:
>pthread_create(&g_thread_data.thread, NULL, &receive_packet, &g_thread_data);

Which I assume creates "pthread" what size of stack is used during the creation?

Chuck.

zoliweiss, 2019-03-14:
If I interpret correctly https://github.com/espressif/esp-idf/blob/master/components/pthread/pthread.c then the default is 2 kB.

stickbreaker, 2019-03-14:
I don't see where you ever init the thread control record? I would expect you to use one of the default config functions like:
[esp_pthread_cfg_t esp_pthread_get_default_config()](https://github.com/espressif/esp-idf/blob/ce6b73ca5f52ceb49fc4b7a76d6b0e343418caac/components/pthread/pthread.c#L175), then override the `.stacksize` field?

zoliweiss, 2019-03-15:
I wasn't aware of that function. Now I tried it, but I get still the same. I don't really understand why the panic happens when we wait for the condition to happen.

zoliweiss, 2019-04-11:
So, after a lot of testing it looks like you need to use `pthread_mutex_init()` and `pthread_cond_init()` functions instead of `PTHREAD_MUTEX_INITIALIZER` and `PTHREAD_COND_INITIALIZER`.

closed 2019-05-26 by zoliweiss

kanakamedalasumanth, 2021-02-19:
> So, after a lot of testing it looks like you need to use `pthread_mutex_init()` and `pthread_cond_init()` functions instead of `PTHREAD_MUTEX_INITIALIZER` and `PTHREAD_COND_INITIALIZER`.

Thank you. We can initialize like this 
pthread_cond_t wait;
pthread_cond_init(&wait , NULL);