espressif/arduino-esp32 #2585: rmtDeinit missing?

yanzixiang, 2019-03-15:
rmtDeinit is defined in esp32-hal-rmt.c,
but it is not defined in esp32-hal-rmt.h,
can someone explain the reason ?

```c
bool rmtDeinit(rmt_obj_t *rmt)
{
    if (!rmt) {
        return false;
    }

    // sanity check
    if (rmt != &(g_rmt_objects[rmt->channel])) {
        return false;
    }

    size_t from = rmt->channel;
    size_t to = rmt->buffers + rmt->channel;
    size_t i;

#if !CONFIG_DISABLE_HAL_LOCKS
    if(g_rmt_objlocks[from] != NULL) {
        vSemaphoreDelete(g_rmt_objlocks[from]);
    }
#endif

    if (g_rmt_objects[from].data_alloc) {
        free(g_rmt_objects[from].data_ptr);
    }
    
    for (i = from; i < to; i++) {
        g_rmt_objects[i].allocated = false;
    }

    g_rmt_objects[from].channel = 0;
    g_rmt_objects[from].buffers = 0;

    return true;
}
```

lbernstone, 2019-03-15:
The arduino and esp-idf RMT drivers should probably not be used interchangeably, as there may be some state information that will not cross over.  Pick one, and stick with it.  The arduino version is recent and may still have some issues (eg, a missing rmtDeinit).  The esp-idf version has examples linked in the docs location you posted.

atanisoft, 2019-03-15:
also note that the arduino driver has a few debug statements in it that will alter the state of a few pins at runtime!!!

PR #2587 (2019-04-09): Added rmtDeinit to header. Fixes #2585
Added rmtDeinit to header. Fixes #2585
closed 2019-03-31 by yanzixiang