espressif/arduino-esp32 #2586: Flash size problem?

SpectorHacked, 2019-03-15:
Hello everyone,
I am new to arduino programing and I want to build DIY IRobot to my room with wifi built it,
I am trying to use esp32s node-mcu and I am just starting with the coding and stuff, so I am trying to upload esp32 code to the board but I get this error:

```
 esptool.py v2.3.1
Connecting........_
Chip is ESP32D0WDQ6 (revision (unknown 0xa))
Features: WiFi, BT, Dual Core, VRef calibration in efuse
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content

```

Any idea to why its happening? Thanks for helping me!

zoliweiss, 2019-03-15:
Here someone has the same problem: https://github.com/espressif/arduino-esp32/issues/1097

lbernstone, 2019-03-15:
What version of arduino-esp32 are you running?  It looks like your version of esptool is old and may not contain info for the chip you have.

SpectorHacked, 2019-03-15:
So I need to upgrade the board? or the software I am using?

lbernstone, 2019-03-15:
You need to upgrade the board in arduino IDE.  https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

SpectorHacked, 2019-03-16:
Problem isnt solved after upgrade the board in arduino IDE still 
"A fatal error occurred: Time out waiting for packet content"
FYI:
The board it self is getting very very hot after trying to upload the code

lbernstone, 2019-03-16:
ESP32 modules do run hot, but they shouldn't feel like a heating element.  You might have a short somewhere on your board.

closed 2019-08-15 by stale[bot]