espressif/arduino-esp32 #2589: Why doesn't esp32-hal-time.c compile against esp-idf v3.1.2 or v3.1.3

william-ferguson-au, 2019-03-17:
Describe what is failing.
'make all' gives:

```
/home/William/esp/wave/components/arduino-esp32/component.mk:9: warning: undefined variable 'CONFIG_ARDUINO_SELECTIVE_FFat'
/home/William/esp/wave/components/arduino-esp32/component.mk:9: warning: undefined variable 'CONFIG_ARDUINO_SELECTIVE_HTTPUpdate'
Arduino libraries in build:    BLE    ESP32          SPI      WiFi WiFiClientSecure Wire
CC build/arduino-esp32/cores/esp32/libb64/cencode.o
CC build/arduino-esp32/cores/esp32/libb64/cdecode.o
CC build/arduino-esp32/cores/esp32/esp32-hal-sigmadelta.o
CC build/arduino-esp32/cores/esp32/esp32-hal-misc.o
CC build/arduino-esp32/cores/esp32/esp32-hal-rmt.o
CC build/arduino-esp32/cores/esp32/esp32-hal-adc.o
CC build/arduino-esp32/cores/esp32/wiring_pulse.o
CC build/arduino-esp32/cores/esp32/esp32-hal-timer.o
CC build/arduino-esp32/cores/esp32/esp32-hal-matrix.o
CC build/arduino-esp32/cores/esp32/esp32-hal-dac.o
CC build/arduino-esp32/cores/esp32/stdlib_noniso.o
CC build/arduino-esp32/cores/esp32/esp32-hal-uart.o
CC build/arduino-esp32/cores/esp32/esp32-hal-time.o
E:/Dev/msys32/home/William/esp/wave/components/arduino-esp32/cores/esp32/esp32-hal-time.c:16:28: fatal error: lwip/apps/sntp.h: No such file or directory
compilation terminated.
make[1]: *** [/home/William/esp/esp-idf/make/component_wrapper.mk:286: cores/esp32/esp32-hal-time.o] Error 1
make: *** [E:/Dev/msys32/home/William/esp/esp-idf/make/project.mk:468: component-arduino-esp32-build] Error 2
```

Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							Custom ESP32 board
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						NA
Flash Frequency:					NA
PSRAM enabled:                                    NA
Upload Speed:					NA
Computer OS:                                        Windows 10

### Description:
I am trying compile a project that contains arduino-esp32 as an esp-if module.
Building against either esp-idf v3.1.2 or v3.1.3 (latest release) gives:

'make all'

```
/home/William/esp/wave/components/arduino-esp32/component.mk:9: warning: undefined variable 'CONFIG_ARDUINO_SELECTIVE_FFat'
/home/William/esp/wave/components/arduino-esp32/component.mk:9: warning: undefined variable 'CONFIG_ARDUINO_SELECTIVE_HTTPUpdate'
Arduino libraries in build:    BLE    ESP32          SPI      WiFi WiFiClientSecure Wire
CC build/arduino-esp32/cores/esp32/libb64/cencode.o
CC build/arduino-esp32/cores/esp32/libb64/cdecode.o
CC build/arduino-esp32/cores/esp32/esp32-hal-sigmadelta.o
CC build/arduino-esp32/cores/esp32/esp32-hal-misc.o
CC build/arduino-esp32/cores/esp32/esp32-hal-rmt.o
CC build/arduino-esp32/cores/esp32/esp32-hal-adc.o
CC build/arduino-esp32/cores/esp32/wiring_pulse.o
CC build/arduino-esp32/cores/esp32/esp32-hal-timer.o
CC build/arduino-esp32/cores/esp32/esp32-hal-matrix.o
CC build/arduino-esp32/cores/esp32/esp32-hal-dac.o
CC build/arduino-esp32/cores/esp32/stdlib_noniso.o
CC build/arduino-esp32/cores/esp32/esp32-hal-uart.o
CC build/arduino-esp32/cores/esp32/esp32-hal-time.o
E:/Dev/msys32/home/William/esp/wave/components/arduino-esp32/cores/esp32/esp32-hal-time.c:16:28: fatal error: lwip/apps/sntp.h: No such file or directory
compilation terminated.
make[1]: *** [/home/William/esp/esp-idf/make/component_wrapper.mk:286: cores/esp32/esp32-hal-time.o] Error 1
make: *** [E:/Dev/msys32/home/William/esp/esp-idf/make/project.mk:468: component-arduino-esp32-build] Error 2
```

This exact issue was referenced here: https://github.com/espressif/arduino-esp32/issues/1853
And was supposed to have been fixed here https://github.com/espressif/arduino-esp32/blob/96822d783f3ab6a56a69b227ba4d1a1a36c66268/cores/esp32/esp32-hal-time.c

But esp32-hal-time.c was changed to  
```
#include "lwip/apps/sntp.h"
```

when it needs to have been
```
#include "lwip/apps/sntp/sntp.h"
```

Am I the only one seeing this?

william-ferguson-au, 2019-03-17:
NB that change only works against the yet unreleased esp-idf v3.2.
What version of esp-idf is arduino-esp32 meant to be built against?

lbernstone, 2019-03-17:
`
larryb@larryb-F5:~/esp32/arduino-esp32$ git log | grep -i idf | head -1
    Update IDF to ebdcbe8c6 (#2539)
larryb@larryb-F5:~/esp32/arduino-esp32$ cd ../esp-idf/
larryb@larryb-F5:~/esp32/esp-idf$ git log | grep ebdcbe8c6
Merge: ebdcbe8c6 27e00cf7a
commit ebdcbe8c60a3b5a228d9e80335d869cc01c527ab
larryb@larryb-F5:~/esp32/esp-idf$ git checkout ebdcbe8c60a3b5a228d9e80335d869cc01c527ab
`

william-ferguson-au, 2019-03-18:
Sorry @lbernstone I don't understand.

1. Why would I be changing my esp-idf version? I would expect to be changing  my arduino-esp32 version to match a released esp-idf.

2. Why would this espi-idf change (on master) that just updates some doco for crc apis have any impact on being able to find sntp.h? https://github.com/espressif/esp-idf/commit/ebdcbe8c60a3b5a228d9e80335d869cc01c527ab

william-ferguson-au, 2019-03-18:
OK so 

- arduino-esp32 1.0.1 refers to lwip/apps/sntp.h
- arduino-esp32 1.0.0 refers to apps/sntp/sntp.h

So 1.0.0 is compatible with esp-idf v3.1
And 1.0.1 is compatible with the yet unreleased esp-idf v3.2+

Really didn't expect to pick up a released version of library that has dependencies on an unreleased one. 

And since arduino-esp32 is pretty heavily coupled to esp-idf it would seem to make sense that the version numbering of arduino-esp-32 reflect the version of esp-idf it depends upon.

kapollo, 2019-03-18:
@william-ferguson-au so have you in the end found what is the most recent and fairly stable pair, e.g. arduino-esp 1.0.1 rel commit and some commit for esp-idf, I tried several and still some problems either, path too long, or like this one here, I got the heads or master branches (don't remember right now) compiling, at some point (several days ago) but this is not really what I'm after ...

william-ferguson-au, 2019-03-19:
If you want a release version of esp-idf then you'll have to go with eps-idf-v3.1.1 and arduino-esp32-1.0.0

On the assumption that the tags aren't slipping we have decided to go with
esp-idf-v3.3-beta2 and arduino-esp32-1.0.2-rc1

closed 2019-03-19 by william-ferguson-au