espressif/arduino-esp32 #2591: Arduino board support/definition for TTGO T8 (v1.7+)

knopserl, 2019-03-17:
would it be possible to support thes nice ESP32 board with SD card reader and PSRAM
https://github.com/LilyGO/TTGO-T8-ESP32#t8-v17

I have seen a similar post asking for missing TTGO boards but asling for other boards than this here.

rorosaurus, 2019-06-23:
FYI in this forum post someone appears to have been able to program it with Arduino using `Board: "ESP32 Dev Module, Enabled, Default, QIO, 80MHz, 4MB (32Mb), 921600, None"`.

knopserl, 2019-06-29:
it can be flashed with any ESP32 board, but the extra board features ROM size, PSRAM, SD-card, more flash etc. and the pinout is not good supported.

atanisoft, 2019-06-29:
@knopserl the pinout is fully support. Don't rely on the aliases as they are inconsistent from one board to the next regardless of the board option picked. PSRAM will automatically be detected and used if present on the board, regardless of board selection

atanisoft, 2019-07-28:
@knopserl in Arduino IDE use the esp32 dev board and enable the right options you need. Everything is exposed already for you!

In platformio you can use a WROVER board type to enable those same features.

SD support is not board specific, you can define the pins as needed when initializing the SD/SPI bus.

knopserl, 2019-10-02:
ok, thanks, I'll close this and open anotherr issue if I face ony problems.

closed 2019-10-02 by knopserl

dizcza, 2021-09-03:
@atanisoft 
> in Arduino IDE use the esp32 dev board and enable the right options you need. Everything is exposed already for you!

Could you clarify this? In particular, what do you mean by "enable the right options you need"? When I select "ESP Dev Module" (there is no "dev board"), I expected to see some popup menu with "enabling the right options" for me. Well, of course, it didn't happen (and should not happen - the users could easily get lost in setting up a new board; that's why there exists a wonderful folder called variants for us, users).

I found that TTGO Lora-OLED v2.x is somewhat similar to TTGO T8 although not in the pinout. I'm currently using it unless someone makes an appropriate TTGO T8 support. Why is it still needed?

> SD support is not board specific, you can define the pins as needed when initializing the SD/SPI bus.

Unfortunately, it's not true. SD support *is* board specific and I spent half a day learning this lesson. In particular, due to the `BOARD_HAS_1BIT_SDMMC` define, which is different from board to board. I was unable to find comprehensive documentation about the particularities of the TTGO T8 board (amazingly, for the same reason: everyone thinks that these boards are alike, and therefore why would one needs board-specific documentation?). Only later on I found out that the SDCard is wired as an SPI device in TTGO T8, so it is a 1-bit mode only. If there was a variant for TTGO T8 with  `BOARD_HAS_1BIT_SDMMC` set to `1`, my day would be perfect. Instead, it was only half perfect.

And, I mean, I tried only SD & WiFi features and already got board-specific issues. BTW, I'm still unable to run the default example of mounting an SD card through SPI in TTGO T8... but I ain't complaining any further. You guys do a decent job by supporting a zoo of ESP32 variants.

atanisoft, 2021-09-03:
> Could you clarify this? In particular, what do you mean by "enable the right options you need"?

Use the "ESP32 Dev Module" as the board, **DON'T** assume default pin assignments for peripherals (SD, OLED, etc). One problem with adding new board variants is maintenance of them (there are 117 boards defined in 2.0.0 today, of which many only differ in display name).

> Unfortunately, it's not true. SD support _is_ board specific and I spent half a day learning this lesson.

Defining `BOARD_HAS_1BIT_SDMMC` is not necessary, it only forces the usage of 1-bit mode. You can accomplish the same via:
```
SD_MMC.begin("/sdcard", true);
```

Almost all TTGO boards are configured only for SD-SPI mode and the full SD_MMC mode would require slight modifications to the board (SD slot has the pads for the extra data lines but they aren't connected).

> BTW, I'm still unable to run the default example of mounting an SD card through SPI in TTGO T8.

Can you share the logs showing any failures?

BT3D, 2022-03-27:
Anyone can share board settings and perhaps simple code for TTGO T8 that tests SD card, which works for him?
I tried multiple examples from IDE and internet but still "mounting card failed...".'
thanks

badre2911, 2022-05-03:
on TTGO T8 thi code work with SD_MMC for me
```
pinMode(2, INPUT_PULLUP);
    if(!SD_MMC.begin("/sdcard",true)){
        Serial.println("Card Mount Failed");
        return;
    }
```