espressif/arduino-esp32 #2593: Is there any GUI to show the RMT data ?

yanzixiang, 2019-03-18:
I config the RMT like this,
I think 
  when RMT_CLK_DIV = 1, a pluse = 12.5ns
  when RMT_CLK_DIV = 2, a pluse = 25.0ns
  when RMT_CLK_DIV = 4, a pluse = 50.0ns
  when RMT_CLK_DIV = 8, a pluse = 100.0ns
Am I right ?

```C
#define RMT_RX_CHANNEL    0    /*!< RMT channel for receiver */
#define RMT_RX_GPIO_NUM  15     /*!< GPIO number for receiver */
//#define RMT_RX_GPIO_NUM  8     /*!< GPIO number for receiver */
#define RMT_CLK_DIV      1  /*!< RMT counter clock divider */
//1 12.5ns
//2 25.0ns
//4 50.0ns
//8 100.0ns
/*
 * @brief RMT receiver initialization
 */
static void nec_rx_init()
{
  periph_module_enable(PERIPH_RMT_MODULE);
  
  rmt_config_t rmt_rx;
  rmt_rx.channel = (rmt_channel_t)RMT_RX_CHANNEL;
  rmt_rx.gpio_num = (gpio_num_t)RMT_RX_GPIO_NUM;
  rmt_rx.clk_div = RMT_CLK_DIV;
  rmt_rx.mem_block_num = 1;
  rmt_rx.rmt_mode = RMT_MODE_RX;
  rmt_rx.rx_config.filter_en = true;
  rmt_rx.rx_config.filter_ticks_thresh = 1;
  rmt_rx.rx_config.idle_threshold = 10000;
    
  esp_err_t rmt_config_result = rmt_config(&rmt_rx);
  log_e("rmt_config_result: %s", esp_err_to_name(rmt_config_result));

  uint8_t real_div_cnt = 0;
  esp_err_t rmt_get_clk_div_result =  rmt_get_clk_div(rmt_rx.channel, &real_div_cnt);
  log_e("rmt_get_clk_div_result: %s,real_div_cnt : %d ", esp_err_to_name(rmt_get_clk_div_result),real_div_cnt);

  //esp_err_t rmt_driver_install(rmt_channel_t channel, size_t rx_buf_size, int intr_alloc_flags);
  esp_err_t rmt_driver_install_result = rmt_driver_install(rmt_rx.channel, 1024 * 100, 0);
  log_e("rmt_driver_install_result: %s", esp_err_to_name(rmt_driver_install_result));

  esp_err_t rmt_set_err_intr_en_result =  rmt_set_err_intr_en((rmt_channel_t)RMT_RX_CHANNEL,true);
  log_e("rmt_set_err_intr_en_result: %s", esp_err_to_name(rmt_set_err_intr_en_result));
  esp_err_t rmt_set_rx_intr_en_result =  rmt_set_rx_intr_en((rmt_channel_t)RMT_RX_CHANNEL,true);
  log_e("rmt_set_rx_intr_en_result: %s", esp_err_to_name(rmt_set_rx_intr_en_result));   
}
```

I Still struggle to decode SENT using ESP32,
when i plug in to sensor,
the console shows this
```
13:39:43.355 -> [E][MasterESP32LAN8720WithArduino.ino:528] rmt_example_nec_rx_task(): xRingbufferReceive: 40
13:39:43.432 -> {
13:39:43.432 ->   "data": [
13:39:43.432 ->     "Index : 0 level0: 0 duration0: 1868 level1: 1 duration1: 996",
13:39:43.432 ->     "Index : 1 level0: 0 duration0: 1868 level1: 1 duration1: 2897",
13:39:43.432 ->     "Index : 2 level0: 0 duration0: 1868 level1: 1 duration1: 4332",
13:39:43.432 ->     "Index : 3 level0: 0 duration0: 1868 level1: 1 duration1: 3132",
13:39:43.432 ->     "Index : 4 level0: 0 duration0: 1868 level1: 1 duration1: 1948",
13:39:43.432 ->     "Index : 5 level0: 0 duration0: 1868 level1: 1 duration1: 4329",
13:39:43.432 ->     "Index : 6 level0: 0 duration0: 1868 level1: 1 duration1: 2660",
13:39:43.432 ->     "Index : 7 level0: 0 duration0: 1868 level1: 1 duration1: 3612",
13:39:43.465 ->     "Index : 8 level0: 0 duration0: 1868 level1: 1 duration1: 4",
13:39:43.465 ->     "Index : 9 level0: 0 duration0: 4 level1: 1 duration1: 0",
13:39:43.465 ->     "Index : 10 level0: 0 duration0: 40 level1: 0 duration1: 0",
13:39:43.465 ->     "Index : 11 level0: 0 duration0: 0 level1: 0 duration1: 0",
13:39:43.465 ->     "Index : 12 level0: 0 duration0: 1868 level1: 1 duration1: 993",
13:39:43.465 ->     "Index : 13 level0: 0 duration0: 1868 level1: 1 duration1: 2900",
13:39:43.465 ->     "Index : 14 level0: 0 duration0: 1868 level1: 1 duration1: 4332",
13:39:43.499 ->     "Index : 15 level0: 0 duration0: 1868 level1: 1 duration1: 3140",
13:39:43.499 ->     "Index : 16 level0: 0 duration0: 1868 level1: 1 duration1: 1945",
13:39:43.499 ->     "Index : 17 level0: 0 duration0: 1868 level1: 1 duration1: 4564",
13:39:43.499 ->     "Index : 18 level0: 0 duration0: 1876 level1: 1 duration1: 2652",
13:39:43.499 ->     "Index : 19 level0: 0 duration0: 1868 level1: 1 duration1: 4",
13:39:43.499 ->     "Index : 20 level0: 0 duration0: 4 level1: 1 duration1: 3132",
13:39:43.499 ->     "Index : 21 level0: 0 duration0: 1868 level1: 1 duration1: 0",
13:39:43.567 ->     "Index : 22 level0: 0 duration0: 44 level1: 0 duration1: 0",
13:39:43.567 ->     "Index : 23 level0: 0 duration0: 0 level1: 0 duration1: 0",
13:39:43.567 ->     "Index : 24 level0: 0 duration0: 1868 level1: 1 duration1: 4",
13:39:43.567 ->     "Index : 25 level0: 0 duration0: 4 level1: 1 duration1: 2892",
13:39:43.567 ->     "Index : 26 level0: 0 duration0: 1868 level1: 1 duration1: 4",
13:39:43.567 ->     "Index : 27 level0: 0 duration0: 4 level1: 1 duration1: 2892",
13:39:43.567 ->     "Index : 28 level0: 0 duration0: 1868 level1: 1 duration1: 4332",
13:39:43.567 ->     "Index : 29 level0: 0 duration0: 1868 level1: 1 duration1: 3140",
13:39:43.600 ->     "Index : 30 level0: 0 duration0: 1868 level1: 1 duration1: 2185",
13:39:43.600 ->     "Index : 31 level0: 0 duration0: 1868 level1: 1 duration1: 996",
13:39:43.600 ->     "Index : 32 level0: 0 duration0: 1868 level1: 1 duration1: 2660",
13:39:43.600 ->     "Index : 33 level0: 0 duration0: 1868 level1: 1 duration1: 4092",
13:39:43.600 ->     "Index : 34 level0: 0 duration0: 1868 level1: 1 duration1: 0",
13:39:43.600 ->     "Index : 35 level0: 0 duration0: 52 level1: 0 duration1: 0",
13:39:43.600 ->     "Index : 36 level0: 0 duration0: 0 level1: 0 duration1: 0",
13:39:43.634 ->     "Index : 37 level0: 0 duration0: 1868 level1: 1 duration1: 4",
13:39:43.634 ->     "Index : 38 level0: 0 duration0: 4 level1: 1 duration1: 3844",
13:39:43.634 ->     "Index : 39 level0: 0 duration0: 4 level1: 1 duration1: 2"
13:39:43.634 ->   ]
13:39:43.634 -> }
```

It's like many times repeating down this,
Is it a real sent frame ?
If it is ,
What value is it ?
```
13:39:43.465 ->     "Index : 12 level0: 0 duration0: 1868 level1: 1 duration1: 993",
13:39:43.465 ->     "Index : 13 level0: 0 duration0: 1868 level1: 1 duration1: 2900",
13:39:43.465 ->     "Index : 14 level0: 0 duration0: 1868 level1: 1 duration1: 4332",
13:39:43.499 ->     "Index : 15 level0: 0 duration0: 1868 level1: 1 duration1: 3140",
13:39:43.499 ->     "Index : 16 level0: 0 duration0: 1868 level1: 1 duration1: 1945",
13:39:43.499 ->     "Index : 17 level0: 0 duration0: 1868 level1: 1 duration1: 4564",
13:39:43.499 ->     "Index : 18 level0: 0 duration0: 1876 level1: 1 duration1: 2652",
13:39:43.499 ->     "Index : 19 level0: 0 duration0: 1868 level1: 1 duration1: 4",
13:39:43.499 ->     "Index : 20 level0: 0 duration0: 4 level1: 1 duration1: 3132",
13:39:43.499 ->     "Index : 21 level0: 0 duration0: 1868 level1: 1 duration1: 0",
13:39:43.567 ->     "Index : 22 level0: 0 duration0: 44 level1: 0 duration1: 0",
13:39:43.567 ->     "Index : 23 level0: 0 duration0: 0 level1: 0 duration1: 0",
```

lbernstone, 2019-03-18:
https://en.wikipedia.org/wiki/Oscilloscope

bertmelis, 2019-03-19:
Or [logic analyzer](https://www.aliexpress.com/premium/logic-analyzer.html?site=glo&d=y&origin=y&spm=2114.search0204.0.0.485e740beAe6C1&filterCat=50502%2C153713%2C50503&groupsort=1&SortType=price_asc&SearchText=logic+analyzer&initiative_id=SB_20190318225510&isViewCP=y) with [Sigrok](https://sigrok.org/)

negativekelvin, 2019-03-26:
Your idle threshold is too low, 125us but sync pulse is 170us.  Your filter threshold is too low, 12.5ns but should be around 1 tick = 3us.

yanzixiang, 2019-03-27:
> Your idle threshold is too low, 125us but sync pulse is 170us. Your filter threshold is too low, 12.5ns but should be around 1 tick = 3us.
> 
> But it would decode to 08E94F79

thanks, can you explain more details ?

negativekelvin, 2019-03-27:
>As far as I can see, 0 need 5 ticks LOW and 7ticks HIGH ?
≥5 but not necessarily 5

993+1868 = 2861 * 12.5ns / 3000ns = 12 => 0

yanzixiang, 2019-04-18:
> But you have to measure the sync pulse to get the tick time first. I just guess based on other values.
can you help to explain how to measure the sync pulse ?

negativekelvin, 2019-04-18:
What is the new output after you changed the threshholds?

yanzixiang, 2019-04-22:
> What is the new output after you changed the threshholds?

I will change the config to this,
And have a look.

  rmt_rx.rx_config.filter_ticks_thresh = 240; //3us
  rmt_rx.rx_config.idle_threshold = 20000; //250us

When I change the config,
It seems that the RMT can NOT stop,
because the sensor nerver stop sending the sent pluses,
so i have to change it back.

closed 2019-05-29 by yanzixiang