espressif/arduino-esp32 #2601: Crash on BluetoothSerial.begin

dquadros, 2019-03-21:
### Hardware:
Board:							DOIT ESP32 Dev Module
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Computer OS:                                        Windows 10

### Description:
 BluetoothSerial.begin will crash if called again after  BluetoothSerial.end

Sketch bellow uses a switch connected to GPIO 39 to trigger the bug

### Sketch:
```cpp

#include "BluetoothSerial.h"

BluetoothSerial SerialBT;
int key;

void setup() {
  Serial.begin (115200);
  Serial.println ("Ready");
  delay(5000);
  SerialBT.begin ("TEST_ESP32");
  pinMode (39, INPUT);
  key = digitalRead(39);
  Serial.println ("Ready");
}

void loop() {
  if (key != digitalRead(39)) {
    // chance in switch triggers test
    key = digitalRead(39);
    Serial.println ("Stopping BT");
    SerialBT.end();
    delay (3000);
    Serial.println ("Re-starting BT");
    SerialBT.begin ("TEST_ESP32");
    Serial.println ("OK!");
  }
}
```

### Debug Messages:
```
Ready
[I][BluetoothSerial.cpp:160] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:227] esp_spp_cb(): ESP_SPP_START_EVT
Stopping BT
Re-starting BT
OK!
[I][BluetoothSerial.cpp:160] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:227] esp_spp_cb(): ESP_SPP_START_EVT
ASSERT_PARAM(2048 0), in rwbt.c at line 311
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40084b22  PS      : 0x00060034  A0      : 0x80088c94  A1      : 0x3ffbe410  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdd48  A7      : 0x60008050  A8      : 0x80084b1d  A9      : 0x3ffbe3f0  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40084a55  LEND    : 0x40084a5c  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x401856a6  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40084b22

Backtrace: 0x40084b22:0x3ffbe410 0x40088c91:0x3ffbe430 0x40089097:0x3ffbe450 0x4008168d:0x3ffbe470 0x401856a3:0x00000000

Core 1 register dump:
PC      : 0x400d2a82  PS      : 0x00060f34  A0      : 0x8008fd48  A1      : 0x3ffc9200  
A2      : 0x3ffc5158  A3      : 0x00000000  A4      : 0x00060023  A5      : 0x3ffbbbe0  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d2a94  A9      : 0x3ffc91d0  
A10     : 0x3ffc4f24  A11     : 0x00000001  A12     : 0x00000020  A13     : 0x80000020  
A14     : 0x00000000  A15     : 0x3ffba3d8  SAR     : 0x00000007  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d2a82:0x3ffc9200 0x4008fd45:0x3ffc9220

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5848
entry 0x40080698
Ready
```
### Decoding:
```
PC: 0x40084b22
EXCVADDR: 0x00000000

Decoding stack results
0x401856a3: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/pm_esp32.c line 487
```

copercini, 2019-03-23:
It seems because you aren't feeding the WDT in the loop(), try with:

```
void loop() {
  if (key != digitalRead(39)) {
    // change in switch triggers test
    key = digitalRead(39);
    Serial.println ("Stopping BT");
    SerialBT.end();
    delay (3000);
    Serial.println ("Re-starting BT");
    SerialBT.begin ("TEST_ESP32");
    Serial.println ("OK!");
  }
  delay(20); //extra delay to feed WTD
}
```

dquadros, 2019-03-25:
I tried adding the delay(), but it made no difference.

I would be surprised if it did, because the loop will run "forever" without the delay as long as I don't trigger the end() / begin().  I also tried putting a delay(3000) between the begin() ant the println("OK"), the panic happens before the OK is printed.

I believe that the WDT is timing out inside the ASSERT_PARAM(2048 0), so the real problem is whatever is causing this assert to fail.

Anyway, thanks for the suggestion.

sebdelsol, 2019-05-11:
Still crashing

closed 2019-08-14 by stale[bot]

rgot-org, 2019-11-10:
i had the same issue, i fixed it here https://github.com/espressif/arduino-esp32/issues/2718#issuecomment-552213656