espressif/arduino-esp32 #2602: The problem with using putChar, getChar in Preferences

DeltaVetal26, 2019-03-22:
Hi,

I have use NVS storage with save configuration settings.
My test Code:
```cpp
#include <WiFi.h>
#include <Preferences.h>

Preferences preferences;
char p_key[5];
char p_name[5] = "Alex";

void setup() {
  preferences.begin("conf", false);
    p_key = preferences.getChar("p_key", 0); // Get key
    preferences.putChar("name", p_name); // Write Name
  preferences.end();
}

void loop() {}
```

I get compilling error:
```txt
 incompatible types in assignment of 'int8_t {aka signed char}' to 'char [5]'
```
Idea`s?

stickbreaker, 2019-03-22:
Do you want a single char or a string of char?

stickbreaker, 2019-03-22:
```c++
Uint16_t outlen = preferences.getString(key, (char*)outbuf, maxlen)
```

closed 2019-08-15 by stale[bot]