espressif/arduino-esp32 #2603: timerBegin () / timerAlarmWrite () do not work with frequencies higher than 500 khz. (IDFGH-812)

emilts, 2019-03-22:
Hi.

I need to generate pulses of 75 nanoseconds and a function that, depending on the number of pulses counted, does certain operations.

I am using the functions timerBegin, timerAlarmWrite and timerAttachInterrupt, which work perfectly until the pulse width decreases to 2 microseconds. From this value I can not make the pulses smaller, checking it with the oscilloscope.

I am using the ESP-WROOM-32 and the code used is as follows:

const int        ledPin     = 13;
volatile uint8_t ledOn   = 1;

//------------------- Variables para el timer ----------------------

hw_timer_t * timer    = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

//------------------- función digitalWrite ------------------------

extern void IRAM_ATTR __digitalWrite(uint8_t pin, uint8_t val){
    if(val) {
        if(pin < 32) {
            GPIO.out_w1ts = ((uint32_t)1 << pin);
        } else if(pin < 34) {
            GPIO.out1_w1ts.val = ((uint32_t)1 << (pin - 32));
        }
    } else {
        if(pin < 32) {
            GPIO.out_w1tc = ((uint32_t)1 << pin);
        } else if(pin < 34) {
            GPIO.out1_w1tc.val = ((uint32_t)1 << (pin - 32));
        }
    }
}

//--------------------- función de control -------------------------

void IRAM_ATTR onTimer(){
  portENTER_CRITICAL_ISR(&timerMux);

  __digitalWrite (ledPin, ledOn ^= 1);
 
  portEXIT_CRITICAL_ISR(&timerMux);
}

//=======================================================

void startTimer() {
  timer = timerBegin(0, 80, true);

  timerAttachInterrupt(timer, &onTimer, true);
  timerAlarmWrite(timer, 2, true);

  yield();
  timerAlarmEnable(timer);
}

//=======================================================

void setup() {

  pinMode(ledPin, OUTPUT);
  
  startTimer();
}

//=======================================================

void loop() {
  vTaskDelay(portMAX_DELAY);
}

stickbreaker, 2019-03-22:
at 500KHz, with a 240MHz CPU clock speed, you only have 240,000,000 / 500,000 or 480 clock cycles between interrupts. the FreeRTOS implementation used by the ESP32 uses a 1ms task slice, so  240,000,000 / 1,000.  This means a task gets 240,000 clock cycles for program execution per slice.  The libraries and hardware HAL layers take this into account when they were designed.  I don't know the cycle cost of an interrupt event, but I would assume it is in the high double digits or low triple digits.  At your frequency, most of the time the processor is jumping into and out of the ISR, I doubt it is executing the foreground task efficiently. 

I don't think your are reporting a problem, as much as you are reporting a capability limit.  Your stated 75ns rate would only allow 240,000,000 /(1/0.000000075) OR 18 CLOCKS?

I would not expect a the ESP32 system to be stable with a continuous interrupt rate greater than 50Khz.  At this rate, there is only 4,800 clock cycles between events, hardly enough time to task switch to ISR, execute the ISR, task switch Back to foreground task.

If you look at the hardware design of the ESP32 you will see all of the peripherals are designed to offload operations to dedicated hardware, the serial UARTS have 1,024 bytes of dedicated FIFO memory shared between three UARTs. The current software design allocates 256 bytes to each UART, 128 in, 128 out.  This means the UART only INTERRUPTS the CPU when 128bytes have moved. At 115,200 baud that is only a 90Hz interrupt rate. The I2C subsystem is more limited, it only uses 32 byte Fifo's so, at 400KHz operation that is a 1,250Hz interrupt rate. Still way, Way, below 50,000Hz.
  
If you need 75ns operations you are going to have to use dedicated hardware, your 75ns period is over 13MHz frequency? 

Chuck.

emilts, 2019-08-27:
Well, in the end I am using PWM to change the frequencies at which those pulses should be emitted.
With each different frequency I recalculate the "duty cycle", so that the pulse is always about 75ns.
In this way I found no problems and the measurements with the oscilloscope are correct.

closed 2019-08-27 by emilts