espressif/arduino-esp32 #2604: WiFiClientSecure connection to IP address(Connection Fails)

telliottosceola, 2019-03-22:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Use WiFiClientSecure to connect to an IP address using CACert.

Describe your system( Hardware, computer, O/S, core version, environment).
ESP32 ESP-WROOM-32 Arduino Platformio using Adafruit_Feather32

Describe what is failing.
WiFiClientSecure(IPaddress,Port) fails to connect to the remote server.

Show the shortest possible code that will duplicate the error.
I was able to simply replicate this issue by modifying the WiFiClientSecure.ino example.  I obtained the IP address for www.howsmyssl.com and used that IP in the WiFiClientSecure.connect function.  Full code available here:
https://github.com/telliottosceola/WiFi_Client_Secure_IP_Error

Show the EXACT error message(it doesn't work is not enough).
Debug level set to 5 outputs the following when executing the code:
```
Attempting to connect to SSID: Travis-WiFi
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START

.[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 192.168.1.21, MASK: 255.255.255.0, GW: 192.168.1.1
Connected to Travis-WiFi

Starting connection to server...
[V][ssl_client.cpp:53] start_ssl_client(): Free internal heap before TLS 288964
[V][ssl_client.cpp:55] start_ssl_client(): Starting socket
[E][ssl_client.cpp:82] start_ssl_client(): Connect to Server failed!
[E][WiFiClientSecure.cpp:118] connect(): start_ssl_client: -1
[V][ssl_client.cpp:243] stop_ssl_socket(): Cleaning SSL connection.
Connection failed!
```

Thank you very much for taking the time to review this.  I ran into this issue while trying to use the PubSub client to make a secure connection to a local(LAN) server.  I tracked it down to this issue.

lbernstone, 2019-03-23:
WiFiClientSecure has a verify method to test certificates. I have also written an [app](https://github.com/lbernstone/setCerts) to test mqtt connections. Troubleshooting SSL is outside the scope of this forum.

copercini, 2019-03-23:
This error is because the socket is not able to connect to your IP, a network error not released with WiFiClientSecure 

And even it connecting, SSL verifies the domain name, if the server of www.howsmyssl.com is really from www.howsmyssl.com, but using only the IP this will not work, because the domain name is part o verification

telliottosceola, 2019-03-25:
Thank you @lbernstone and @copercini very much for your input.

How would this function be used in normal instances then?
int connect(IPAddress ip, uint16_t port, const char *rootCABuff, const char *cli_cert, const char *cli_key);

Is it possible to sign an SSL Certificate with an IP address?  I do not know much about SSL(just a low level C/C++ developer).  If a domain name is part of verification then I do not understand how the above function would ever work unless the certificate could be created using the IP address.

copercini, 2019-03-25:
> Is it possible to sign an SSL Certificate with an IP address? 
It's possible but unusual 

The `connect(IPAddress ip...` is for these unusual cases, you can get more details here: https://stackoverflow.com/questions/2043617/is-it-possible-to-have-ssl-certificate-for-ip-address-not-domain-name

telliottosceola, 2019-04-02:
I'm going to go ahead and close this issue.  Seems like the issue I was having is more of an issue with TLS standards as a whole rather than an issue with WiFiClientSecure.

closed 2019-04-02 by telliottosceola