espressif/arduino-esp32 #2609: Feature Request... Synchronize the WiFi (And WiFiMulti) APIs between ESP 8266 and ESP 32

dachshund-digital, 2019-03-25:
Feature Request... Synchronize the WiFi (And WiFiMulti) APIs between ESP 8266 and ESP 32.  This would be a good thing.  I just finished writing programming logic identical on ESP 32 and ESP 8266 devices, and frankly there is not good reason the APIs differ.  It is quite user unfriendly.  I wasted a lot of time having to use #defines and such to deal with the variances between the APIs in question.

fabiuz7, 2019-03-25:
What are the exact differences?

rpannekoek, 2019-04-01:
One difference I'm aware of is `WiFi.hostname()`  vs `WiFi.setHostName()` for ESP8266 and ESP32, respectively.

dachshund-digital, 2019-04-02:
Unfortunately, I, for one, cannot publish deltas, since the development done, is privately owned.  That said, there are quite a few differences... And I don't think I need to itemize them here, regardless if I could nor not.  Someone, one of the major contributors or supports of the ESP8266 Wifi and WifiMulti modules should do the analysis.  Moreover the major contributors for ESP32 and ESP8266 should coordinate together to address this feature request.

It is obvious that the two APIs should aligned and consistently implemented.  Frankly, the fact they are not already is odd, because almost every ESP8266 module solution is applicable to ESP32 modules.  It makes no sense not to address this.  It would have saved me, for example, 100s lines of code.  making transparent library use between the various ESP modules should be a strategic goal for all contributors to the growing ESP* APIs.

Oh, the same can be said for the WebServer modules, where ESP32 and ESP8266 deltas exist as well, although they are not as frequent in total.

closed 2019-08-15 by stale[bot]