espressif/arduino-esp32 #2612: Stability of rom_phy_get_vdd33 ???

robertgregor, 2019-03-25:
Hi,

seems I have a problem with rom_phy_get_vdd33() and it's reading. My WiFi is always ON in STA mode, connected to AP. However, seems, that call to rom_phy_get_vdd33() get correct response only if there is i.e. socket opened. If there is no socket opened, the reading is 0, even if ESP is connected to AP. 

I have also tried:

    btStart();
    int vdd = rom_phy_get_vdd33();
    btStop();

but the result is the same. Has anyone some trick or description, why this is happening? I am thinking, that there is maybe some ESP current management functionality, so when there is no incomming / outgoing connection, then the WiFi is in some sleep mode and thus the reading is 0???

Thanks Robert

lbernstone, 2019-03-25:
No driver = no phy = no voltage.  You can start the driver with (from WiFiGeneric.cpp):
```
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
        esp_err_t err = esp_wifi_init(&cfg);
```

robertgregor, 2019-03-26:
Hi, OK, thanks for the answer. However, when I connect to WiFi, then the driver is loaded isn't it...
I am doing:
void connectToWifiNoWait(String ssid, String pwd, String ip, String gw, String mask, String dns1, String dns2, String bssid, String chnl ) {
  if (WiFi.status() == WL_CONNECTED) return;
  logSketch('I', PSTR("Connecting wifi..."));
  long st = millis();
  WiFi.persistent(true);
  WiFi.mode(WIFI_STA);
  if (ip.length() != 0) WiFi.config(getIpAddr(ip), getIpAddr(gw), getIpAddr(mask), getIpAddr(dns1), getIpAddr(dns2));
  if ((ssid.length() != 0) && (pwd.length() == 0)) {
      WiFi.begin(ssid.c_str());
  } else if ((ssid.length() != 0) && (pwd.length() != 0)) {
    if (bssid[0] == 0) {
      WiFi.begin(ssid.c_str(), pwd.c_str());
    } else {
      byte iparr[6];
      for (int i=0; i<6; i++) {
         String part = "0x" + bssid.substring(0,bssid.indexOf(':'));
         bssid.remove(0, bssid.indexOf(':')+1);
         iparr[i] = strtol(part.c_str(),NULL,0);               
      }
      WiFi.begin(ssid.c_str(), pwd.c_str(), chnl.toInt(), iparr);
    }
  }  
  WiFi.persistent(false);

lbernstone, 2019-03-26:
Sorry, misread your issue.  Unfortunately, rom_phy_get_vdd33 is an unsupported command.  This is probably because it is unreliable.  Set up a voltage divider and read your battery level on an adc pin.

robertgregor, 2019-03-26:
Yes. I know, it is not supported command, however it is there and I am trying to have stable and working replacement of ESP8266 getVcc()...
So I am trying to figure out, how it works.
It is interesting. If my wifi is not connected to AP, then this works perfectly and it is 100% stable:

btStart();
int vdd = rom_phy_get_vdd33();
btStop();

However, when WiFi is connected to AP, then the reading is not stable and sometimes I will get reasonable result, sometimes not. Is it possible to somehow disable the WiFi for reading, but keep the TCP connections open? Or swith to BT and then back to WiFi without loosing the existing connections?

closed 2019-08-15 by stale[bot]