espressif/arduino-esp32 #2613: arduino API incompatibility for BSSID and macAddress

aster94, 2019-03-25:
working in a project i noted that the WIFI api is different and i thought to open an issue if you would like to fix it, all the resources are here:

BSSID official API
site: https://www.arduino.cc/en/Reference/WiFiBSSID 
code: https://github.com/arduino-libraries/WiFi/blob/master/src/WiFi.cpp#L189
macAddress
site: https://www.arduino.cc/en/Reference/WiFiMACAddress
code: uint8_t* WiFiClass::macAddress(uint8_t* mac)

ESP32
BSSID: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.cpp#L521 and https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiScan.cpp#L234
macAddress: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.cpp#L416

aster94, 2019-03-30:
@me-no-dev would you like to accept a PR with a fix to make it compatible with the arduino API?

closed 2019-08-15 by stale[bot]

aster94, 2019-08-17:
Just commenting to suggest to reopen it

atanisoft, 2019-08-17:
@aster94 feel free to reopen the issue and send in a PR for it.