espressif/arduino-esp32 #2614: Sparkfun ESP32 Thing: Timed out waiting for packet header

lilcommieblyat, 2019-03-26:
### Hardware:
Board:							Sparkfun ESP32 Thing

IDE name:						Arduino IDE 1.8.8
Flash Frequency:					80Mhz
Upload Speed:					115200
Computer OS:                                          Windows 10

### Description:
Describe your problem here

I have a Sparkfun ESP32 Thing board and this is my first Sparkfun product. 
To start my project I followed the instruction on Sparkfun's website: https://learn.sparkfun.com/tutorials/esp32-thing-hookup-guide
Everything went OK until I tried to upload the blink example code to the board. 
Here is the code: 

### Sketch:  (leave the backquotes for [code formatting]

int ledPin = 5;

void setup()
{
    pinMode(ledPin, OUTPUT);
    Serial.begin(115200);
}

void loop()
{
    Serial.println("Hello, world!");
    digitalWrite(ledPin, HIGH);
    delay(500);
    digitalWrite(ledPin, LOW);
    delay(500);
}

The error message I received is: 

Archiving built core (caching) in: C:\Users\Alienware\AppData\Local\Temp\arduino_cache_663458\core\core_287c685320db3c376f0bf5c727e873af.a
Linking everything together...
"C:\\Users\\Alienware\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -nostdlib "-LC:\\Users\\Alienware\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/lib" "-LC:\\Users\\Alienware\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group "C:\\Users\\Alienware\\AppData\\Local\\Temp\\MW_ArduinoHWSetup\\MW_ArduinoBuild\\sketch\\blinkk.ino.cpp.o" "C:\\Users\\Alienware\\AppData\\Local\\Temp\\MW_ArduinoHWSetup\\MW_ArduinoBuild\\core\\core.a" -lgcc -lopenssl -lbtdm_app -lfatfs -lwps -lcoexist -lwear_levelling -lesp_http_client -lprotobuf-c -lhal -lnewlib -ldriver -lbootloader_support -lpp -lfreemodbus -lmesh -lsmartconfig -ljsmn -lwpa -lethernet -lphy -lfrmn -lapp_trace -lfr_coefficients -lconsole -lulp -lwpa_supplicant -lfreertos -lbt -lmicro-ecc -lesp32-camera -lcxx -lxtensa-debug-module -ltcp_transport -lmdns -lvfs -lmtmn -lespcoredump -lesp_ringbuf -lsoc -lcore -lfb_gfx -lsdmmc -llibsodium -lcoap -ltcpip_adapter -lprotocomm -lesp_event -limage_util -lc_nano -lesp-tls -lasio -lrtc -lspi_flash -lwpa2 -lwifi_provisioning -lesp32 -lface_recognition -lapp_update -lnghttp -lspiffs -lface_detection -lefuse -lunity -lesp_https_server -lespnow -lnvs_flash -lesp_adc_cal -llog -ldl_lib -lsmartconfig_ack -lexpat -lfd_coefficients -lm -lmqtt -lc -lheap -lmbedtls -llwip -lnet80211 -lesp_http_server -lpthread -ljson -lesp_https_ota -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\\Users\\Alienware\\AppData\\Local\\Temp\\MW_ArduinoHWSetup\\MW_ArduinoBuild/blinkk.ino.elf"
"C:\\Users\\Alienware\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/gen_esp32part.exe" -q "C:\\Users\\Alienware\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/partitions/default.csv" "C:\\Users\\Alienware\\AppData\\Local\\Temp\\MW_ArduinoHWSetup\\MW_ArduinoBuild/blinkk.ino.partitions.bin"
"C:\\Users\\Alienware\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/esptool/esptool.exe" --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\Alienware\\AppData\\Local\\Temp\\MW_ArduinoHWSetup\\MW_ArduinoBuild/blinkk.ino.bin" "C:\\Users\\Alienware\\AppData\\Local\\Temp\\MW_ArduinoHWSetup\\MW_ArduinoBuild/blinkk.ino.elf"
esptool.py v2.6

"C:\\Users\\Alienware\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-size" -A "C:\\Users\\Alienware\\AppData\\Local\\Temp\\MW_ArduinoHWSetup\\MW_ArduinoBuild/blinkk.ino.elf"
Sketch uses 209855 bytes (16%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13032 bytes (3%) of dynamic memory, leaving 314648 bytes for local variables. Maximum is 327680 bytes.
C:\Users\Alienware\Documents\Arduino\hardware\espressif\esp32/tools/esptool/esptool.exe --chip esp32 --port COM11 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\Alienware\Documents\Arduino\hardware\espressif\esp32/tools/partitions/boot_app0.bin 0x1000 C:\Users\Alienware\Documents\Arduino\hardware\espressif\esp32/tools/sdk/bin/bootloader_dio_80m.bin 0x10000 C:\Users\Alienware\AppData\Local\Temp\MW_ArduinoHWSetup\MW_ArduinoBuild/blinkk.ino.bin 0x8000 C:\Users\Alienware\AppData\Local\Temp\MW_ArduinoHWSetup\MW_ArduinoBuild/blinkk.ino.partitions.bin 
esptool.py v2.6
Serial port COM11
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

My laptop recognizes the board successfully. However, no matter how I press the boot button (button 0) or the Reset button during the upload, I got the same result every time. I tried to reinstall everything including the Arduino IDE but nothing comes to work. 
I found one possible solution here:  https://forum.arduino.cc/index.php?topic=558213.0 
which tells me to lower the baud rate but still, didn't work for me. 

Can someone help me with this situation? I have never encountered something like this before......

lbernstone, 2019-03-26:
Getting a COM port only means that your computer recognizes the UART bridge chip, not the rest of the board.  Open a command prompt and run `python esptool.py chip_id`.  As mentioned in the other post, hold down both buttons, then release the reset, and try the chip_id again.  Sparkfun generally has good QC, but things do break...
```
$:~/esp32/arduino-esp32/tools$ python esptool.py chip_id
esptool.py v2.6
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
MAC: 30:ae:a4:36:c7:18
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 30:ae:a4:36:c7:18
Hard resetting via RTS pin...
```

lilcommieblyat, 2019-03-26:
> Getting a COM port only means that your computer recognizes the UART bridge chip, not the rest of the board. Open a command prompt and run `python esptool.py chip_id`. As mentioned in the other [...]

Thanks for the comment. I just tried your method but unfortunately it didn't work for me. The message in my command prompt was exactly the same as what I see in the Arduino IDE.

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32\tools>python esptool.py chip_id
esptool.py v2.6
Found 1 serial ports
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____....._____
COM3 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header
A fatal error occurred: All of the 1 available serial ports could not connect to a Espressif device.

However, I tried coding with  microPython and it worked perfectly. I was confused.

closed 2019-08-15 by stale[bot]

iwojapa, 2019-09-13:
I'm having this problem in Mac OS 10.15 Catalina beta (19A558d). It was working fine a couple on months ago in Mojave.