espressif/arduino-esp32 #2615: Bluetooth a2dp   ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4008e878

Flo082002, 2019-03-26:
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Upload Speed:					115200
Computer OS:                                        Windows 10

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10312
load:0x40080400,len:6460
entry 0x400806a4
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4008e878
file: "D:\Users\Flo\Desktop\Verstärker\Esp32\a2dp_sink\a2dp_sink.ino" line 89
func: void setup()
expression: esp_bt_controller_mem_release(ESP_BT_MODE_BLE)

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4008e348:0x3ffb1f00 0x4008e87b:0x3ffb1f20 0x400d14d8:0x3ffb1f40 0x400d1b5f:0x3ffb1fb0 0x40093201:0x3ffb1fd0

[a2dp_sink.zip](https://github.com/espressif/arduino-esp32/files/3008231/a2dp_sink.zip)

Petros144, 2019-03-31:
Same here, i think its something in line 89:

  ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));

maybe someone can help... looking for the same...

lbernstone, 2019-03-31:
Use btStart() in arduino-esp32

Flo082002, 2019-04-03:
Where do I have to usetbStart() ?

atanisoft, 2019-04-03:
Maybe instead of calling esp_bt_controller_mem_release you can call btStop() instead? This will take care of the release of data as well.

Flo082002, 2019-04-03:
> 
> 
> Maybe instead of calling esp_bt_controller_mem_release you can call btStop() instead? This will take care of the release of data as well.

No, thats not working.
Now the ESP is no longer displayed on my Smartphone in the Bluetooth device list.

atanisoft, 2019-04-03:
> No, thats not working.
> Now the ESP is no longer displayed on my Smartphone in the Bluetooth device list.
Normally the only reason to call esp_bt_controller_mem_release is when you don't want to use any of the BT stack. Without the memory allocated it is not really usable at all.

Flo082002, 2019-04-03:
> Normally the only reason to call esp_bt_controller_mem_release is when you don't want to use any of the BT stack. Without the memory allocated it is not really usable at all.

I used th example from the IDF.
In the example there is esp_bt_controller_mem_release

atanisoft, 2019-04-03:
> I used the example from the IDF.
> In the example there is esp_bt_controller_mem_release

There are very likely a number of other lines that come before it that make it an IDF example. There may be a fix to use this in an arduino-esp32 project, such as btStart(), but are you using arduino-esp32?

Flo082002, 2019-04-03:
> but are you using arduino-esp32?

I use the Arduino Ide with arduino-esp32

atanisoft, 2019-04-03:
Part of the problem is that you are trying to use the IDF example "as-is" by simply renaming app_main to setup. That won't work exactly. You can call btStart() to replace [these lines](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/a2dp_sink/main/main.c#L88-L99) and remove [these lines](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/a2dp_sink/main/main.c#L47-L53) entirely.

Petros144, 2019-04-03:
tried it out, but no sucsess, still same error...

Flo082002, 2019-04-04:
Thanks a lot for your help. Now It is working. But it only works, when I use I2S_MODE_PDM in the i2s config.
But the audio signal is
pretty noisy when I play something.
Besides, the sounds are weird.
When I play a sine wave, the frequency sounds deeper.
In addition, I can see on an oscilloscope that the lower half of the sine is compressed and the upper one is pulled wide.

Here is my new code:
[a2dp_sink.zip](https://github.com/espressif/arduino-esp32/files/3043296/a2dp_sink.zip)

Petros144, 2019-04-04:
@Flo082002 

Im getting this error:

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
assertion "dataSize == readSize" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_storage.cpp", line 419, function: esp_err_t nvs::Storage::readMultiPageBlob(uint8_t, const char*, void*, size_t)
abort() was called at PC 0x40143cd3 on core 0

any idea? updated the ESP SW from board manager to 1.0.1

Flo082002, 2019-04-05:
Try to use my code.
https://github.com/espressif/arduino-esp32/files/3043296/a2dp_sink.zip

My code is working on my esp.

Flo082002, 2019-04-09:
> any idea? updated the ESP SW from board manager to 1.0.1

Erease the flash of the ESP before flashing the BLuetooth program.

Petros144, 2019-04-11:
tried on a fresh Win Instalation and worked fine... maybe its my ESP Lib thats bricked.

ps: did you know how to read out AVRCP informations? like title, album, playtime?

closed 2019-08-15 by stale[bot]