espressif/arduino-esp32 #2616: Incompatibility Webserver and NetBIOS

luc-github, 2019-03-26:
This is an old bug, which exists since the begining of NetBIOS and Webserver on ESP32.
 
With NetBIOS enabled, Webserver stop responding after 10 to 15 min.

Board:							ESP32 Dev Module
Core Installation version:			all versions and latest git
IDE name:						Arduino IDE (any version)
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 7/10

### MCVE:
Use modified HelloServer.ino example with NetBIOS

Just wait 10 to 15 min and  WebServer stop responding - no crash - no error, need to restart board to recover Webserver during few minutes.

Note: I did a sync version of NetBIOS before the async was released and there is no issue on mine but it is sync - prefered version for ESP32 was async per @me-no-dev 

```
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <NetBIOS.h>

const char* ssid = "........";
const char* password = "........";

WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, 0);
}

void handleNotFound() {
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void) {
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  NBNS.begin("ESP");

  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
}

```

luc-github, 2019-04-13:
seems fixed in latest GIT as I cannot reproduce anymore

closed 2019-04-13 by luc-github

hr-kapanakov, 2021-01-06:
I just hit the same issue. If I remove the `NBNS.begin("ESP");` everything is working normally. The only difference is that I am using `AsyncWebServer`.