espressif/arduino-esp32 #2617: ArduinoOTA throwing class has no member errors using example code

treii28, 2019-03-26:
I am trying to update old code to work on ESP32 platform (originally on ESP8266)  so I completely replaced all of the OTA specific code using the example in the BasicOTA sketch in the espressif framework code imported into platformio.

I am creating and compiling my code on a Linux Ubuntu AMD machine using an updated version of platformio with code built in CLion.

When I try to compile using the Example code I get a series of errors:

```
src/main.cpp: In function 'void initOTA()':
src/main.cpp:279:6: error: 'class ArduinoOTAClass<WiFiServer, WiFiClient>' has no member named 'onStart'
.onStart([]() {
^
src/main.cpp: In lambda function:
src/main.cpp:281:22: error: 'class ArduinoOTAClass<WiFiServer, WiFiClient>' has no member named 'getCommand'
if (ArduinoOTA.getCommand() == U_FLASH)
^
src/main.cpp:281:38: error: 'U_FLASH' was not declared in this scope
if (ArduinoOTA.getCommand() == U_FLASH)
^
src/main.cpp: In function 'void initOTA()':
src/main.cpp:295:17: error: 'ota_error_t' has not been declared
.onError([](ota_error_t error) {
^
src/main.cpp: In lambda function:
src/main.cpp:297:20: error: 'OTA_AUTH_ERROR' was not declared in this scope
if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
^
src/main.cpp:298:25: error: 'OTA_BEGIN_ERROR' was not declared in this scope
else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
^
src/main.cpp:299:25: error: 'OTA_CONNECT_ERROR' was not declared in this scope
else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
^
src/main.cpp:300:25: error: 'OTA_RECEIVE_ERROR' was not declared in this scope
else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
^
src/main.cpp:301:25: error: 'OTA_END_ERROR' was not declared in this scope
else if (error == OTA_END_ERROR) Serial.println("End Failed");
^
src/main.cpp: In function 'void initOTA()':
src/main.cpp:304:20: error: no matching function for call to 'ArduinoOTAClass<WiFiServer, WiFiClient>::begin()'
ArduinoOTA.begin();
^
In file included from src/main.cpp:11:0:
lib/ArduinoOTA/src/ArduinoOTA.h:49:8: note: candidate: void ArduinoOTAClass<NetServer, NetClient>::begin(IPAddress, const char*, const char*, OTAStorage&) [with NetServer = WiFiServer; NetClient = WiFiClient]
void begin(IPAddress localIP, const char* name, const char* password, OTAStorage& storage) {
^
lib/ArduinoOTA/src/ArduinoOTA.h:49:8: note:   candidate expects 4 arguments, 0 provided

```

The code I extracted from the example is placed in an 'initOTA()' void function called from the setup() loop:

```
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

void initOTA() {
    DBG_OUTPUT_PORT.println("setting OTA handlers");
  // Port defaults to 3232
  // ArduinoOTA.setPort(3232);

  // Hostname defaults to esp3232-[MAC]
  // ArduinoOTA.setHostname("myesp32");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA
    .onStart([]() {
      String type;
      if (ArduinoOTA.getCommand() == U_FLASH)
        type = "sketch";
      else // U_SPIFFS
        type = "filesystem";

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      Serial.println("Start updating " + type);
    })
    .onEnd([]() {
      Serial.println("\nEnd");
    })
    .onProgress([](unsigned int progress, unsigned int total) {
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    })
    .onError([](ota_error_t error) {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
      else if (error == OTA_END_ERROR) Serial.println("End Failed");
    });

  ArduinoOTA.begin();
  DBG_OUTPUT_PORT.println("finished setting up OTA handlers");
}

void setup()
{
    initDebug(); // start serial
    initDNSServer();
    initMDNS();
    //initWebserver();
    initSPIFFS();
    initOTA();
    initWifi();
}
```

I have tried adding and removing ArduinoOTA as a dependency in lib_deps in the platformio.ini file with no luck.

lbernstone, 2019-03-27:
Make sure you are using c++11 extensions for all those lambda expressions.
`-std=c++11`

closed 2019-08-15 by stale[bot]

zinen, 2020-03-17:
Are you by any change using platform.io? If yes, if found this info very clarifying on a similarly problem of mine. 
PlatformIO Community => [ESP8266 OTA Example Won’t Compile](https://community.platformio.org/t/esp8266-ota-example-wont-compile/10124/7).

It basically says to delete builtin libraries from platformio.ini file, delete folder`.pio` then retry compiling.

Nailik, 2021-07-11:
> Are you by any change using platform.io? If yes, if found this info very clarifying on a similarly problem of mine.
> PlatformIO Community => [ESP8266 OTA Example Won’t [...]

unfortunately that didn't help me at all, i still have the same problem :/