espressif/arduino-esp32 #2622: Automatic open browser after connect to AP to load index.html saved on the SPIFFS

williamesp2015, 2019-03-30:
I really having difficulty to have a automatic loading index.html saved on the SPIFFS. I need to automatically open index.html without user require open browser and type the IP.

  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    Serial.println("/");
    Serial.println("redirecting to /index.html");
    request->redirect("/index.html");
  });
  #if ESP8266
  if(!SPIFFS.begin()){
      Serial.println("SPIFFS Mount Failed");
  }
  #endif
  #if ESP32
  if(!SPIFFS.begin(true)){
      Serial.println("SPIFFS Mount Failed on ..");
  }
  #endif

  listDir(SPIFFS, "/", 0);
  server.serveStatic("/", SPIFFS, "/").setCacheControl("max-age=600").setDefaultFile("index.html");
 server.begin();
Thank you

lbernstone, 2019-03-30:
https://github.com/espressif/arduino-esp32/blob/master/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino

nkYellOw, 2019-04-08:
> https://github.com/espressif/arduino-esp32/blob/master/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino

WiFiServer with to much code. In real time all use WebServer.h and dont need to know while client.available()
But Wifiserver.h dont have open browser after connect to server

closed 2019-08-15 by stale[bot]