espressif/arduino-esp32 #2624: Best option for wireless serial communication?

karan-sood, 2019-03-31:
### Hardware:
Board:							Adafruit HUZZAH32 Feather 

### Description:
Not facing an issue as such, just want to gather feedback on my best option for communicating wirelessly with the serial, for the posting of data from an analog sensor. I hope to either use this data to generate XML for the real-time graphing of data on a web application or alternatively implement something like in this video, where an AVR breakout with an additional Anaren BLE module is used to send serial data to a native smartphone app:

https://www.youtube.com/watch?v=9IfFaX2Mr6c

Is this example something I can easily replicate with an ESP32? The digital output is just a 3 or 4 digit int which will be read every 500 ms. 

Thank you!

sansillusion, 2019-03-31:
You may want to look at the BluetoothSerial example.

karan-sood, 2019-03-31:
Thanks @sansillusion this is useful, however I'm definitely keener to utilise BLE instead if this is a possibility

karan-sood, 2019-04-02:
Hi @sansillusion , I'm making use of this and have attempted an implementation like so:

`#include "BluetoothSerial.h"
 
BluetoothSerial SerialBT;

void setup() {
  SerialBT.begin("ESP32");
}
 
void loop() {
  int sensorValue = analogRead(A1);
  SerialBT.println(sensorValue);
  delay(1000);
}`

It seems to be transmitting to the serial wirelessly, but is posting the same value constantly. This does not occur with the same loop, making use of a USB serial connection with Serial.begin(9600); 

Any ideas?

sansillusion, 2019-04-03:
I dont know the current state of analogRead support for esp32, I also dont
know to what pin A1 maps to on your esp32 module but make sure that it is a
pin on ADC1 not ADC2 because ADC2 can't be used while using the radio
subsystem. ADC1 pins are 32,33...up to 39.

karan-sood, 2019-04-03:
> I dont know the current state of analogRead support for esp32, I also dont know to what pin A1 maps to on your esp32 module but make sure that it is a pin on ADC1 not ADC2 because ADC2 can't be used [...]

This was the exact issue - I was connecting to an ADC2 GPIO. Thank you again

closed 2019-08-15 by stale[bot]