espressif/arduino-esp32 #2627: Can't upload sketch while ESP32 is connected to LED Display and Power Supply

GeorgeFlorian, 2019-04-01:
Hello !

I'm trying to use a SMD 2727 P5 64x32 LED Display with the ESP32. I've opted for the following library https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel

I've soldered the wires for the default constructor.
I've passing the wires through 2 Level Logic Converters, from 5V to 3.3V.

As you can say I don't use GPIO 1 and 3, which are UART 0 TX and UART 0 RX respectively.

But I can't upload a sketch while the ESP is mounted on the board and connected to the Power Supply. I have to take the ESP out and upload a sketch than put it back in.

How did this happen ?

I am sure I soldered the wires as it says in the above mentioned library.

Thank you !

Jeroen88, 2019-04-01:
Does your power supply deliver enough current?

GeorgeFlorian, 2019-04-01:
> Does your power supply deliver enough current?

I sure hope so, these are the specs:

https://i.postimg.cc/prt2vnW7/20190401-145102.jpg

Jeroen88, 2019-04-01:
That should be enough :D

GeorgeFlorian, 2019-04-01:
> That should be enough :D

Great ! Than what looks to be the problem ?

lbernstone, 2019-04-01:
https://desire.giesecke.tk/index.php/2018/07/06/reserved-gpios/

GeorgeFlorian, 2019-04-01:
> https://desire.giesecke.tk/index.php/2018/07/06/reserved-gpios/

Thank you !
I've read that before making the connections.

Also, this guy's [library](https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel) makes use of the GPIO 12. He's works, mine doesn't.

I don't know how to check if the GPIO12 is low or high... What does that mean ?

GeorgeFlorian, 2019-04-01:
Update: Now I can't upload anything even if the ESP32 is off the board and connected only to the USB Cable.

closed 2019-04-01 by GeorgeFlorian