espressif/arduino-esp32 #2628: ESP-IDF SDK need update

asetyde, 2019-04-01:
I register many bugs on WiFi and other peripherals that are closed with upgrade of SDK beta 3.3 b3, at now on 1.0.2 rc2 we use 3.3 b1 , is possible upgrade manually on my code or i must wait ? 

About new realese, I think you wait 3.3 sdk stable to update ? I correct ? 

At now we use more thank 2000 esp32 chips for our device but as esp8266 firmware stability is not so stable, complicating job of developer work on big scale use.

asetyde, 2019-04-12:
@me-no-dev  do you update to  v3.2 286202c ESP-IDF Release , I need this to resolve a bug on my system

asetyde, 2019-04-12:
I see commit on 3.2 , perfect. Thanks

closed 2019-04-12 by asetyde

me-no-dev, 2019-04-12:
yup 3.2 final is almost ready :) That means that 1.0.2 is also around the corner, but be aware that after release, I will sync master again with IDF master, so you might want to stick to 1.0.2 as a stable version :)