espressif/arduino-esp32 #2630: Upload doesn't work anymore ! ESP too hot to touch ! Is it broken ?

GeorgeFlorian, 2019-04-01:
Hello !

Earlier today I tried using the ESP Wroom-32 with a P5 SMD 2727 LED Display and the following library:
https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel

I've soldered the wires for that default constructor and tried uploading some codes to the ESP32.
Now, the board looked like this:
https://i.postimg.cc/tR6mRKZQ/20190329-164451.jpg

I've attached a power supply to it and used 2 LLC to shift the 3V3 accepted by the ESP32 to 5V needed for the LED Display to work.

https://i.postimg.cc/VkCHqbsd/20190401-135241.jpg

The RED (PIN 1) wire from the HUB75 is on R1 (IO25).

These are the specs of my power supply:

https://i.postimg.cc/prt2vnW7/20190401-145102.jpg

I noticed that I couldn't upload any code to the ESP32, because the boot button seemed like it did not work, so I took off the ESP from the board and uploaded the code than put it back in and noticed a bizarre behavior:

If the ESP32 is half-way inside [(like this)](https://i.postimg.cc/gcXndYgG/20190401-141209.jpg), the screen is [white](https://i.postimg.cc/hPVvz2Fw/20190401-141143.jpg) if I insert it all the way it get's [red](https://i.postimg.cc/1t41G38X/20190401-141233.jpg)

If I reset the ESP, the display will be off, and the Serial Monitor will output:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```
repeating itself without stopping.

This began to make me think of the GPIO 12 that is a bootstrapping pin, but I thought if the guy that made the library used GPIO 12 than I can use it to.

The ESP ran around 1h with that error, which I think was actually the ESP resetting every second, made it go so hot that I couldn't touch it. I disconnected it at let it cool off only to realize that I can't upload code to it anymore. 

I am now trying to upload code to the ESP, by having it connected to the PC via a USB Cable and keeping boot button pressed, but nothing happens.
Also it keeps getting very hot because the previous code is still making it reset itself.

Is there a way to check if the GPIO 12 is LOW or HIGH ?

Is the ESP broken ?

Flo082002, 2019-04-03:
> Is there a way to check if the GPIO 12 is LOW or HIGH ?
Use a multimeter.

> Is the ESP broken ?
Which part of the ESP is hot?
The voltage regulator or the ESP?

GeorgeFlorian, 2019-04-03:
> > Is there a way to check if the GPIO 12 is LOW or HIGH ?
> 
> Use a multimeter.
> 
> > Is the ESP broken ?
> 
> Which part of the ESP is hot?
> The voltage regulator or the ESP?
> 
> Is it getting hot, when you attach nothing to the ESP?

The chip on the ESP can get so hot that all the ESP gets hot as well.
It get's hot even if it's only connected to the PC via the USB cable.

Flo082002, 2019-04-03:
> The chip on the ESP can get so hot that all the ESP gets hot as well.
> It get's hot even if it's only connected to the PC via the USB cable.

I can touch my ESP without burning my fingers.
My thermometer says 39°C./ 102,2°F

GeorgeFlorian, 2019-04-04:
> > The chip on the ESP can get so hot that all the ESP gets hot as well.
> > It get's hot even if it's only connected to the PC via the USB cable.
> 
> I can touch my ESP without burning my fingers.
> My thermometer says 39°C./ 102,2°F
> I think there is something wrong with your ESP.
> 
> Did you remove the metal cover?
> The cover is also used as a heat sink.
> 
> Maybe the voltage regulator is broken and the ESP gets too high voltage.

What voltage regulator ? He gets power from the USB. A USB port can output 5V.

Flo082002, 2019-04-04:
> He gets power from the USB. A USB port can output 5V.

I mean the voltage regulator on ESP's the PCB.
[image]

GeorgeFlorian, 2019-04-04:
> > He gets power from the USB. A USB port can output 5V.
> 
> I mean the voltage regulator on ESP's the PCB.
> [image]

Can I check if it's broken or we just have to assume that ?

Flo082002, 2019-04-04:
Measure with a multimeter.

[image]

Pin 1 (green) = GND
Pin 2 (yellow) = 3. 3V
Pin 3 (blue) = USB voltage (4. 5V - 5. 5V)
Measure from pin 1 to pin 3. There should be 3. 3V.

GeorgeFlorian, 2019-04-05:
> Measure with a multimeter.
> 
> 
> 
> Pin 1 (green) = GND
> Pin 2 (yellow) = 3. 3V
> Pin 3 (blue) = USB voltage (4. 5V - 5. 5V)
> Measure from pin 1 to pin 3. There should be 3. 3V.

From pin 1 to 2 there are 3.25V and from pin 1 to 3 there are 4.35V.

Flo082002, 2019-04-05:
That's OK.
How much current does the ESP draw?

Flo082002, 2019-04-05:
>I am now trying to upload code to the ESP, by having it connected to the PC via a USB Cable and keeping boot button pressed, but nothing happens.

Mesure with you multimeter, weather the button works.

GeorgeFlorian, 2019-04-05:
>>> Pin 1 (green) = GND
>>> Pin 2 (yellow) = 3. 3V
>>> Pin 3 (blue) = USB voltage (4. 5V - 5. 5V)
>>> Measure from pin 1 to pin 3. There should be 3. 3V.
>
> > From pin 1 to 2 there are 3.25V and from pin 1 to 3 there are 4.35V.
> 
> That's OK.
> How much current does the ESP draw from USB or the Vin Pin?

But you said that it should be 3V3 from 1 to 3...

Flo082002, 2019-04-05:
>But you said that it should be 3V3 from 1 to 3...

Sorry. That's my mistake. I mean 1 and 2

GeorgeFlorian, 2019-04-05:
> > I am now trying to upload code to the ESP, by having it connected to the PC via a USB Cable and keeping boot button pressed, but nothing happens.
> 
> Mesure with you multimeter, weather the button works.

The measurements for the buttons are : 
-unpressed: 1.25V
-pressed: 0V

GeorgeFlorian, 2019-04-05:
> > From pin 1 to 2 there are 3.25V and from pin 1 to 3 there are 4.35V.
> 
> That's OK.
> How much current does the ESP draw from USB or the Vin Pin?

How do I measure that ?

Flo082002, 2019-04-05:
>The ESP ran around 1h with that error, which I think was actually the ESP resetting every second, made it go so hot that I couldn't touch it.

Are
GPIO 6 
GPIO 7
GPIO 8
GPIO 9 
GPIO 10
GPIO 11
Connect to anything?
They are connected internally to the flash of the ESP. If you connect these pins to anything, the esp is in a reboot loop.

Flo082002, 2019-04-05:
>How do I measure that ?

Use a measuring range around 2A

https://encrypted-tbn0.gstatic.com/images?q=tbn:ANd9GcTjs6kH7rFKz8B4_t0JsWTgLYvrkyhWldlS479ahgME_udyUjqCUw

Flo082002, 2019-04-05:
>pressed: 0V

That's OK. 

>unpressed: 1.25V

I will mesure on my esp when I am at home.

GeorgeFlorian, 2019-04-05:
> > pressed: 0V
> 
> That's OK.
> 
> > unpressed: 1.25V
> 
> I will mesure on my esp when I am at home.

On my ESP32 Wrover-B its 3.3V when not pressed and 0V when pressed.

GeorgeFlorian, 2019-04-05:
> > How do I measure that ?
> 
> Use a measuring range around 2A
> 
> https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.electronics-notes.com/articles/test-methods/meters/how-to-measure-current.php&ved=2ahUKEwjZ2JS01bjhAhXPKlAKHaoWDDgQFjAFegQIaBAb&usg=AOvVaw1IxVuy6Z3Lig5is4n8oEnh
> 
> The measured current should not be more than 300mA

I am sorry but I won't be able to measure that.

> Are
> GPIO 6
> GPIO 7
> GPIO 8
> GPIO 9
> GPIO 10
> GPIO 11
> Connect to anything?
> They are connected internally to the flash of the ESP. If you connect these pins to anything, the esp is in a reboot loop.

I've got the 30 pins variant of the ESP Wroom-32 DevKitV1 so I don't even have pins 6,7,8,9,10,11.
Also, I've attached a schematic in the top post.

Flo082002, 2019-04-05:
>unpressed: 1.25V

I found out in the internet, that this is normal for your board. 

>I am sorry but I won't be able to measure that.

Is your multimeter not able to mesure current?

GeorgeFlorian, 2019-04-05:
> > unpressed: 1.25V
> 
> I found out in the internet, that this is normal for your board.
> 
> > I am sorry but I won't be able to measure that.
> 
> Is your multimeter not able to mesure current?

On the Wroom-32 DevKitV1:

- BOOT Button not pressed: 1.24V. Pressed: 0V
- EN Button not pressed: 3.3V. Pressed: 0V;

On the Wrover-B DevKitV4:

- BOOT Button not pressed: 3.3V . Pressed: 0V
- EN Button not pressed: 3.3V. Pressed: 0V;

Why is there a difference since the boards are almost the same ?

I've measured the current at VIN and it's 370 mA.

Flo082002, 2019-04-05:
> I found out in the internet, that this is normal for your board.

Don't truste the INternet. 
When I searched further on the internet, I found this problem several times and several people said that this is not correct with the 1. 5 V.
I've only seen one Peron so far who thinks that's right.
Now I don't know what's right.
I'm on 3. 3V.
I also have an ESP-WROOM-32.

Flo082002, 2019-04-05:
```
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> ets_main.c 371 
> ets Jun  8 2016 00:22:57
```
This looks like a memory error.
Try to erease the flash with the esptool.

GeorgeFlorian, 2019-04-08:
> ```
>  rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
>  flash read err, 1000
>  ets_main.c 371 
>  ets Jun  8 2016 00:22:57
> ```
> This looks like a memory error.
> I get the same error, when I haver no programm in the esp.
> Try to erease the flash with the esptool.
> 
> If you have istalled the arduino-esp32 with de boardmanager:
> open cmd and type:
> ` C:\Program Files (x86)\Arduino\hardware\espressif\esp32\tools\esptool\esptool.exe erase_flash`
> 
> Then try to Upload a sketch.

I am using Linux Mint 19.1.

I've tried erasing the flash using the esptool.py, but it couldn't connect to the ESP.

This is the command: esptool.py erase_flash

And this is the output:
```
esptool.py v2.6
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyUSB0 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header
Serial port /dev/ttyS0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyS0 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header

A fatal error occurred: All of the 2 available serial ports could not connect to a Espressif device.
```

Flo082002, 2019-04-08:
Is the Voltage over the boot button 3,3V
when you use a 1kΩ - 50kΩ resistor from 3.3V pin to one side of the button?
If yes, try to upload a program with the resistor. When the computer wants to connect to the ESP, keep the resistor on the button, press and hold the boot button.

Paraphraser, 2019-04-27:
GeorgeFlorian - FYI none of the postimage.cc links in your original post seem to work. I get a 404 error page on postimage. 
[image]

Flo082002, 2019-04-27:
> GeorgeFlorian - FYI none of the postimage.cc links in your original post seem to work. I get a 404 error page on postimage.

I have no 404 on my PC.

Paraphraser, 2019-04-27:
Weird. The earlier post was from my Safari on an iPad. I've just checked again, this time from an iMac and with both Safari and FireFox. I get 404 errors on both browsers.

I wondered if it was a regional thing (I'm in Australia) but firing up a VPN and pretending to be in San Francisco - still 404 errors.

The only other thing I can think of is that I'm not registered on postimage but I have not tried registering to see if that makes a difference.

[image]

Flo082002, 2019-04-27:
I am not registered too

Flo082002, 2019-04-27:
I am not registered too.

I have downloaded the pictures.
Here are the pictures :

Flo082002, 2019-04-27:
[image]
[image]
[image]
[image]

Paraphraser, 2019-04-27:
Well, that's obviously not the explanation, then.

The only reason I mentioned it was because I assumed that, if I couldn't see them, nobody else could either. If nobody else Is having 404 issues then I don't think we should waste time on figuring out why I'm having some odd-ball problem. I was just curious about this thread because I've very occasionally had ESP32 boards misbehave and get hot. It's one of those mysteries that I hope to solve sometime.

While I was typing that, you sent the pictures. Thanks *heaps* for that. I'll shut up now and stop wasting everyone else's time.

closed 2019-08-01 by GeorgeFlorian

Cemu0, 2020-05-16:
Hi !
on my ESP32S board (esp-wroom-32) after connected some wire while it still online, my board stop functioning and I got the same error as your 
```
[06:45:44]rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
[06:45:44]flash read err, 1000
[06:45:44]ets_main.c 371 
[06:45:44]ets Jun  8 2016 00:22:57
```
The board also get really hot and not able to flash the esp again event using Flash Download Tools from Espressif or PlatformIO.
Followed this post [ESP32 Boot Mode Selection](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection), I tried to keep the GPIO5 as HIGH or LOW but it wasn't solved the problem. 
I guess the board was broken while wiring and ... yup going to buy a new one.
Hope this helpful.

GuillaumeLeclerc, 2022-06-12:
It happened on two of my boards. Any idea on how to avoid this happening in the Future ?

malig06, 2022-07-25:
> C:\Program Files (x86)\Arduino\hardware\espressif\esp32\tools\esptool\esptool.exe erase_flash

Sir did you fixed the problem or you just passed it .
Today, When I was working with ESP32 I recognized that the ESP32 has heat problem from the shield when I touched it I felt it was too hot. Since I had the problem I tried to find out but I can't do anything did you fixed the problem? 
I need some help. A couple of days later I have compititon with esp32s' Please Help.

GeorgeFlorian, 2022-07-27:
> Sir did you fixed the problem or you just passed it . Today, When I was working with ESP32 I recognized that the ESP32 has heat problem from the shield when I touched it I felt it was too hot. Since [...]

It was just a broken board.

GeorgeFlorian, 2022-07-27:
> Weird. The earlier post was from my Safari on an iPad. I've just checked again, this time from an iMac and with both Safari and FireFox. I get 404 errors on both browsers.
>
> I wondered if it was a [...]

The image hosting probably expired. I can't see the image either.
The problem was a broken board.

SharlSherif, 2022-08-13:
> Sir did you fixed the problem or you just passed it . Today, When I was working with ESP32 I recognized that the ESP32 has heat problem from the shield when I touched it I felt it was too hot. Since [...]

Did you find a solution to this or was it just a broken board as well?

malig06, 2022-08-16:
@SharlSherif  it was just a broken board but after some searching. I'll try to fix it. The solution which I'm thinking is using the cold sprey find the broken resistor or what ever it is.

Ganesh540-crypto, 2024-06-19:
> > > pressed: 0V
> > 
> > 
> > That's OK.
> > > unpressed: 1.25V
> > 
> > 
> > I will mesure on my esp when I am at home.
> 
> On my ESP32 Wrover-B its 3.3V when not pressed and 0V when pressed.
> 
> > > How do I measure that ?
> > 
> > 
> > Use a measuring range around 2A
> > https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.electronics-notes.com/articles/test-methods/meters/how-to-measure-current.php&ved=2ahUKEwjZ2JS01bjhAhXPKlAKHaoWDDgQFjAFegQIaBAb&usg=AOvVaw1IxVuy6Z3Lig5is4n8oEnh
> > The measured current should not be more than 300mA
> 
> I am sorry but I won't be able to measure that.
> 
> > > The ESP ran around 1h with that error, which I think was actually the ESP resetting every second, made it go so hot that I couldn't touch it.
> > 
> > 
> > Are
> > GPIO 6
> > GPIO 7
> > GPIO 8
> > GPIO 9
> > GPIO 10
> > GPIO 11
> > Connect to anything?
> > They are connected internally to the flash of the ESP. If you connect these pins to anything, the esp is in a reboot loop.
> 
> I've got the 30 pins variant of the ESP Wroom-32 DevKitV1 so I don't even have pins 6,7,8,9,10,11. Also, I've attached a schematic in the top post.

bro can any one help me I am also facing the same issue.

jefflord, 2024-08-19:
For posterity, I'll add to this.

I encountered the same issue: my ESP32 worked fine initially, but then something unknown happened. (I was experimenting with code to discover the PWM pins, using a rheostat, a servo, USB power, and an external 5V supply.)

- At some point, the ESP32 started running very warm—not extremely hot, but noticeably warm. The current draw was 400mA, which is very high for the task it was performing. Another ESP32 running the same code (and an OLED) only draws 80mA.

- If I hold the BOOT button while powering up, I get this message:
`rst:0x1 (POWERON_RESET), boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))`, and I'm able to upload firmware. Afterward, I need to power cycle the device to get the firmware to boot.

- Upon rebooting, without holding the BOOT button, the firmware loads, and the reset message shows:
`rst:0x1 (POWERON_RESET), boot:0x13 (SPI_FAST_FLASH_BOOT)`.

- So, it "works," but not as it should. I suspect, as mentioned above, that something is damaged. I'll order an identical part and compare them to see if I can identify the fault, though I'm not optimistic about finding it.

FYI:

- I also tried `erase_flash`, but it made no difference.
- I compared all the GPIOs with another ESP32 (after each was running the firmware), and they were pretty much the same.