espressif/arduino-esp32 #2632: Calling configTime without WiFi breaks.

a-c-sreedhar-reddy, 2019-04-02:
### Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200
Computer OS:                                        Ubuntu

### Description:
When configTime is called without WiFi the code breaks.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```
#include "time.h"
String getLocalTime()
{
  String currentTime = "NULL";
  struct tm timeStruct;
  if (getLocalTime(&timeStruct))
  {
    currentTime = asctime(&timeStruct);
    currentTime.replace("\r", "");
    currentTime.replace("\n", "");
  }
  return currentTime;
}
void setup(){
    Serial.begin(115200);
    const char *ntpServer = "pool.ntp.org";
    const long gmtOffset_sec = 18000 - 1800;
    const int daylightOffset_sec = 3600;
    configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
    
}
void loop(){
    Serial.println(getLocalTime());
}
```

### Debug Messages:
```
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40087879 on core 1

```

### ESP exception decoder
```

Decoding stack results
0x400850c8: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x400852f5: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x40087011: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c line 1441
0x400f1f52: sys_mutex_lock at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c line 78
0x400f1fd2: sys_arch_protect at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c line 469
0x400e6d74: do_memp_malloc_pool at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c line 302
0x400e6ddd: memp_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c line 398
0x400ecf99: udp_new at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c line 1145
0x400ecfbe: udp_new_ip_type at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c line 1177
0x400e5923: sntp_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/apps/sntp/sntp.c line 547
0x400d121b: configTime at /home/acreddy/Desktop/arduino-1.8.5/hardware/espressif/esp32/cores/esp32/esp32-hal-time.c line 55
0x400d1196: setup() at /home/acreddy/Arduino/testtime/testtime.ino line 18
0x400d1b2b: loopTask(void*) at /home/acreddy/Desktop/arduino-1.8.5/hardware/espressif/esp32/cores/esp32/main.cpp line 20
0x400891a9: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```

stickbreaker, 2019-04-02:
How do you expect to access the internet without a connection?

a-c-sreedhar-reddy, 2019-04-02:
@stickbreaker If the connection is not there it should not break right

lbernstone, 2019-04-03:
The driver needs to at least be initialized to get the needed structures.

gitdevus, 2019-04-04:
@a-c-sreedhar-reddy same issue here. This code works with ESP8266 but not with ESP32. It shouldn't crash.

gitdevus, 2019-04-04:
@a-c-sreedhar-reddy same issue here. This code works with ESP8266 but not with ESP32. Calling configTime once the WiFi connection is established works sometimes but not always apparently. It seems that the behavior is not as expected and configTime shouldn't fail that way too.

ThaChoppa, 2019-04-30:
Same here as well. If you explicitly set WiFi.mode(WIFI_STA); right before calling configureTime() it works for me. As far as i can see an actual wifi-connection is not required (as long as you don't whant to get the time from a NTP-Server at this stage), but for what ever reason the wifi module must be at least active.

edalongeville, 2019-08-01:
Same issue here, attempting to use NTPClient.h before configuring the WiFi causes the ESP32 to reboot.

lbernstone, 2019-08-01:
@ThaChoppa 's comment is the solution.  You must initialize the driver before calling any socket functions.

a-c-sreedhar-reddy, 2019-08-01:
Yes but can something be done so that esp does not reboot.

edalongeville, 2019-08-01:
@lbernstone It's a workaround but not a solution, especially when @gitdevus have mentioned the problem is not present on ESP8266.

atanisoft, 2019-08-01:
the only difference between the esp32 and esp8266 configTime methods is you can pass a null for the time servers and it will not enable SNTP. So the usecase from the OP is not entirely valid without an internet connection as it specifies a timeserver which should be polled for time data.

esp32: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-time.c#L51-L54
esp8266: https://github.com/esp8266/Arduino/blob/master/cores/esp8266/time.cpp#L68-L73

atanisoft, 2019-08-01:
From the above links, it would appear that possibly just no-op the calls when the servers are null and that should fix this issue.

closed 2019-10-14 by stale[bot]

lbernstone, 2019-11-10:
Fixed in #3470