espressif/arduino-esp32 #2633: Can't upload sketch to ESP32-EVB

GeorgeFlorian, 2019-04-02:
OS: Linux Mint 19.1
Board: ESP32-EVB

I am trying to upload a simple sketch (like anything) to an ESP32-EVB.
I've been able to upload just fine to the other ESP32, and now I can't upload to it either.
More of this here: https://github.com/espressif/arduino-esp32/issues/2630

Now I can't even upload to this new, out of the box ESP32-EVB.
The board is connected to the PC via a USB Cable.

What happened. Are they both broken ? 

How can I not be able to upload to any ESP32, if, until yesterday, I was able to ?!

This is the error I'm receiving:
`A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header`

I am keeping the BUT1 button pressed.

Jeroen88, 2019-04-02:
- Is another process (another Arduino IDE) active occupying the USB serial?
- Aren't Rx and Tx of UART0 connected to something?
- Is the USB cable functioning?
- Is the USB Port functioning?
- Did you select the correct board in the Arduino IDE?

GeorgeFlorian, 2019-04-02:
> * Is another process (another Arduino IDE) active occupying the USB serial?
> * Aren't Rx and Tx of UART0 connected to something?
> * Is the USB cable functioning?
> * Is the USB Port functioning?
> * Did you select the correct board in the Arduino IDE?

I've just opened the PC and the Arduino, there is no other process occupying the USB Serial.
As I've mentioned in my original post, the ESP is only connected to the PC via a USB Cable.
The USB Cable is functional. I've tried with others just in case.
I've tried all the USB ports available on the motherboard and nothing changed.
Here are the Arduino IDE settings:
https://i.postimg.cc/J4DR3jrC/Screenshot-at-2019-04-02-14-14-31.png

chegewara, 2019-04-02:
> I am keeping the BUT1 button pressed.

Why?

GeorgeFlorian, 2019-04-02:
> > I am keeping the BUT1 button pressed.
> 
> Why?

Why ? Because that's the boot button. You are supposed to press it !

That's how I've always uploaded sketches to the ESP32. Before the ESP32-EVB, I've used a ESP32- Wroom-32 and this one uploaded ONLY while holding down the button.

chegewara, 2019-04-02:
But you dont have to do it anymore with new board, and for sure you cant keep it pressed.

GeorgeFlorian, 2019-04-02:
> But you dont have to do it anymore with new board, and for sure you cant keep it pressed.

This is beside the point ! It should be obvious that I've tried uploading with and without pressing the button.
It doesn't work either way.

Why can't I upload code to the ESP32-EVB ?!

GeorgeFlorian, 2019-04-08:
I've stopped using that board so I will close this !

I am disappointed in the lack of help I got !

closed 2019-04-08 by GeorgeFlorian