espressif/arduino-esp32 #2635: Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

robertgregor, 2019-04-02:
Hi, my sketch after some time - around 5 - 10 minutes, crash with the error:

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4003c771  PS      : 0x00060734  A0      : 0x800551d8  A1      : 0x3ffd9ec0  
A2      : 0x3ff71000  A3      : 0x80007007  A4      : 0x3ffb9510  A5      : 0x80000000  
A6      : 0x00060720  A7      : 0x00000000  A8      : 0x8003c724  A9      : 0x3ffd9eb0  
A10     : 0x08000b00  A11     : 0x00000000  A12     : 0x000001b0  A13     : 0x3ffb96c0  
A14     : 0x00000000  A15     : 0x0000001b  SAR     : 0x0000001f  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4003c771:0x3ffd9ec0 0x400551d5:0x3ffd9ef0 0x40167c16:0x3ffd9f10 0x4009498d:0x3ffd9f40

Core 1 register dump:
PC      : 0x40083abe  PS      : 0x00060134  A0      : 0x800837a6  A1      : 0x3ffca2c0  
A2      : 0x20b9106b  A3      : 0x00000000  A4      : 0x5c000000  A5      : 0x00000020  
A6      : 0x0000001e  A7      : 0x3ffc8400  A8      : 0x00000000  A9      : 0x3ff47030  
A10     : 0x77ffffd8  A11     : 0x00000001  A12     : 0x00000050  A13     : 0x1f83c290  
A14     : 0x77ffffd8  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400012e5  LEND    : 0x40001309  LCOUNT  : 0x800ee46b  

Backtrace: 0x40083abe:0x3ffca2c0 0x400837a3:0x3ffca2e0 0x4008167f:0x3ffca300 0x400def07:0x3ffca320 0x400ef71d:0x3ffca350 0x4009498d:0x3ffca370

After decode:

C: 0x4003c771
EXCVADDR: 0x00000000

Decoding stack results
0x4009498d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

I am using 1.0.1 released version. 
What is the reason for that error?

My sketch is runing webserver and the webpage periodically requesting the server for the json.

robertgregor, 2019-04-02:
Another case when it crashed:

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4013cd23  PS      : 0x00060130  A0      : 0x800e0b08  A1      : 0x3ffcd960  
A2      : 0x00000000  A3      : 0x3ffcd9bc  A4      : 0x00000020  A5      : 0x0000a688  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80146c91  A9      : 0x3ffcd940  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffccdd4  A13     : 0x00000000  
A14     : 0x3ffcce50  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000034  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4013cd23:0x3ffcd960 0x400e0b05:0x3ffcd990 0x40135310:0x3ffcd9b0 0x4009498d:0x3ffcd9e0

After decode:

PC: 0x4013cd23: tcp_output at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp_out.c line 1004
EXCVADDR: 0x00000034

Decoding stack results
0x4013cd23: tcp_output at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp_out.c line 1004
0x400e0b05: _tcp_output_api(tcpip_api_call_data*) (C:\Program Files at x86)\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp line 284
0x40135310: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 124
0x4009498d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

GeorgeFlorian, 2019-04-08:
Provide the sketch or one that can emulate that error.

closed 2019-08-15 by stale[bot]