espressif/arduino-esp32 #2637: Starting work with Wizet W5500(questions on compatibility with ESP32)

telliottosceola, 2019-04-02:
What you are trying to do.
Add compatibility for Wizet W5500 

Describe your system( Hardware, computer, O/S, core version, environment)
ESP32 Wroom32, developing on PlatformIO

Describe what is failing
No failures, just looking for insight.

I am in the preliminary phase of adding Ethernet to a project we currently have that is based on the ESP32.  Most tasks are MQTT, MQTTS, and HTTP/HTTPS.  I found [Arduino's Ethernet Library](https://github.com/arduino-libraries/Ethernet) which has support for the Wizen W550 chip I'd like to use.  However it appears that Arduino-esp32 has opted not to use this library.  Is there a particular reason for this?  I'd like to make sure there is not already support for the W5500 before I go trying to port everything.  Essentially this device will have a setting to switch all communications between Ethernet and WiFi so I'll pretty much need full functionality over Ethernet.

Also @copercini I'd love to have your input on porting WiFiSecureClient over to Ethernet for this.  I took a quick glance at the source for WiFiClientSecure but have not dug in too deep.  Do you foresee any glaring issues with doing this?

Thank you everyone

atanisoft, 2019-04-02:
https://github.com/PaulStoffregen/Ethernet

pipi61, 2019-04-02:
Try lan8720  from library. lowest price.

telliottosceola, 2019-04-02:
Thanks guys.

@atanisoft I saw that library.  Looks like a simple port to ESP32.  Looks like it pretty much has the exact same functionality as the Arduino Ethernet Library.  Any reason you'd see using this one from Paul rather than Arduino's Ethernet library?

@pipi61 It would certainly be much simpler to use a PHY converter already supported in the official Arduino-esp32. I'm just trying to work with the EEs brought me at the moment with is a Wiznet W5500.  However if it gets to be too big of a pain I'll definitely suggest we move over to that chip instead.

Do either of you have any insights into secure client sockets over Ethernet because I'm not seeing anything at all so far.  I with WiFiClientSecure had a constructor where you passed in a client object, that would be pretty sweet since you could just pass in an Ethernet client that way.

atanisoft, 2019-04-02:
@telliottosceola because the work to port it has been done. There is no reason to duplicate the effort.

I have no experience on the secure usages, in all of my usages it has been extremely basic usages.

telliottosceola, 2019-04-03:
Thanks @atanisoft I'll be sure to post if I get it going.  I have ported the WiFiClientSecure library to use the Ethernet client and have the Ethernet client working on the ESP32.  I used the Arduino Ethernet client which worked right out of the box.  It's nearly identical to the one you linked to.  The EthernetSecureClient library is compiling.  Time to start flashing, testing and debugging.  I'll post updates here for anyone interested.

telliottosceola, 2019-04-03:
So debugging this port of the WiFiClientSecure I have a test repo for it here:
https://github.com/telliottosceola/Ethernet_Test
Sorry structure is a little "wonky" but I just threw it together for testing.  It's setup for PlatformIO.  Ethernet.init is using GPIO 4 as the SS line since that's what the EE's brought me.

Anyway running main.cpp fails with the following:
`My IP address: 192.168.1.22

Starting connection to server...
[E][EthernetClientSecure.cpp:113] connect(): about to call start_ssl_client

[V][ssl_client.cpp:53] start_ssl_client(): Free internal heap before TLS 370640
[V][ssl_client.cpp:55] start_ssl_client(): Starting socket
[V][ssl_client.cpp:57] start_ssl_client(): 1
2, 1, 6
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40087d75 on core 1

Backtrace: 0x400856b4:0x3ffb1b30 0x400858e5:0x3ffb1b50 0x40087d75:0x3ffb1b70 0x401171f2:0x3ffb1bb0 0x4011739a:0x3ffb1bd0 0x4010a560:0x3ffb1bf0 0x4010a5c9:0x3ffb1c10 0x401182be:0x3ffb1c30 0
x40117464:0x3ffb1c50 0x40108ee0:0x3ffb1c90 0x400d3bf7:0x3ffb1cb0 0x400d2bc4:0x3ffb1f20 0x400d2cc9:0x3ffb1f60 0x400d313e:0x3ffb1f80 0x400d6461:0x3ffb1fb0 0x40087175:0x3ffb1fd0`

I've tracked this down to line 61 in ssl_client.cpp

Any insight would be greatly appreciated.  Thanks All!

telliottosceola, 2019-04-08:
(Update) See latest on repo here:
https://github.com/telliottosceola/Ethernet_Test
Failing on ssl_client.cpp line 210 with MbedTLS message code: -69(0x0045) which appears to be:
`MBEDTLS_ERR_NET_INVALID_CONTEXT `
So it's saying that &ssl_client->ssl_ctx has been freed.  Any ideas?

gjanpjetro, 2019-06-18:
Dear @telliottosceola, I'm thinking to use the ESP32 as a gateway, in https,  and manage also the ethernet. One possibility is to use the w5500 device. Have you got any results in your project?

telliottosceola, 2019-06-18:
Hi @gjanpjetro,

Unfortunately I do this stuff for a living and the plug was pulled on this project.  I determined that there was not going to be a straight forward way to route TLS connections through the W5500 interface in a reasonable amount of time currently.  It is absolutely possible but just not cost effective from a business standpoint.  You would need to register callbacks for MBEDTLS and route data through the W5500.  Hopefully one day I'll have time to revisit this project.

closed 2019-08-31 by stale[bot]

Voha888, 2019-11-22:
> (Update) See latest on repo here:
> https://github.com/telliottosceola/Ethernet_Test

Hello! Is your project working? How can I contact you? Or send me an email: 888voha@gmail.com
Thanks!

ntwallace, 2020-04-28:
> Thanks @atanisoft I'll be sure to post if I get it going. I have ported the WiFiClientSecure library to use the Ethernet client and have the Ethernet client working on the ESP32. I used the Arduino [...]

@telliottosceola, do you mind adding the ported WiFiClientSecure lib to your Ethernet_Test repo? I'm about to attempt something similar and would love to save the trouble of redoing that work. Thanks!

telliottosceola, 2020-04-28:
@ntwallace,
WiFiClientSecure is an ESP Arduino root library.  You can find it here:
https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFiClientSecure
I don't remember making any changes to the stock library, this was the library I was trying to replace.

ntwallace, 2020-04-28:
@telliottosceola Gotcha, I was misremembering which library was which. I was thinking of WiFiGeneric, which looks like it could potentially be modified to handle W5500 ethernet events directly. Thanks.