espressif/arduino-esp32 #2639: ESP32 AP WIFI loss all packets in WIFI_AP_STA mode after attempt station reconnection

danlu01, 2019-04-04:
Using ESP32 in WIFI_AP_STA mode, computer can connect to ESP32's AP WIFI and ping results ok.  However, as soon as trying to connect to Station WiFi twice, ESP32's AP WiFi loss all ping packets.  PC and phone can still connect to AP WiFi but can not communicate with it.

Here's the scenario:  ESP32 connected to STATION WiFi and ESP32's AP WiFi as backup.  If Station WiFi router goes down temporary, ESP32 should auto reconnect to Station WiFi when Station WiFi is back up.  In the meantime, ESP32 AP WiFi acts as backup.  However, when attempting to reconnect to Station WiFi, ESP32's AP WiFi will stop working (can still connect to it but not communicate to it).

Arduino Code:
`#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>
const char *ssid = "yourAP";
const char *password = "yourPassword";

void setup() {

  WiFi.mode(WIFI_AP_STA);
  
  

  Serial.begin(115200);
  Serial.println();
  Serial.println("Configuring access point...");

  // You can remove the password parameter if you want the AP to be open.
  WiFi.softAP(ssid, password);
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  WiFi.begin("testingSSID", "ksiii3j3jjd");
  delay(10000);
  WiFi.disconnect();

  WiFi.begin("testingSSID", "ksiii3j3jjd");
  delay(10000);
  WiFi.disconnect();
  
  WiFi.begin("testingSSID", "ksiii3j3jjd");
  delay(10000);
  WiFi.disconnect();
  
  WiFi.begin("testingSSID", "ksiii3j3jjd");
  delay(10000);
  WiFi.disconnect();
  
  WiFi.begin("testingSSID", "ksiii3j3jjd");
  delay(10000);
  WiFi.disconnect();
  
  Serial.println("test now");
}

void loop() {
}
`

### Hardware:
Board:							Generic ESP32 Dev Module
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					921600
Computer OS:                                          Windows 10

### Debug Messages:
`Configuring access point...
AP IP address: 192.168.4.1
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
test now
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 16 - AP_STADISCONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 15 - AP_STACONNECTED
`

boarchuz, 2019-04-07:
This is not unusual. AP performance is shaky whenever STA is disconnected as it will be actively scanning channels trying to reconnect, causing issues for clients connected to the AP on a particular channel.
The trick is to disable STA after a connection attempt fails (you can pass a parameter to disconnect(true) to do this, or enableSTA(false)) so the AP can stabilise, and try again after some interval. So your test code should be more stable if it looks something like this: 
 ```
  WiFi.begin("testingSSID", "ksiii3j3jjd"); //Enable STA
  delay(10000);
  WiFi.disconnect(true); //Disable STA
  delay(5000); //AP only for 5 seconds
  WiFi.begin("testingSSID", "ksiii3j3jjd"); //Enable STA
  delay(10000);
  WiFi.disconnect(true); //Disable STA
  delay(5000); //AP only for 5 seconds
```
If you still have issues, try lowering 10000.

danlu01, 2019-04-07:
Thank you, that worked.

closed 2019-04-07 by danlu01