espressif/arduino-esp32 #2644: Using a partition as a manager for OTA rollback?

LucasFeliciano21, 2019-04-07:
Hi, I'm trying to build a fail control partition for rolling back to a previous storages image, so if something goes wrong with the new OTA image I can point to other partition and do the report of bad image. 

It's possible to point to other partition without resetting the esp? An if that is possible, can I use a standard image on one of the partition as this "docker manager"?

I saw the esp_ota_set_boot_partition(), but that requires a reset, so if on the running code, something went wrong, even if I set some non volatile memory as a flag, I would not have something to read it and to point to other "safe image"

Thanks in advance and congrats on the good work.

chegewara, 2019-04-07:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/ota.html#app-rollback

LucasFeliciano21, 2019-04-07:
@chegewara, thanks, I'll see if the CONFIG_APP_ROLLBACK_ENABLE is by default on the Kconfig. Also the logic for calling esp_ota_mark_app_valid_cancel_rollback().

Thanks man. I'll mark this as closed as I look into this.

closed 2019-04-07 by LucasFeliciano21