espressif/arduino-esp32 #2647: i2s dac sounds strange.

Flo082002, 2019-04-09:
I used the bluetooth a2dp with the internal DACs.
https://github.com/espressif/arduino-esp32/files/3043296/a2dp_sink.zip
I use esp32-arduino 1.0.1 from the board manager.
I noticed that you only hear something when you use `I2S_MODE_PDM` anstadt from `I2S_MODE_ADC_BUILT_IN`. 
Now it plays with the built-in DACs. But the audio signal pretty noisy when I play something.
Besides, the sounds are weird.
When I play musik, the frequency sounds deeper.
When play 1000Hz sine and my phone sais, that it is 914Hz.
When I play a sine wave, I can see on an oscilloscope that the lower half of the sine is compressed and the upper one is pulled wide.

Besides, I hear lots of crackling noises. 
When I play the musik very quiet, there is lots of nois.

1000 hz sine at 100% Volume : [100%.wav.zip](https://github.com/espressif/arduino-esp32/files/3060216/100.wav.zip)

1000 hz sine at 25% Volume : [25%.wav.zip](https://github.com/espressif/arduino-esp32/files/3060215/25.wav.zip)

Sorry for the ZIP file format. I can't upload wav files here.

I have also tried this with older versions of esp32-arduino. That didn't work too. 

I hope, someone can help me.

me-no-dev, 2019-04-12:
hey :) so I will throw a couple of guesses here.
1. You should be using the DAC mode really, strange you selected PDM and you hear the DACs
2. the actual reason why you see those wave issues is probably the fact that you are writing a signed data to unsigned register. Audio data is AC or oscillates around some value which we take as 0. For full range on 8bit DAC that will equal to 128. so you need to turn the data from -128 to 127 instead to be from 0 to 255. This should do the trick :)

Flo082002, 2019-04-22:
Thanks for your help.
But where do I have to add 

> ```c++
> left = left + 0x8000;
> right = right + 0x8000;
> ```

to my code?

me-no-dev, 2019-04-23:
you need to do this for every sample of your WAV data before you push it to the DACs

Flo082002, 2019-04-23:
> you need to do this for every sample of > > your WAV data before you push it to the
> DACs

I know that. But, where do I have to modify my arduino code to achieve that ?
I have no idea.

me-no-dev, 2019-04-23:
you read the file and then you push it to I2S, so at that point :)

Flo082002, 2019-04-23:
I have no file. I play the audio with Bluetooth.
https://github.com/espressif/arduino-esp32/files/3043296/a2dp_sink.zip

me-no-dev, 2019-04-23:
please use some common sense and go through the code you are using. find the function with name: bt_app_a2d_data_cb and do it there before you write to i2s.
If you continue to have issues with your own code, feel free to post in IDFs repo issues or visit our gitter chat (link in readme)

closed 2019-04-23 by me-no-dev