espressif/arduino-esp32 #2650: ESP32 wroom32: BluetoothSerial.write doesn't work anymore at client reconnection

Alteregoxxx, 2019-04-10:
### Hardware:
Board:							ESP32 Dev Module rev.C
Core Installation/update date:		1.7.0 (PlatformIO reported)/3.1.3(esp-idf framework)
IDE name:					        Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                      No
Upload Speed:				        115200
Computer OS:                                          Windows 10

### Description:
Hello everyone,
As for the subject, using the BluetoothSerial, if it's sending fast lots of a data using BluetoothSerial.write() method and the client to which it was connected disconnects, at the client reconnection, ESP32 Wroom32 accept the connection (I'm sure because I can see it from the serial prints I perform in the callback called at client connection/disconnection) , but the BluetoothSerial.write() method doesn't send data anymore.

A snippet worths a thousand words:

### Sketch:  

```cpp
#include <Arduino.h>
#include <BluetoothSerial.h>
BluetoothSerial MyBT;
bool connesso = 0;
/* I use some other variables here
...
...
*/
 
void callback(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)
{
  if (event == ESP_SPP_SRV_OPEN_EVT)
  {
    Serial.println("Client Connected");
    connesso = true;
  }
 
  if (event == ESP_SPP_CLOSE_EVT)
  {
    Serial.println("Client disconnected");
    connesso = false;
  }
}
 
/* I implement some other functions here
.....
.....
*/
 
void setup()
{
 
  Serial.begin(9600);
  MyBT.register_callback(callback);
 
  MyBT.begin("ESP32-BT");
  while (!MyBT.hasClient())
  {
    delay(1000);
    Serial.println("ESP32 is ready to pair.");
  }
  Serial.println("ESP32 PAIRED!!");
  delay(1000);
}
 
void loop()
{
 
/* I initialize here all the variables that I'll send using BluetoothSerial.write()
.....
.....
*/
 
  if (connesso)
  {
 
/*I perform some other tasks on the aforementioned variables here
.......
.......
*/
 
    MyBT.write(AddParityAndPosition(&MyUint8_t_here, true));
    MyBT.write(AddParityAndPosition(&AnotherUint8_t_here, false));
    /*If I insert some delay in the order of hundreds of milliseconds, as in the subsequent line of code, the issue disappear*/
    //delay(100);  
 
  }
 
  else{
    Serial.println("Waiting for a client connection");
    delay(500);
  }
 
}

```

Is this a normal behaviour? Can someone be so kind in trying to reproduce the issue?

Any help/explanation on the issue is very welcome! :-)

Thanks!

bestbai, 2019-07-13:
I had the same problem. i'll try it your way.

RealParity, 2019-08-19:
I examined the problem for quite some time, the main loop will stop running if you connect/disconnect to the bluetooth serial port several times. Verifyable by blinking a LED in the main loop that will stop when the problem occures. Verified with both, laptop computers and Smartphones and different ESP-32 modules.

Seems to be a elementary problem that can be reproduced with this very simple sketch:
`
#include "BluetoothSerial.h"

BluetoothSerial Blue;

void setup() {
  Blue.begin("Smart Rf 1 (SI-I)");
}

void loop() {
        Blue.write('X');
        delay(20);
}
`
With the delay(20) like I am using above, it usually takes only a few tries (2-20) to stop the loop
The longer you make the delay, the less likely it seems to be that the problem occurs, but it still happens.

RealParity, 2019-08-19:
I examined the problem for quite some time, the main loop will stop running if you connect/disconnect to the bluetooth serial port several times. Verifyable by blinking a LED in the main loop that will stop when the problem occures. Verified with both, laptop computers and Smartphones and different ESP-32 modules.

Seems to be a elementary problem that can be reproduced with this very simple sketch:
```
#include "BluetoothSerial.h"

BluetoothSerial Blue;

void setup() {
  Blue.begin("BTname");
}

void loop() {
        Blue.write('X');
        delay(20);
}
```
With the delay(20) like I am using above, it usually takes only a few tries (2-20) to stop the loop
The longer you make the delay, the less likely it seems to be that the problem occurs, but it still happens.

closed 2019-11-01 by stale[bot]