espressif/arduino-esp32 #2651: Error Installing the ESP32 in Arduino IDE on a Raspberry Pi 3

RamanujamV, 2019-04-10:
Need help resolving thie following:
Error Installing the ESP32 in Arduino IDE on a Raspberry Pi 3 

Pi3 running NOOBS v3_0_1 (downloaded on Aril 10th)

IDE installed using "arduino-nightly-linuxarm.tar.xz " (downloaded on Aril 10th)

In the Installation step
Tools >Board ... > Boards Manager
Selected  ESP32 by Espressif Systems
Clicked on "Install"  (version 1.0.1)
Get the Error Message
"Tool xtensa-esp32-elf-gcc is not available for your operating system"

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2019-04-10:
The toolchain is not compiled regularly for ARM.  See #789

RobertPHeller, 2019-04-10:
I built a *working* toolchain on a Pi2 (running Raspbian GNU/Linux 8, 4.19.25-v7+).  Available for download here: http://files.deepsoft.com/Other/xtensa-esp32-elf-armv7l/xtensa-esp32-elf-armv7l-6c4433a-01092018.tar.bz2

RamanujamV, 2019-04-10:
Dear Mr. Heller,Thank you for your prompt response to my post.
I presume that this will work on Pi3 also.I will try and will let you know.
Thanks again.Ram Venkataswamy---------------------------------------------------------

me-no-dev, 2019-04-12:
Starting with GCC 8, PI is on the regular toolchain build :) so stay tuned for full official support

greeneagle317, 2019-06-24:
Ramanujam!
Is it OK now. Is the arduino IDE working for esp32 in RPi 3 ?

RobertPHeller, 2019-06-24:
At Mon, 24 Jun 2019 10:01:50 -0700 espressif/arduino-esp32 <reply@reply.github.com> wrote:

> 
> 
> Ramanujam!
> Is it OK now. Is the arduino IDE working for esp32 in RPi 3 ?

I don't know if the Espressif people have now incorporated a build of the
build toolchain for arm (RPi) in their board manager code. *I* built the
toolchain on a RPi 2 and uploaded it to
http://files.deepsoft.com/Other/xtensa-esp32-elf-armv7l/
There are some bugs in the python code to upload to the ESP32 boards, I don't 
know if that is fixed or not.

> 

-- 
Robert Heller             -- 978-544-6933
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
heller@deepsoft.com       -- Webhosting Services

ghhann, 2019-07-07:
Any update on when this will be available for rpi3?  There is no entry point for it in the Json file

lbernstone, 2019-07-07:
When you see an ARM installer [here](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/linux-setup.html), then it will be available.  The toolchain is upstream.

syswombat, 2019-07-31:
Hello
same Problem with Raspberry 4 - hope there is a way to solve it.
have a nice day
vinc

syswombat, 2019-07-31:
Arduino IDE 1.8.9
Tools --> Boards --> Boards Manager
search esp32

esp32 by Espressif Systems
1.0.2 Install

and gives the same error as written on the top

RobertPHeller, 2019-07-31:
At Wed, 31 Jul 2019 00:28:02 -0700 espressif/arduino-esp32 <reply@reply.github.com> wrote:

> 
> 
> Arduino IDE 1.8.9
> Tools --> Boards --> Boards Manager
> search esp32
> 
> esp32 by Espressif Systems
> 1.0.2 Install
> 
> and gives the same error as written on the top

Until such time as the Espressif Systems people get their act together, you
*cannot* install the esp32 on the Raspberry Pi *using the Boards Manager*. You
have to use the (evil?) command line. Search for installing via the command
line. Once you install the all of the board, library, and platform stuff, you
can then download the toolchain from my site and drop it into place.

> 

-- 
Robert Heller             -- 978-544-6933
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
heller@deepsoft.com       -- Webhosting Services

syswombat, 2019-08-09:
@RobertPHeller thanks for your information.
the toolchain is done, but where excatly it have to stay in the Buster (Debian10) place?
also the esp32 Hardware - where on Debian 10 have to be copied?
have a nice day
vinc

me-no-dev, 2019-09-14:
Starting soon Espressif will build the toolchain for ARM just like for the other operating systems. For now you can use the following toolchain (which is not officially supported but seemed to work fine) https://dl.espressif.com/dl/xtensa-esp32-elf-linux-armel-1.22.0-87-gb57bad3-5.2.0.tar.gz
follow the github install guide to get esp32 arduino on the pi, run get.py as per instructions, then download and decompress the archive to [parth-to-esp32-arduino]/tools/ and give it a go. Quite slow ;)

Mike09, 2019-10-20:
Hello everyone,
I encountered similar problems using the ESP32 with the Arduino IDE on a Pi 3 B+.

Can someone explain me the definition for toolchain, maybe I am mixing it up?
Is "toolchain" meant for the building process under the Arduino IDE or does it refer to setting up the system to get it working with the ESP hardware?

My current system is set up as follows:
- Raspberry Pi 3 B+ with Raspbian 10 (buster)
- Arduino IDE version 2:1.0.5 (installed via apt-get)

I used the installation instruction as given at [/docs/arduino-ide/debian_ubuntu.md](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md).
I didn't install pip because it was pre-installed on raspbian as well as pyserial.

But the ESP board does not show up in the Arduino IDE.
What am I missing?
Has something changed for the Arduino IDE, because there is also no Board-Manger showing up?
For me it seems the IDE cannot see the installed files, but this is just a guess.

@me-no-dev why do I need to additionally install the files mentioned in the link above? These files are already existing!?

RobertPHeller, 2019-10-20:
At Sun, 20 Oct 2019 02:08:23 -0700 espressif/arduino-esp32 <reply@reply.github.com> wrote:

> Hello everyone,
> I encountered similar problems using the ESP32 with the Arduino IDE on a Pi 3 B+.
>
> Can someone explain me the definition for toolchain, maybe I am mixing it up?
> Is "toolchain" meant [...]

"Toolchain" refers to the compiler and related programs (assembler, linker, 
library archiver, etc.).

> 
> My current system is set up as follows:
> - Raspberry Pi 3 B+ with Raspbian 10 (buster)
> - Arduino IDE version 2:1.0.5 (installed via apt-get)

You need to install  Arduino IDE 1.8.5, which is probably *not* available via 
apt-get. You need to download the tarball from the Arduino website.

> 
> I used the installation instruction as given at [/docs/arduino-ide/debian_ubuntu.md](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md).
> I didn't install pip because it was pre-installed on raspbian as well as pyserial.

The instructions above are for x86/amd64 machines (Intel/AMD-based "PC" 
hardware).

There are instructions (somewhere) for installing the ESP32 support from
github, except for the actual "toolchain" built for an ARM host (Raspberry Pi,
etc.). There is a copy of the "toolchain" built for an ARM host available on
my server at http://files.deepsoft.com/Other/xtensa-esp32-elf-armv7l/.
Supposedly the espressif people will be providing a "toolchain" built for an
ARM host Real Soon Now...  (This has supposedly been in the works for a while 
now, but people are still waiting...)

> But the ESP board does not show up in the Arduino IDE.
> What am I missing?
> Has something changed for the Arduino IDE, because there is also no Board-Manger showing up?
> For me it seems the IDE cannot [...]

-- 
Robert Heller             -- 978-544-6933
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
heller@deepsoft.com       -- Webhosting Services

lbernstone, 2019-10-20:
Sorry there was no press release, but the ARM toolchain is now available. https://github.com/espressif/arduino-esp32/commit/5ad468f9dcb5f9cc7249a4fb987aead4f1d9b136

atanisoft, 2019-10-20:
As @lbernstone mentions above, boards manager can now be used on rPi as of 1.0.4 (current stable release)

RamanujamV, 2019-10-20:
Information on Dwnloading and Installing the current version of the Arduino IDE in Linux ARM (eg. Raspberry Pi) systems
   
   - Download the Arduino IDE
            Arduino - Software

| 
| 
|  | 
Arduino - Software

Open-source electronic prototyping platform enabling users to create interactive electronic objects.
 |

 |

 |

 Select “Linux ARM32 bits”

---------------------------
   
   - Install the ArduinoSoftware (IDE) on Linux
            This documentexplains how to install the Arduino Software (IDE) on Linux machines.            Arduino - Linux

| 
| 
|  | 
Arduino - Linux

Open-source electronic prototyping platform enabling users to create interactive electronic objects.
 |

 |

 |

--------------------------------------------------------------------------------------------------------------------------------------------

 

> Hello everyone,
> I encountered similar problems using the ESP32 with the Arduino IDE on a Pi 3 B+.
>
> Can someone explain me the definition for toolchain, maybe I am mixing it up?
> Is "toolchain" meant [...]

"Toolchain" refers to the compiler and related programs (assembler, linker, 
library archiver, etc.).

> 
> My current system is set up as follows:
> - Raspberry Pi 3 B+ with Raspbian 10 (buster)
> - Arduino IDE version 2:1.0.5 (installed via apt-get)

You need to install Arduino IDE 1.8.5, which is probably *not* available via 
apt-get. You need to download the tarball from the Arduino website.

> 
> I used the installation instruction as given at [/docs/arduino-ide/debian_ubuntu.md](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md).
> I didn't install pip because it was pre-installed on raspbian as well as pyserial.

The instructions above are for x86/amd64 machines (Intel/AMD-based "PC" 
hardware).

There are instructions (somewhere) for installing the ESP32 support from
github, except for the actual "toolchain" built for an ARM host (Raspberry Pi,
etc.). There is a copy of the "toolchain" built for an ARM host available on
my server at http://files.deepsoft.com/Other/xtensa-esp32-elf-armv7l/.
Supposedly the espressif people will be providing a "toolchain" built for an
ARM host Real Soon Now... (This has supposedly been in the works for a while 
now, but people are still waiting...)

> But the ESP board does not show up in the Arduino IDE.
> What am I missing?
> Has something changed for the Arduino IDE, because there is also no Board-Manger showing up?
> For me it seems the IDE cannot [...]

Mike09, 2019-10-21:
Thank you guys for the quick replies!

I was able to install the current Arduino version via the tarball.
Also I could add the esp32 source for the board manager to install the arm boards.

Now I'm having problems to use the library installed via the esp32.
The Arduino environment detects two "WiFi.h" files.
Even if I delete the original WiFi directory contained in Arduino, the compiler stops because it found two libraries.
Other topics say that this error should be solved in the current Arduino IDE.
Any ideas?

RamanujamV, 2019-10-23:
Information as copied From a post in https://randomnerdtutorials.com/esp32-access-point-ap-web-server/
With the ESP32 and Arduino we use the WiFi.h library. However, those libraries are different for the ESP32 and ESP8266. If you’re having trouble compiling ESP32 code that uses the WiFi.h library, you must remove the Arduino WiFi library from your Arduino IDE installation.
The ESP8266 uses the ESP8266WiFi.h library.
----------------------------------------------------------------------------------------------------------------------------------------

closed 2020-01-05 by stale[bot]

vubon, 2020-01-11:
> Starting soon Espressif will build the toolchain for ARM just like for the other operating systems. For now you can use the following toolchain (which is not officially supported but seemed to work [...]

Hi, I am getting a warring in build time. It is a version mismatch problem. But I did not find anywhere 1.22.0-80-g6c4433a version for ARM. Could you help me?
 
WARNING: Toolchain version is not supported: 1.22.0-87-gb57bad3
Expected to see version: 1.22.0-80-g6c4433a