espressif/arduino-esp32 #2653: Changing channel breaks ESP-Now Master/Slave example

skrenes, 2019-04-10:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

dopyoman, 2019-11-28:
Can someone comment on why this occurs?

skrenes, 2019-12-02:
Support seems none-existent.

sheminasalam, 2022-06-25:
was this solved?

VojtechBartoska, 2022-06-27:
This needs to be checked, reopening.

VojtechBartoska, 2022-06-27:
Can you please take a look @me-no-dev?

me-no-dev, 2022-06-27:
@VojtechBartoska I'm not sure what the problem was. it seems like OP tried to set one channel for master and another for slave and expected the code to work. Should be closed IMHO

sheminasalam, 2022-06-28:
I had a similar situation with ESP8266. On investigation I found that even when we set channel as anything other than 1 while adding peers, the device doesn't execute that. The solution that I found with esp8266 is that we need to first set the channel with wifi.begin and then disconnect wifi with wifi.disconnect and proceed with adding peer. In Code it would look like below.

```
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password, WIFI_CHANNEL,NULL,false);
    WiFi.disconnect(); //we do not want to connect to a WiFi network

    if (esp_now_init() != 0) {
      if (DEBUG_MODE == 1) {
        Serial.println("ESP-NOW initialization failed");
      }
      return;
    }
```

reppad, 2022-07-20:
Hi,

> is anyone able to retest this on v2.0.3? Thanks in advance!

@VojtechBartoska Same on my side on v2.0.4 (not tested before)

> I'm not sure what the problem was. it seems like OP tried to set one channel for master and another for slave and expected the code to work. Should be closed IMHO

Indeed if we leave the default channel values in sketches (master=3 and slave=1) we can't expect the code to work !
In this case we got an explicit message in master output :
```
Found 7 devices
Found a Slave.
1: Slave_1 [XX:XX:XX:XX:XX:XX] (-28)
Slave Found, processing..
Slave Status: Already Paired
Sending: 27
E (205350) ESPNOW: Peer channel is not equal to the home channel, send fail!
Send Status: Invalid Argument
```

But if I set the same channel for slave and master, it not work better and I got this output
```
Found 7 devices
Found a Slave.
1: Slave_1 [XX:XX:XX:XX:XX:XX] (-28)
Slave Found, processing..
Slave Status: Already Paired
Sending: 4
Send Status: Success
Last Packet Sent to: xx:xx:xx:xx:xx:xx
Last Packet Send Status: Delivery Fail
```

On the slave side I don't have any output after init :
```
ESPNow/Basic/Slave Example
AP Config Success. Broadcasting with AP: Slave_1
AP MAC: 7C:DF:A1:02:DD:8B
ESPNow Init Success
```

I tried a simpler way to get working ESPNow: 2 ESPs in STA mode and hardcoding the peer address on sender side but without success.

My setup:
Win10 / Arduino 1.8.19 / arduino-esp32 2.0.4
Master: Lolin C3 mini (ESP32-C3FH4) / Slave: Lilygo ESP32_S2_WOOR (ESP32-S2-WROOM)

reppad, 2022-07-20:
Hi,

> is anyone able to retest this on v2.0.3? Thanks in advance!

@VojtechBartoska Same on my side on v2.0.4 (not tested before)

> I'm not sure what the problem was. it seems like OP tried to set one channel for master and another for slave and expected the code to work. Should be closed IMHO

Indeed if we leave the default channel values in sketches (master=3 and slave=1) we can't expect the code to work !
In this case we got an explicit message in master output :
```
Found 7 devices
Found a Slave.
1: Slave_1 [XX:XX:XX:XX:XX:XX] (-28)
Slave Found, processing..
Slave Status: Already Paired
Sending: 27
E (205350) ESPNOW: Peer channel is not equal to the home channel, send fail!
Send Status: Invalid Argument
```

But if I set the same channel for slave and master, it not work better and I got this output
```
Found 7 devices
Found a Slave.
1: Slave_1 [XX:XX:XX:XX:XX:XX] (-28)
Slave Found, processing..
Slave Status: Already Paired
Sending: 4
Send Status: Success
Last Packet Sent to: xx:xx:xx:xx:xx:xx
Last Packet Send Status: Delivery Fail
```

On the slave side I don't have any output after init :
```
ESPNow/Basic/Slave Example
AP Config Success. Broadcasting with AP: Slave_1
AP MAC: XX:XX:XX:XX:XX:XX
ESPNow Init Success
```

I tried a simpler way to get working ESPNow: 2 ESPs in STA mode and hardcoding the peer address on sender side but without success.

My setup:
Win10 / Arduino 1.8.19 / arduino-esp32 2.0.4
Master: Lolin C3 mini (ESP32-C3FH4) / Slave: Lilygo ESP32_S2_WOOR (ESP32-S2-WROOM)

VojtechBartoska, 2022-07-26:
Adding this to our Roadmap as it needs further testing.

PilnyTomas, 2022-08-09:
Hi, I have opened PR to fix the small problem with the channel, but I could not replicate your issue with `Delivery Fail`
Tested on ESP32 DevKit

PilnyTomas, 2022-08-10:
Thanks for the info.
I have tried using the same chips as you (different boards) - Master C3; Slave S2 - and it is working for me ok.
Can you please try compiling it with a verbose debug level?
Maybe changing both channels to different number might do something.

reppad, 2022-08-14:
Hi,

Some news after several tests.

ESPNow does not work with the Lolin C3 Mini board specifically, I was able to get it to work successfully on another ESP32-C3 based board with native USB.
It may be related to the fact that the Lolin C3 Mini uses an ESP32-C3FH4 chip with built-in flash memory ? The working board uses an external SPI flash chip.

Here are the "verbose" logs of the Lolin C3 Mini running "Master" sketch
```
[   249][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino EventESPNow/Basic/Master Example
STA MAC: 60:55:F9:29:FE:FC
ESPNow Init Success
[  3112][V][WiFiGeneric.cpp:381] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 6
[  3113][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE

Found 6 devices 
Found a Slave.
1: Slave_1 [7C:DF:A1:02:DD:8B] (-31)
Slave Found, processing..
Slave Status: Pair success
Sending: 1
Send Status: Success
Last Packet Sent to: 7c:df:a1:02:dd:8b
Last Packet Send Status: Delivery Fail
[ 11041][V][WiFiGeneric.cpp:381] _arduino_event_cb(): SCAN Done: ID: 129, Status: 0, Results: 4
[ 11042][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE

Found 4 devices 
Found a Slave.
1: Slave_1 [7C:DF:A1:02:DD:8B] (-33)
Slave Found, processing..
Slave Status: Already Paired
Sending: 2
Send Status: Success
Last Packet Sent to: 7c:df:a1:02:dd:8b
Last Packet Send Status: Delivery Fail
```

On the slave side, nothing happens.

On the other hand I detected another problem, even with the board for which ESPNow works (by re-reading I think this is the problem initially reported by @skrenes):

If I use a channel other than 1 (same value on the master and slave side of course), the communication does not work and we get the behavior below

```
[   247][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino EventESPNow/Basic/Master Example
STA MAC: 60:55:F9:79:A4:5C
ESPNow Init Success

Found 10 devices 
Found a Slave.
1: Slave_1 [7C:DF:A1:02:DD:8B] (-47)
Slave Found, processing..
Slave Status: Pair success
Sending: 1
Send Status: Invalid Argument

Fou[ 20394][V][WiFiGeneric.cpp:381] _arduino_event_cb(): SCAN Done: ID: 130, Status: 0, Results: 12
[ 20394][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE

Found 12 devices 
Found a Slave.
1: Slave_1 [7C:DF:A1:02:DD:8B] (-33)
Slave Found, processing..
Slave Status: Already Paired
Sending: 2
E (20481) ESPNOW: Peer channel is not equal to the home channel, send fail!
Send Status: Invalid Argument
```

reppad, 2022-08-17:
There is definitely a hardware design issue with the board I used for my 1st test :
https://www.wemos.cc/en/latest/tutorials/c3/get_started_with_arduino_c3.html#wifi
https://github.com/arendst/Tasmota/discussions/15443

So the issue is only about using channel other than 1 :

> If I use a channel other than 1 (same value on the master and slave side of course), the communication does not work and we get the behavior below
>
> ```
> [   247][D][WiFiGeneric.cpp:929] [...]

PilnyTomas, 2022-08-19:
Hi, it seems that the WiFi library does not allow changing channel for Station, so it is impossible to use any other channel than 1.
BTW could please explain why you need other channels? It might be possible to add this ability later, but I'm not promising anything...
In [ESP-IDF example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/espnow) it is possible to change the channel in menuconfig, however, this results in the same error as for you: `ESPNOW: Peer channel is not equal to the home channel, send fail!
`
I will create an issue for IDF example nad hopefully we will hear more from the IDF team.

reppad, 2022-08-19:
> Hi, it seems that the WiFi library does not allow changing channel for Station, so it is impossible to use any other channel than 1.

OK, that explains the behavior.

> BTW could please explain why you need other channels?

In my case, I tried to change to use a channel as less congested as possible. 
But honestly it's not critical and if there hadn't been a need to modify the default channel to make the sketch work, I probably wouldn't have tried to configure it!

> It might be possible to add this ability later, but I'm not promising anything...

That would be great, in the meantime it might be helpful to put a comment on the CHANNEL constant?

Thanks a lot for the help!

PilnyTomas, 2022-08-19:
> put a comment on the CHANNEL constant

I will just remove the option to change `CHANNEL` until the real issue is solved (if ever).

laverdanny, 2023-04-04:
Hi there !
Is this issue still running ?

As [juantinazzo] said the problem is when you need to connect one device (at least) to router, in order to communicate out of local network.

Even if [juantinazzo] workaround works, all ESP-Now devices have to get connected on the same router to synchronise their operating channel, but they often can't be in router's range ....

P-R-O-C-H-Y, 2024-04-03:
This issue can be closed as solved by the ESP-NOW library PR #9395.
Channel of the ESP-NOW peer can be specified, but must be same with the Wifi Interface channel.

closed 2024-04-03 by P-R-O-C-H-Y