espressif/arduino-esp32 #2654: I2C stop working

Boia11, 2019-04-10:
Hi, I have a LOLIN D32 Pro and with multiple I2C communication the I2C stop to respond.
I have create sketch master (esp32) and slave (Arduino micro) but after send rotation command the communication crash and not respond.

Sketch master:
```
#include <Wire.h>
#include "I2C_Anything.h"

#define MOTOR_STEPS 200

#define I2C_ADDR_MOTOR1 0x01

char StatoMotore = 'R';
long int PassiMotore = 100*MOTOR_STEPS;
int RPM = 120;

void setup() {
  Wire.begin();
  Serial.begin(115200);
  Wire.beginTransmission(I2C_ADDR_MOTOR1);
  I2C_writeAnything('A');
  Wire.endTransmission();
  delay(5000);
  Wire.beginTransmission(I2C_ADDR_MOTOR1);
  //Wire.write(a);
  I2C_writeAnything(StatoMotore);
  I2C_writeAnything(PassiMotore);
  I2C_writeAnything(RPM);
  //Wire.write(b);
  Wire.endTransmission();
  Serial.print("Dati inviati: ");
  Serial.print(StatoMotore);
  Serial.print("    ");
  Serial.print(PassiMotore);
  Serial.print("    ");
  Serial.println(RPM);
  delay(1000*15);
  Wire.beginTransmission(I2C_ADDR_MOTOR1);
  I2C_writeAnything('E');
  Wire.endTransmission();
  Serial.println("Force - Motor Stop");
  delay(1000*5);
  Wire.beginTransmission(I2C_ADDR_MOTOR1);
  I2C_writeAnything('D');
  Wire.endTransmission();
  Serial.println("Force - Motor Disable");
}

void loop() { 
}
```
Sketch slave:
```
#include <Wire.h>
#include "I2C_Anything.h"

#define I2C_ADDR 0x01

//Specifiche Motore
#define MOTOR_STEPS 200
int RPM = 120;
#define MICROSTEPS 1 // Microstepping 1=full step, 2=half step etc.

// PIN Driver
#define DIR 4
#define STEP 5
#define ENABLE 12

//Motore
#include <DRV8825.h>
#define MODE0 9
#define MODE1 10
#define MODE2 11
DRV8825 stepper(MOTOR_STEPS, DIR, STEP, ENABLE, MODE0, MODE1, MODE2);

//Menu
#define M_ACTIVE 'A'
#define M_DISABLE 'D'
#define M_RUN 'R'
#define M_EMERGENCY 'E'

char StatoMotore = M_DISABLE; //Active|Disable|Run|Emergency
long int PassiMotore = 0;
bool MotorArmed = false;

long previousMillis = 0;
int interval = 1000;

long previousMillisLED = 0;
int interval_LED = 500;
int ledState = LOW;

void setup() {

  //Seriale
  Serial.begin(115200);

  //LED
  pinMode(LED_BUILTIN, OUTPUT);
  
  //I2C
  Wire.begin(I2C_ADDR);
  delay(2000);
  Serial.println("I2C init");
  Wire.onReceive(Command);

  //Motor
  stepper.begin(RPM, MICROSTEPS);
  stepper.setEnableActiveState(LOW);
  Serial.println("Motor Init");
  
}

void loop() {
  //Serial.println("Loop");
  unsigned wait_time_micros = stepper.nextAction();

  unsigned long currentMillis = millis();

  if(MotorArmed == true) {
    // 0 wait time indicates the motor has stopped
    if (wait_time_micros <= 0) {
      //stepper.disable();       // comment out to keep motor powered
      if(currentMillis - previousMillis > interval) {
        Serial.println("Motore in Stand-by");
        previousMillis = currentMillis;
      }
      if(currentMillis - previousMillisLED > interval_LED) {
        if (ledState == LOW)
          ledState = HIGH;
        else
          ledState = LOW;
        digitalWrite(LED_BUILTIN, ledState);
        previousMillisLED = currentMillis;
      }
    }

    // (optional) execute other code if we have enough time
    if (wait_time_micros > 100){
      //Code to execute
    }
  } else {
    if(currentMillis - previousMillis > interval) {
    Serial.println("Motore Spento");
    previousMillis = currentMillis;
      }
  }

}

void Command() {

  I2C_readAnything(StatoMotore);
  I2C_readAnything(PassiMotore);
  I2C_readAnything(RPM);
  Serial.print("Dati ricevuti:   StatoMotore = ");
  Serial.print(StatoMotore);
  Serial.print("    PassiMotore = ");
  Serial.print(PassiMotore);
  Serial.print("    RPM = ");
  Serial.println(RPM);

  stepper.setRPM(RPM);
  
  switch(StatoMotore) {
    case M_DISABLE:
    {
      stepper.disable();
      MotorArmed = false;
      Serial.println("Motore Disabilitato");
    } break; //end case Disable
    
    case M_ACTIVE:
    {
      stepper.enable();
      MotorArmed = true;
      Serial.println("Motore Attivato");
    } break; //end case Active

    case M_EMERGENCY:
    {
      stepper.stop();
      Serial.println("Arresto d'Emergenza");
    } break; //end case Emergency

    case M_RUN:
    {
      Serial.println("Inzio Rotazione");
      //stepper.startMove(PassiMotore * MOTOR_STEPS * MICROSTEPS);
      stepper.startMove(PassiMotore * MICROSTEPS);
      ledState = HIGH;
      digitalWrite(LED_BUILTIN, ledState);
    } break; //end case Run
    
    default:
    {
      StatoMotore = M_EMERGENCY;
    } 
  }
}
```

stickbreaker, 2019-04-11:
Try reading the error codes on the esp32.

```
void setup() {
  Wire.begin();
  Serial.begin(115200);
  Wire.beginTransmission(I2C_ADDR_MOTOR1);
  I2C_writeAnything('A');
  uint8_t err= Wire.endTransmission();
  If (err != I2C_ERROR_OK){ // problem
    Serial.printf("Problem sending Activate command endTransmission returned %d(%s).",Wire.lastError(), Wire.getErrorText(Wire.lastError()));
  }
  delay(5000);
  Wire.beginTransmission(I2C_ADDR_MOTOR1);
  //Wire.write(a);
  I2C_writeAnything(StatoMotore);
  I2C_writeAnything(PassiMotore);
  I2C_writeAnything(RPM);
  //Wire.write(b);
  err =  Wire.endTransmission();
  If (err != I2C_ERROR_OK){ // problem
    Serial.printf("Problem sending statoMorre command endTransmission returned %d(%s).",Wire.lastError(), Wire.getErrorText(Wire.lastError()));
  }
  Serial.print("Dati inviati: ");
  Serial.print(StatoMotore);
  Serial.print("    ");
  Serial.print(PassiMotore);
  Serial.print("    ");
  Serial.println(RPM);
  delay(1000*15);
  Wire.beginTransmission(I2C_ADDR_MOTOR1);
  I2C_writeAnything('E');
  err =  Wire.endTransmission();
  If (err != I2C_ERROR_OK){ // problem
    Serial.printf("Problem sending E command endTransmission returned %d(%s).",Wire.lastError(), Wire.getErrorText(Wire.lastError()));
  }
  Serial.println("Force - Motor Stop");
  delay(1000*5);
  Wire.beginTransmission(I2C_ADDR_MOTOR1);
  I2C_writeAnything('D');
  err =  Wire.endTransmission();
  If (err != I2C_ERROR_OK){ // problem
    Serial.printf("Problem sending D command endTransmission returned %d(%s).",Wire.lastError(), Wire.getErrorText(Wire.lastError()));
  }
  Serial.println("Force - Motor Disable");
}
```

Chuck.

Boia11, 2019-04-11:
> Try reading the error codes on the esp32.
>
> ```
> void setup() {
>   Wire.begin();
>   Serial.begin(115200);
>   Wire.beginTransmission(I2C_ADDR_MOTOR1);
>   I2C_writeAnything('A');
>   uint8_t err= [...]

Nothing
```
⸮.⸮⸮⸮⸮ (POWERON_RESET),⸮⸮⸮0x37 (SPI_FASTe1SH_BOOT)
config⸮⸮ 0, SPIWP:0xee
clk_dr⸮⸮⸮⸮0,q_drv:0x00,d_drv:0⸮0,cs0_drv:0x00,hd_drv:0x00⸮⸮}⸮rv:0x00
mode:DIO, c⸮k⸮⸮div:1
load:0x3ff⸮⸮⸮bb⸮⸮⸮4
load:0x&⸮⸮⸮⸮c,len:1100
load'⸮⸮0078000,len:103L⸮H⸮⸮+⸮⸮0x40080400,⸮⸮:6460
entry 0x⸮6a4
Test operativo controllo motori

Motore abilitato

Dati inviati: R    20000    120

Force - Motor Stop
```
Where is the problem? I try to connect the ESP32 (SDA, SCL and GND) with Arduino Micro but it's the same

stickbreaker, 2019-04-12:
@Boia11 are you directly connecting the ESP32 to the UNO?  The ESP32 is a 3.3v device, the UNO is 5V  you need voltage level converter circuit to successfully connect them.

[adafruit i2c level shifter](https://www.adafruit.com/product/757?gclid=EAIaIQobChMIjprErbDJ4QIVeyCtBh2CGwK1EAQYASABEgJp-_D_BwE)

Except, the ESP32 would error if it could not see the UNO?  

You have a problem with your UNO slave code, the `onReceive()` function need to be defined as:
[Arduino onReceive](https://www.arduino.cc/en/Reference/WireOnReceive)
```c++
void myHandler(int numBytes);
```
So, your `void Command();` needs to be changed to `void Command(int numbytes);`  Also, `Command()` is an interrupt function, therefore you must only use function and variables that are safe to be accessed inside an interrupt, `Serial()` cannot be accessed.

Change your code to something like this:
```c++
#define DEBUGMAX 300
static uint16_t debugPos =0;
static char debugBuffer[DEBUGMAX];
void Command(int numberBytes) {
 //I don't know the format the readAnything converts the data too?
// it should be possible to use numberBytes to identify which message type has been send.
// numberBytes contains how long the total message is.
// this is my assumption:
// 1 = just command
// 7 = command + passiMorte + rpm

  I2C_readAnything(StatoMotore);
if(numberBytes==7){
  I2C_readAnything(PassiMotore);
  I2C_readAnything(RPM);
  stepper.setRPM(RPM);
} else {
 PassiMotore = 0;
 RPM= 0;
}
  // clear out old msg
  debugPos = sprintf((char*)&debugBuffer[debugPos),"Dati ricevuti:   StatoMotore = %d PassiMotore =%d rpm=%d\n",StatoMotore,PassiMotore,RPM);
  
  switch(StatoMotore) {
    case M_DISABLE:
      stepper.disable();
      MotorArmed = false;
      debugPos += sprintf((char)&debugBuffer(debugPos),"Motore Disabilitato\n");
     break; //end case Disable
    
    case M_ACTIVE:
       stepper.enable();
       MotorArmed = true;
       debugPos += sprintf((char)&debugBuffer(debugPos),"Motore Attivato\n");
       break; //end case Active

    case M_RUN:
        debugPos += sprintf((char)&debugBuffer(debugPos),"Inzio Rotazione\n");
        stepper.startMove(PassiMotore * MICROSTEPS);
        ledState = HIGH;
        digitalWrite(LED_BUILTIN, ledState);
        break; //end case Run
    
    case M_EMERGENCY:
    default :
       stepper.stop();
       debugPos += sprintf((char)&debugBuffer(debugPos),"Arresto d'Emergenza\n");
       break; //end case Emergency
       StatoMotore = M_EMERGENCY;
    
  }
}

void loop() {
  //Serial.println("Loop");
  if(debugPos != 0){
    Serial.print(debugBuffer);
    debugPos = 0;
  }
  unsigned wait_time_micros = stepper.nextAction();

  unsigned long currentMillis = millis();

  if(MotorArmed == true) {
    // 0 wait time indicates the motor has stopped
    if (wait_time_micros <= 0) {
      //stepper.disable();       // comment out to keep motor powered
      if(currentMillis - previousMillis > interval) {
        Serial.println("Motore in Stand-by");
        previousMillis = currentMillis;
      }
      if(currentMillis - previousMillisLED > interval_LED) {
        if (ledState == LOW)
          ledState = HIGH;
        else
          ledState = LOW;
        digitalWrite(LED_BUILTIN, ledState);
        previousMillisLED = currentMillis;
      }
    }

    // (optional) execute other code if we have enough time
    if (wait_time_micros > 100){
      //Code to execute
    }
  } else {
    if(currentMillis - previousMillis > interval) {
    Serial.println("Motore Spento");
    previousMillis = currentMillis;
      }
  }

}
```
This code as not been tested or compiled, I expect there are syntax errors.

Chuck.

Boia11, 2019-04-13:
> 
> 
> @Boia11 are you directly connecting the ESP32 to the UNO? The ESP32 is a 3.3v device, the UNO is 5V you need voltage level converter circuit to successfully connect them.
> 
> [adafruit i2c level shifter](https://www.adafruit.com/product/757?gclid=EAIaIQobChMIjprErbDJ4QIVeyCtBh2CGwK1EAQYASABEgJp-_D_BwE)
> 
> Except, the ESP32 would error if it could not see the UNO?
> 
> You have a problem with your UNO slave code, the `onReceive()` function need to be defined as:
> [Arduino onReceive](https://www.arduino.cc/en/Reference/WireOnReceive)
> 
> ```c++
> void myHandler(int numBytes);
> ```
> 
> So, your `void Command();` needs to be changed to `void Command(int numbytes);` Also, `Command()` is an interrupt function, therefore you must only use function and variables that are safe to be accessed inside an interrupt, `Serial()` cannot be accessed.
> 
> Change your code to something like this:
> 
> ```c++
> #define DEBUGMAX 300
> static uint16_t debugPos =0;
> static char debugBuffer[DEBUGMAX];
> void Command(int numberBytes) {
>  //I don't know the format the readAnything converts the data too?
> // it should be possible to use numberBytes to identify which message type has been send.
> // numberBytes contains how long the total message is.
> // this is my assumption:
> // 1 = just command
> // 7 = command + passiMorte + rpm
> 
>   I2C_readAnything(StatoMotore);
> if(numberBytes==7){
>   I2C_readAnything(PassiMotore);
>   I2C_readAnything(RPM);
>   stepper.setRPM(RPM);
> } else {
>  PassiMotore = 0;
>  RPM= 0;
> }
>   // clear out old msg
>   debugPos = sprintf((char*)&debugBuffer[debugPos),"Dati ricevuti:   StatoMotore = %d PassiMotore =%d rpm=%d\n",StatoMotore,PassiMotore,RPM);
>   
>   switch(StatoMotore) {
>     case M_DISABLE:
>       stepper.disable();
>       MotorArmed = false;
>       debugPos += sprintf((char*)&debugBuffer(debugPos),"Motore Disabilitato\n");
>      break; //end case Disable
>     
>     case M_ACTIVE:
>        stepper.enable();
>        MotorArmed = true;
>        debugPos += sprintf((char*)&debugBuffer(debugPos),"Motore Attivato\n");
>        break; //end case Active
> 
>     case M_RUN:
>         debugPos += sprintf((char*)&debugBuffer(debugPos),"Inzio Rotazione\n");
>         stepper.startMove(PassiMotore * MICROSTEPS);
>         ledState = HIGH;
>         digitalWrite(LED_BUILTIN, ledState);
>         break; //end case Run
>     
>     case M_EMERGENCY:
>     default :
>        stepper.stop();
>        debugPos += sprintf((char*)&debugBuffer(debugPos),"Arresto d'Emergenza\n");
>        break; //end case Emergency
>        StatoMotore = M_EMERGENCY;
>     
>   }
> }
> 
> void loop() {
>   //Serial.println("Loop");
>   if(debugPos != 0){
>     Serial.print(debugBuffer);
>     debugPos = 0;
>   }
>   unsigned wait_time_micros = stepper.nextAction();
> 
>   unsigned long currentMillis = millis();
> 
>   if(MotorArmed == true) {
>     // 0 wait time indicates the motor has stopped
>     if (wait_time_micros <= 0) {
>       //stepper.disable();       // comment out to keep motor powered
>       if(currentMillis - previousMillis > interval) {
>         Serial.println("Motore in Stand-by");
>         previousMillis = currentMillis;
>       }
>       if(currentMillis - previousMillisLED > interval_LED) {
>         if (ledState == LOW)
>           ledState = HIGH;
>         else
>           ledState = LOW;
>         digitalWrite(LED_BUILTIN, ledState);
>         previousMillisLED = currentMillis;
>       }
>     }
> 
>     // (optional) execute other code if we have enough time
>     if (wait_time_micros > 100){
>       //Code to execute
>     }
>   } else {
>     if(currentMillis - previousMillis > interval) {
>     Serial.println("Motore Spento");
>     previousMillis = currentMillis;
>       }
>   }
> 
> }
> ```
> 
> This code as not been tested or compiled, I expect there are syntax errors.
> 
> Chuck.
> 
> edited to fix syntax

Yes, I use one voltage level converter circuit.
I have solve the problem! The problem was the function for send long I2C command (I2C_readAnything) with ESP32 not work....I change those commands with array.
THANKS for help man!
Can I ask you, with ESP32 and two Arduino Micro (with single i2c level shifter) I need pull-up resistence? if yes, can you explain how need (for normal and fast i2c mode)?

stickbreaker, 2019-04-13:
@Boia11 If you are using a converter similar to AdaFruit's, you will need to add pullups to each side, the AdaFruit circuit uses 10k pullups on each side. For 400khz speed you need to get the combined resistance down to 2.4k or 3.3k, with two 10k's it is down to 5k.  Add two more 10k resistors, one connected on the ESP32 side to 3.3v and the other on the Arduino Micro's side connected to 5v.  That will create an equivalent resistance of 2.5k.  You will need to do this for both, SCL and SDA.

Chuck.

Boia11, 2019-04-13:
> @Boia11 If you are using a converter similar to AdaFruit's, you will need to add pullups to each side, the AdaFruit circuit uses 10k pullups on each side. For 400khz speed you need to get the [...]

I use uno of that i2c level shifter:  [https://s.click.aliexpress.com/e/bAlFjuqU](https://s.click.aliexpress.com/e/bAlFjuqU)
Please can you draw me the schema with level shifter for both mode (standard 100 kbit/s and fast 400 kbit/s)?

Boia11, 2019-04-17:
@stickbreaker I use one of that i2c level shifter:  [https://s.click.aliexpress.com/e/bAlFjuqU](https://s.click.aliexpress.com/e/bAlFjuqU). This module have 10k pull-up SMD resistance bult-in. I need to add other resistence for standard mode at 100 kbit/s? and for fast mode at 400 kbit/s (i need other like capacitors for this mode)?
Thanks for help

closed 2019-04-23 by Boia11