espressif/arduino-esp32 #2657: Slow ADC Sampling Rate

bharath1000, 2019-04-11:
### Hardware:
Board:							 Adafruit Huzzah feather board, wroom-32
IDE name:					          Arduino IDE

### Description:
Hi all,

I'm pretty novice at both arduino, esp-idf, and programming in general, but I am working on a project to take EMG signals (muscle activity) and send it over WiFi to a computer for analysis. So far, I seem to have gotten it working overall, except for the fact that I cannot get a sample rate faster than 125 Hz! Ideally, I need to reach at least 1000 Hz sample rate to get a good signal.

I have tried interrupts, and rtos for multitasking, but nothing has worked. I have even tried an empty loop simply printing micros() since the last loop, and even that only goes ~600 Hz Everywhere I look online, everyone is able to reach 30k+ Hz sample rate. What am I doing wrong?

I've attached my code that uses interrupts (again, much of this is pasted from resources online). Using the adafruit feather board, wroom-32

Thank you!

```cpp

//Change the code below by your sketch
// Include Libraries
#include "Arduino.h"
#include <WiFi.h>
#include <WiFiClient.h>
#include "SparkFunLSM6DS3.h"
#include "Wire.h"
#include "SPI.h"
#include <MadgwickAHRS.h>
#include "esp_wpa2.h" //wpa2 library for connections to Enterprise networks

LSM6DS3 myIMU(SPI_MODE,21); //Default constructor is I2C, addr 0x6B
Madgwick filter;
float EMA_a = 0.3;    //initialization of EMA alpha
uint16_t EMA_S = 0;        //initialization of EMA S

#define EAP_ANONYMOUS_IDENTITY "anonymous@example.com"
#define EAP_IDENTITY "******"
#define EAP_PASSWORD "*******"
const char* ssid = "*****"; // Eduroam SSID
//const char* host = "arduino.php5.sk"; //external server domain for HTTP connection after authentification
uint16_t counter = 0;
bool wifistat = false;
WiFiClient client;

unsigned long microsPerReading, microsPrevious, timer, microfreq, currentaccel, currentemg, currentprint, previousaccel=millis(), previousemg=millis(), previousprint=millis();

//// Set these to your desired credentials.
//const char *ssid = "GGesp32";
//const char *password = "capstone";

WiFiServer server(80);

//Timer Interrupts
hw_timer_t *acceltimer = NULL;
hw_timer_t *emgtimer = NULL;
hw_timer_t *printtimer = NULL;
bool updateEMG = false;
bool updateAccel = false;
bool updatePrint = false;

// Initialize Analog Pins
uint16_t EMGLeftRA_pin = 39; //A3
uint16_t EMGRightRA_pin = 34; //A2
uint16_t EMGLeftOb_pin = 36; //A4
uint16_t EMGRightOb_pin = 33; //33
uint16_t EMGErect_pin = 32; //32

// Initialize EMG Aquisition Variables
uint16_t EMGleftRA=0;
uint16_t EMGrightRA=0;
uint16_t EMGleftob=0;
uint16_t EMGrightob=0;
[... 247 lines cut ...]
      Serial.print(",");
      client.write(e);
      Serial.print(e);
      client.write(",");
      Serial.print(",");
      client.write(Ax);
      Serial.print(Ax);
      client.write(",");
      Serial.print(",");
      client.write(Ay);
      Serial.print(Ay);
      client.write(",");
      Serial.print(",");
      client.write(Az);
      Serial.println(Az);

//ENDING DATA TRANSMISSION 
        client.write("\r");
      
    }
    //close the connection
    client.stop();
    Serial.println("Client Disconnected.");
}
}

josefinHeilig, 2019-04-11:
Hi, 
 You did not set your boolean variable `updateEMG = false;`  again in your `if (updateEMG ==true){}` condition, so this is always entered and executed, which slows it down.
And I would try using AsyncUDP for the Wifi like in the example ESP32 Async UDP. 
The Wifi TCP protocol  always has to wait for the handshake with the server and that possibly slows the main loop down.
Best, Katrin

bharath1000, 2019-04-11:
Thank you for your reply! 

yes sorry that is fixed in the actual running code, must have removed it when copying by accident. Running the code even without any networking, without any extra functions, just doing a single analog read in the loop and nothing else is maxing out at around 300 Hz. This is confusing to me as the documentation says that it can go up to 6KHz and while through googling I've seen people go even higher than that.

Below is an example of the code we're using (with most of the extra features cut out), still giving us a very slow sampling rate. We are tracking this sampling rate by using the micros() function.

``
// Include Libraries
#include "Arduino.h"

// Initialize Analog Pins
uint16_t EMGLeftRA_pin = 39; //A3

// Initialize Mapping for EMG data
uint16_t Min = 0;
uint16_t Max = 4095;
uint16_t Map_Min= 0;
uint16_t Map_Max= 2200;

// Initialize EMG Aquisition Variables
uint16_t EMGleftRA=0;

// Setup the essentials for your circuit to work. It runs first every time your circuit is powered with electricity.
void setup() 
{
     Serial.begin(115200);
  delay(10);
}

// Main logic of your circuit. It defines the interaction between the components you selected. After setup, it runs over and over again, in an eternal loop.
void loop() 
{
  
    EMGleftRA = analogRead(39);
    delayMicroseconds(1);
    EMGleftRA=constrain(EMGleftRA*1.814,Min,Max);
    EMGleftRA=map(EMGleftRA,Min,Max,Map_Min,Map_Max);

    Serial.println(micros());
}

zekageri, 2019-04-26:
Hi bharath1000! I hope you need some suggestions after all. I was struggeling with the analog read too and i managed to find a piece of code on a group in facebook a couple of weeks ago. I wanted to measure a 3Khz signal. The thing is if you want to measure a signal you need twice reading speed than the measured one. So the code in Arduino ide is capable to measure with 44Khz but unfortunatelly this is slows down everything on the esp except the measure. Because if we measure we have to disable all interrutps wich of course affect the wifi and things like that that are on your loop.

//================================ ADC Setup =================================
#define CHANNEL 34
int offset = 1757; //middle DC level reading
int vReal[1024];
const double samplingFrequency = 44000;  //Hz
unsigned int sampling_period_us;
unsigned long microseconds;
int samples = 1024;
//================================ ADC Setup =================================
void setup(){
Serial.begin(115200);
}

void loop(){

    portDISABLE_INTERRUPTS();
    microseconds = micros();
    sampling_period_us = round(1000000*(1.0/samplingFrequency));
  for(int i=0; i<samples; i++) {
    vReal[i] = analogRead(CHANNEL)/*-offset*/;
  while(micros() - microseconds < sampling_period_us){ } //empty loop
    microseconds += sampling_period_us;
      }
    portENABLE_INTERRUPTS(); 

}

Iam sending the measured data via websocket to a webserver for ChartJS visualization. Iam able to measure properly a 4Khz sinus or square wave signals with this. I hope i was able to help you a little bit. And if anyone know a better way iam listening :D

closed 2019-08-14 by stale[bot]