espressif/arduino-esp32 #2658: Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.

nomanshahsyed, 2019-04-11:
My code is giving this error. I have updated the Arduino to the latest version and the esp32 is also updated.

ESP exception decoder gives following:

PC: 0x40135674
EXCVADDR: 0x00000000

Decoding stack results
0x4008ef25: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

### Hardware:
Board:							ESP32 WROOM 32

### Description:
The Error is coming only when WPS is started. Otherwise everything is fine but WPS is the main step to connect to WiFi. I had the same Error when running WPS example from ESP32 into Sparkfun ESP32 thing. 

I was also getting the brownout detector triggered error but solved it by.
```
    WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
```

```
if (WiFi.status() != WL_CONNECTED)
    {
        WiFi.onEvent(WiFiEvent);
        WiFi.mode(WIFI_MODE_STA);

        Serial.println("Starting WPS");

        wpsInitConfig();
        esp_wifi_wps_enable(&config);
        esp_wifi_wps_start(0);
    }

```
Error is below.
```

Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40135670: eeeeeeee ffffffef ffffffff
Core 0 register dump:
PC      : 0x40135674  PS      : 0x00060d30  A0      : 0x8008cdf8  A1      : 0x3ffb5640  
A2      : 0x00000003  A3      : 0x00000001  A4      : 0x3ffc6960  A5      : 0xffffffff  
A6      : 0x00000001  A7      : 0x3ffbf6bc  A8      : 0x80084147  A9      : 0x3ffb5630  
A10     : 0x00000000  A11     : 0x3ffb5744  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060d20  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40135674:0x3ffb5640 0x4008cdf5:0x3ffb5670 0x4008ef25:0x3ffb56b0
```
Exception decoder says this.
```
PC: 0x40135674
EXCVADDR: 0x00000000

Decoding stack results
0x4008ef25: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

lbernstone, 2019-04-11:
You have a power issue.  Turning off the alarm does not stop the brownout...

nomanshahsyed, 2019-04-11:
Hi, yes I was already considering the power issue. Strange that the same error was coming on Sparkfun esp32thing. I have put on 1mF alluminium cap and the issue has not come again. WS2812 LEDs are good but the power spikes should be filtered better if they are used. Thanks.

closed 2019-04-11 by nomanshahsyed

aistisraudys, 2019-12-04:
I confirm, attaching 2200uf capacitor to 3v pin solved the problem.