espressif/arduino-esp32 #2659: ESP32-Cam issue

zpowellman, 2019-04-11:
I'm trying to get two new ESP32-Cam Boards running with CameraWebServer.
This is the error I get:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10312
load:0x40080400,len:6460
entry 0x400806a4
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
⸮
[E][camera.c:1084] esp_camera_init(): Camera probe failed with error 0x20001
Camera init failed with error 0x20001

I've selected AI Thinker as the camera type.  The Cameras were purchased from BangGood and described as AI Thinker Cameras.  Pictures are included:

The configuration I'm using is:
#define PWDN_GPIO_NUM     -1
#define RESET_GPIO_NUM    32
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

I'm using Arduino 1.8.9  with ESP32 Board Set Version 1.0.1
I've tested the Cameras with an M5 Stack board and both camera modules are working on M5 stack correctly using the CameraWebServer App.
I'm looking for tips to help debug the devices.  I've requested a schematic from BangGood.  We'll see what happens.  Hopefully someone will recognize the photos.
Thanks in advance
Dan
[image]
[image]

me-no-dev, 2019-04-12:
There already is configuration for this board in the example. Was reported working. M5 forks our own examples, so it's kinda strange. In any case, the error is in I2C. I just updated many things, the camera driver included (now uses Hardware I2C1), so you might want to give it a shot. Also on the top of the example you undefine the proper board and make sure you have slected the "Huge App" partition (if you are not on the AI-Thinker-Cam board)

zpowellman, 2019-04-12:
Thanks for the response and update. Downloaded new driver and new examples, etc.  All works well now. I appreciate your effort...on this and ESP32 et al.  Just one weird thing FYI.  When one of the boards started up, it was very slow...about 0.6fps.  It ran that way for a minute then picked up to about 8 or 9 fps.  Bad connection maybe??  Again thanks much
Dan

me-no-dev, 2019-04-12:
yeah, this would point to network troubles :) the sensor itself is running at the proper speed.

keith721, 2019-06-10:
You need to declare that an esp32dev board type.
```
esphome:
  name: esp32cam01
  platform: ESP32
  board: esp32dev
```

Suwin1993, 2019-11-18:
zpowellman please show us how did you get the camera module working again???

zpowellman, 2019-11-20:
Nothing extraordinary. My devices worked best when I supplied 5V to the module instead of 3.3V.   I also downloaded the latest drivers and newest examples from this site.  

Take a look at this page for additional info.
https://randomnerdtutorials.com/esp32-cam-video-streaming-face-recognition-arduino-ide/

Suwin1993, 2019-11-21:
You tested with the espressif module right?? Not the Ai thinker module

zpowellman, 2019-11-21:
I'm using Arduino 1.8.10.  The selected board is the  "Ai Thinker ESP32-Cam".  The code is the CameraWebServer from the ESP32 > Camera >directory.  The only change I made is to change the #define CAMERA_MODEL_WROVER_KIT to #define CAMERA_MODEL_AI_THINKER  and to run the board on 5V.  It's won't run correctly on 3.3V.  Let me know if you want the code.

Suwin1993, 2019-11-21:
Ok thank you very much!!

absi9797, 2020-02-28:
Do they support face recognition or do I have to use an AI thinker module

RobertSasak, 2020-03-03:
I have experienced slow video as well. IO0 cannot be connected to anything, not even 1 centimeter long cable. Especially not to breadboard. I made a video about it here https://youtu.be/NvmyCBbTGPs

dimitre, 2020-08-11:
@RobertSasak comment was super useful thanks. mine was plugged to a protoboard and even ping was more than 1000ms and erratic. video stream was erratic too. most of the time not working.

APteH, 2020-11-10:
I suppose that IO0 must be connected by a resistor(abt 10K) to +3.3V., Like you do normally on all IO ports. The port are very high impedance and if not connected to plus, very sensitive for statics.

insanity67, 2020-12-29:
Hey guys, thank you so much for the idea with GPIO 0.... banged my head several days...
Simply soldering a 10K resistor between IO0 and 3.3v on my PCB solved the problem with high ping/low framerate completely!

insanity67, 2021-01-13:
Short Update... not all modules seem to be the same. On the last one I had to use a 43 ohm resistor between GPIO0 and 3.3v to get rid of the performance issues and latency. Very strange, on the first board 10k was ok.
If I connect GPIO0 directly to 3.3v the latency is completely fine, but no camera image. With the 43 ohm resistor all fine.

APteH, 2021-01-13:
Well, try to use a capacitor between GPIO0 and ground. Perhaps the HF from the very close antenna  is the  reason. Value of capacitor , somewere between a 3-47PF (if it is the radiation from antenne)  and 10--100NF(spikes from the circuit). You have to play around with different values.

APteH, 2021-01-17:
I almost forgot: Add 2 capacitors Between + and - supply, direct and leads short as possible om the esp32 board. Values I use are 1 pc 10uF in parallel wit 1nF. Good luck

kshitijgarg2609, 2022-09-01:
Recently I have bought 4 esp32 cam one by one and three out of not responding on serial at all even after clicking reset button, i see no flash light blinking at all even tried to run esptool.py to erase_flash and one of the esp has still flash light on. Earlier purchased esp32 cam worked. What to do in this situation?

zpowellman, 2022-09-01:
I too have recently had troubles with esp32 devices.  Specifically, the nodemcu boards, but it may also affect esp32-cam.  If you are using the latest esp32 core, I've noticed the my boards don't reset and boot like they used to.  You have to press down the boot button during first part of the load process. Why, I'm not sure.  Earlier versions of the core don't require that. I think 2.0.1 doesn't have the issue. You might try that with your espcams

kshitijgarg2609, 2022-09-09:
@zpowellman I have found a solution that esp32 cam has changed little bit of internal circuit, you can only use gnd which is next to 5v and no other gnd of esp32 cam should be connect to this gnd.

zoe9216, 2023-02-15:
> @zpowellman I have found a solution that esp32 cam has changed little bit of internal circuit, you can only use gnd which is next to 5v and no other gnd of esp32 cam should be connect to this gnd.

I read your comment and connected GPIO0 to this ground next to 5V during boot, while the other ground next to VOT I connected to FTDI programmer's ground. It still doesn't connect when I press reset button while it is connecting and I get 'A fatal error occurred. Failed to connect to ESP32. No serial data received'. I also connected GND next to GIO0 to it while GND next to 5V was connected to FTDI. But same result. 
In your solution how did you connect the ground pins while loading? Which one to GPIO0 and which one to FTDI? Many thanks.

closed 2023-03-29 by mrengineer7777