espressif/arduino-esp32 #2660: PWM channels of ESP32

mimansamaheshwari, 2019-04-12:
### Hardware:
Board:							   ESP32 Dev Module
Core Installation/update date:	           11/jul/2017
IDE name:						   Arduino IDE
Flash Frequency:					   80Mhz
PSRAM enabled:                                         no
Upload Speed:				           921600
Computer OS:                                            Windows 8.1

### Description:
I want to use 10 PWM channels of ESP32 for certain a task. I found no issue in working with 8 PWM channels at a time but as soon as I configure the 9th channel it does not work correctly. I am sharing a code that should change PWM of 10 channels after small delays. Is there anything wrong done by me in configuring channels?

### Sketch: 
```int pin1 = 2;
int pin2 = 4;
int pin3 = 5;
int pin4 = 12;
int pin5 = 13;
int pin6 = 14;
int pin7 = 15;
int pin8 = 18;
int pin9 = 19;
int pin10 = 21;
void setup() {
  // put your setup code here, to run once:
  for (int i = 0 ; i < 10; i++)
  {
    ledcSetup(i, 50, 8);                      //50Hz , 8 bit
  }
  ledcAttachPin(pin1, 0);
  ledcAttachPin(pin2, 1);
  ledcAttachPin(pin3, 2);
  ledcAttachPin(pin4, 3);
  ledcAttachPin(pin5, 4);
  ledcAttachPin(pin6, 5);
  ledcAttachPin(pin7, 6);
  ledcAttachPin(pin8, 7);
  ledcAttachPin(pin9, 8);
  ledcAttachPin(pin10, 9);

}

void loop() {
  // put your main code here, to run repeatedly:
  for (int i = 0; i < 10; i++)
  {
    ledcWrite(i, 255);
  }
  delay(2000);
  for (int i = 0; i < 10; i++)
  {
    ledcWrite(i, 128);
  }
  delay(2000);
  
  for (int i = 0; i < 10; i++)
  {
    ledcWrite(i, 0);
  }
  delay(2000);
  
}
```

me-no-dev, 2019-04-12:
what do you see as result on the 9th and 10th channels?

mimansamaheshwari, 2019-04-12:
Voltage on both channels keeps changing between 3.3V and 1.6V. But the actual change should be 3.3V to 1.6V to 0V as per the commands given in code.  My board has a builtin LED on gpio 2 which is connected to channel 0 in code. It seems to work fine as per code. But 9th and 10th channel pins behave as mentioned above.

lbernstone, 2019-04-12:
@me-no-dev : I am not good enough at bitwise math to understand what you are trying to get to here, but it doesn't change the value at all: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-ledc.c#L207

me-no-dev, 2019-04-12:
channels 0-7 are connected to HS LEDC and 8-15 are to LS LEDC. There are some differences in the registers and the way they set the clock is different. That value there @lbernstone would change if before that the channel was on. See the bit being set a few lines above.

me-no-dev, 2019-04-12:
Done :) nice find!

closed 2019-04-12 by me-no-dev