espressif/arduino-esp32 #2661: httpUpdate OTA crashes in ESP32

thedim-witted, 2019-04-12:
Make your question, not a Statement, inclusive. Include all pertinent information:

Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		11/jul/2017
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                          Windows 10

### Description:
Describe your problem here
I am trying to integrate http OTA update feature in the ESP32 WROOM board. My code has been perfect uptil now. However, as soon as I add the line for the http, the code crashes at runtime right at the httpUpdate line. The error code shows it is causing some problem with the httpClient code. However, I am not able to debug it.
I was able to reproduce the error with the below sketch.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

/**
   httpUpdate.ino

    Created on: 27.11.2015

*/

#include <Arduino.h>

#include <WiFiMulti.h>

#include <HTTPClient.h>
#include <ESP32httpUpdate.h>

#define FIRMWARE_UPDATE_VERSION "2001"

String updateURL = "XYZ.php";

WiFiMulti wifiMulti;
WiFiClientSecure client;

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

//  for (uint8_t t = 4; t > 0; t--) {
//    Serial.printf("[SETUP] WAIT %d...\n", t);
//    Serial.flush();
//    delay(1000);
//  }

  WiFi.begin("xxxxxxxxxx", "xxxxxxxxxx");
}

void loop() {

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }
  // wait for WiFi connection
  if ((WiFi.status() == WL_CONNECTED)) {

    t_httpUpdate_return ret1 = ESPhttpUpdate.update(updateURL, FIRMWARE_UPDATE_VERSION);

    switch (ret1) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    }
  }
}
```

### Debug Messages:
```

PC: 0x400d3eb0: HTTPClient::connect() at C:\Users\Steiner\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2-rc1\libraries\HTTPClient\src\HTTPClient.cpp line 959
EXCVADDR: 0x00000000

Decoding stack results
0x400d3eb0: HTTPClient::connect() at C:\Users\Steiner\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2-rc1\libraries\HTTPClient\src\HTTPClient.cpp line 959
0x400d3f3d: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:\Users\Steiner\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2-rc1\libraries\HTTPClient\src\HTTPClient.cpp line 512
0x400d3ffb: HTTPClient::GET() at C:\Users\Steiner\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2-rc1\libraries\HTTPClient\src\HTTPClient.cpp line 458
0x400d4ebb: ESP32HTTPUpdate::handleUpdate(HTTPClient&, String const&, bool) (C:\Program Files at x86)\Arduino\libraries\esp32-http-update-master\src\ESP32httpUpdate.cpp line 199
0x400d509b: ESP32HTTPUpdate::update(String const&, String const&) (C:\Program Files at x86)\Arduino\libraries\esp32-http-update-master\src\ESP32httpUpdate.cpp line 48
0x400d1def: loop() at C:\Users\Steiner\AppData\Local\Temp\arduino_modified_sketch_277684/httpUpdate.ino line 63
0x400d793d: loopTask(void*) at C:\Users\Steiner\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2-rc1\cores\esp32\main.cpp line 25
0x4008ef29: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

me-no-dev, 2019-04-12:
since this happens on open, maybe your parameters are not correct? you have them replaced here i guess. What will happen if you make `updateURL` to be `const char *` instead of `String`?

thedim-witted, 2019-04-12:
> since this happens on open, maybe your parameters are not correct? you have them replaced here i guess. What will happen if you make `updateURL` to be `const char *` instead of `String`?

Throwing the same error.

atanisoft, 2019-04-12:
@thedim-witted the failure is likely due to an invalid updateURL, it is missing the http://host/ piece which HTTPClient::connect() is expecting.

Also ESP32HTTPUpdate is not maintained in this repo from what I can tell so maybe this issue would be better suited with the maintainer of that library?

thedim-witted, 2019-04-13:
> @thedim-witted the failure is likely due to an invalid updateURL, it is missing the http://host/ piece which HTTPClient::connect() is expecting.
> 
> Also ESP32HTTPUpdate is not maintained in this repo from what I can tell so maybe this issue would be better suited with the maintainer of that library?

Sure. I will repost it there.

closed 2019-04-13 by thedim-witted