espressif/arduino-esp32 #2665: Latest Git d922557e0183fcf5faf0e59c3e0f6dee52d091d4 break Captiveportal + webserver

luc-github, 2019-04-13:
Webserver stop responding in AP with captive portal with latest GIT
No issue with last month GIT

Using Arduino IDE 1.8.9 / Windows 10 64Bit 

Board:							ESP32 Dev Module
Core Installation version:			1.0.2-git (d922557e0183fcf5faf0e59c3e0f6dee52d091d4)
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Windows 10? ?Mac OSX

### Description:
Just updating with latest git (4/13/2019) previous program using Webserver and Captive portal in AP mode stop responding, disabling captive portal solved the issue

Using Captive Portal Sample code + Wifi Server => Ok
Using Webserver Sample Code => Ok
Using a combination of Captive Portal + Webserver Code => failed

here debug output:
```
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
192.168.0.1
HTTP server started
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: -1, ERR: 104
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 16 - AP_STADISCONNECTED

```
No crash - not stack - just stop responding
Here MCVE using sample code:

```
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <DNSServer.h>

const byte DNS_PORT = 53;
IPAddress apIP(192, 168, 0, 1);
DNSServer dnsServer;
WebServer server(80);

void handleRoot() {
  server.send(200, "text/plain", "hello from esp32!");
}

void handleNotFound() {
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP("ESP3D","12345678");
  
  Serial.println(apIP);
  // if DNSServer is started with "*" for domain name, it will reply with
  // provided IP to all DNS request
  dnsServer.start(DNS_PORT, "*", apIP);
  
  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
  
}

void loop() {
  dnsServer.processNextRequest();
  server.handleClient();
}

```
In MCVE disabling Captive portal, make websrever to be responsive

me-no-dev, 2019-04-13:
when you say "disable captive portal" you really mean "not start dns"?

me-no-dev, 2019-04-13:
those `[E][WebServer.cpp:602] _handleRequest(): request handler not found` expected?

luc-github, 2019-04-13:
>when you say "disable captive portal" you really mean "not start dns"?

I comment :
`dnsServer.start(DNS_PORT, "*", apIP);`
and
`dnsServer.processNextRequest();`

>those [E][WebServer.cpp:602] _handleRequest(): request handler not found expected?
with captive portal disabled it goes to root and I get the expected page `"hello from esp32!"`

even typing the IP : 192.168.0.1 does not work when captive portal is runing, it is really strange

me-no-dev, 2019-04-13:
can you print out what the actual request is? I doubt the issue is in the web server or the dns server.
maybe change in IDF that causes captives to request different page (there were some iOS issues)

luc-github, 2019-04-13:
Client is my android phone
what you mean display `_currentUri` in error log ?

me-no-dev, 2019-04-13:
yes :) I have noticed a while back that my Iphone actually is calling a particular URL, like /captive-smth but to not remember the actual thing.

luc-github, 2019-04-13:
Yes there are always some background request from phone or PC, but these should not prevent WebServer to respond on proper request like 192.168.0.1 which exists

I changed line 602 in WebServer.cpp by :
`log_e("request handler not found : %s", _currentUri.c_str());`
and see the background requests:
```
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
192.168.0.1
HTTP server started
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[E][WebServer.cpp:602] _handleRequest(): request handler not found : /generate_204
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: -1, ERR: 104
[E][WebServer.cpp:602] _handleRequest(): request handler not found : /generate_204
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: -1, ERR: 104
[E][WebServer.cpp:602] _handleRequest(): request handler not found : /chat
[E][WebServer.cpp:602] _handleRequest(): request handler not found : /generate_204
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: -1, ERR: 104
[E][WebServer.cpp:602] _handleRequest(): request handler not found : /mmtls/384eea67
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[E][WebServer.cpp:602] _handleRequest(): request handler not found : /chat
[E][WebServer.cpp:602] _handleRequest(): request handler not found : /generate_204
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: -1, ERR: 104
[E][WebServer.cpp:602] _handleRequest(): request handler not found : /mmtls/06b2440a
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[E][Parsing.cpp:82] _parseRequest(): Invalid request: ⸮
[E][WebServer.cpp:602] _handleRequest(): request handler not found : /chat
[E][WebServer.cpp:602] _handleRequest(): request handler not found : /chat
[E][WebServer.cpp:602] _handleRequest(): request handler not found : /generate_204
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128

```

and it now work !!! - I revert and it still work on MCVE !!!
back to my program and now no more issue !!! - I am little bit lost 

I have reinstalled twice the repo to verify issue before submission to be sure I did not missed something
Seems I cannot reproduce issue anymore now.

The only thing I can imagine is : I just restarted my PC but I do not see the relation

luc-github, 2019-04-13:
Sorry - I do not know why I cannot reproduce anymore ....
So better to close

closed 2019-04-13 by luc-github

me-no-dev, 2019-04-13:
:D i am pretty confused as well. Sounds like a connection was open but not closed

luc-github, 2019-04-13:
yes very weird... anyway it seems  there are 3 possible root cause : 
1 - PC had memory issue and corrupt compilation resolved by rebbot (it is Windows after all)
2 - Murphy law, likely possible too
3 - Me but related to point 2 so better I do not submit issues found the same day

Positive point : I closed 3 issues today 😉 thank you ^_^.

me-no-dev, 2019-04-13:
no thank you :D

OmarAlkassab, 2019-09-27:
@luc-github I faced the same problem, can you tell me how did you solved the issue?