espressif/arduino-esp32 #2666: SD library with SPI devices

e3bble, 2019-04-13:
How to run it with multiple SPI devices?

-I trying to run sd with 2x ADS1118 and w5500 modules.

Main fail are coming then I try to run it with 2x ADS1118 , it give "sdcard_mount(): f_mount failed 0x(3) Card Mount Failed"

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.7.0
IDE name:						Platform.io
Flash Frequency:					240Mhz
Upload Speed:					115200
Computer OS:                                         linux

### Description:
Then I try to run sd with 2x ADS1118 I get an error "sdcard_mount(): f_mount failed 0x(3)" but 2x ADS1118  work fine. In case I run 1x ADS1118 and sd it work like a charm, but then I connect another it somehow break sd communication. in case I use only 2x ADS1118 it work like a charm.

//sketch:
#include <Arduino.h>
#include "FS.h"
#include "SD6.h"
#include "SPI.h"

//Ads1118
#include "sixbox_ADS1118.h"
#define CS 5
#define CS_2 4
#define AI 4
//ADS1118
Ads1118 ads1118(CS);
Ads1118 ads1118_2(CS_2);
float ai[8];
float val_d;
word ai_chan[] = {ads1118.AIN0,ads1118.AIN1,ads1118.AIN2,ads1118.AIN3};
word ai_chan_2[] = {ads1118_2.AIN0,ads1118_2.AIN1,ads1118_2.AIN2,ads1118_2.AIN3};
float ai_compens[] = {150.0, 150.0, 150.0, 150.0, 150.0, 150.0, 150.0, 150.0};
int16_t ai_min[] = {  0,   0,   0,   0,   0,   0,   0,   0};
int16_t ai_max[] = {100, 100, 100, 100, 100, 100, 100, 100};
float ai_h[]={0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2};

String st;

void readFile(fs::FS &fs, const char * path){
    Serial.printf("Reading file: %s\n", path);

    File file = fs.open(path);
    if(!file){
        Serial.println("Failed to open file for reading");
        return;
    }

    Serial.print("Read from file: ");
    while(file.available()){
        Serial.write(file.read());
    }
    file.close();
}

void writeFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Writing file: %s\n", path);

    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }
    if(file.print(message)){
        Serial.println("File written");
    } else {
        Serial.println("Write failed");
    }
    file.close();
}

void appendFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Appending to file: %s\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("Failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("Message appended");
    } else {
        Serial.println("Append failed");
    }
    file.close();
}

void deleteFile(fs::FS &fs, const char * path){
    Serial.printf("Deleting file: %s\n", path);
    if(fs.remove(path)){
        Serial.println("File deleted");
    } else {
        Serial.println("Delete failed");
    }
}

void setup(){
    Serial.begin(115200);
    while(!Serial)
    ;
    delay(10);

    if(!SD.begin(2)){
        Serial.println("Card Mount Failed");
        return;
    }

    uint8_t cardType = SD.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD card attached");
        return;
    }
    readFile(SD, "/Ain.txt");
    deleteFile(SD, "/Ain.txt");
    writeFile(SD, "/Ain.txt", "6box:\n");

delay(5);
    //ads1118
    ads1118.begin();
    ads1118_2.begin();
}
int i = 2;

void loop(){
  while(i>1){
    i--;

    st="";
    //ANALOG READ
    for (int i=0; i < AI; i++){
      val_d = (ads1118.adsRead(ai_chan[i])*1000);
      delay(2);
      val_d = (ads1118.adsRead(ai_chan[i])*1000);
      st = st + String(123)+
      ",A" + i + "," + String(val_d) + "\r\n";
    }

    //ANALOG_ READ
    for (int i=0; i < AI; i++){
      val_d = (ads1118_2.adsRead(ai_chan_2[i])*1000);
      delay(2);
      val_d = (ads1118_2.adsRead(ai_chan_2[i])*1000);
      st = st + String(123) +
      ",A" + (i+4) + "," + String(val_d) + "\r\n";
    }

    Serial.println(st.length()+1);

    char ch[st.length()+1];
    st.toCharArray(ch, st.length()+1);
    Serial.println(ch);
   appendFile(SD, "/Ain.txt", ch);
    delay(5);
    appendFile(SD, "/Ain.txt", "\n**********\n");
    
  }

}

### Debug Messages:
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
Card Mount Failed
129

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino

colarocker, 2019-04-18:
i had the same problem with SD card and w5500, using HSPI and VSPI. the default pins gave me some headache, in the examples it says this: 
vspi:
SCLK = 18, MISO = 19, MOSI = 23, SS = 5
hspi:
SCLK = 14, MISO = 12, MOSI = 13, SS = 15

i left vspi standard, using it for the w5500, for the hspi connection for the SD card i had to change port 12 because it produced errors with the booting routine, so i used port 27 for MISO ->

SCLK = 14, MISO = 27, MOSI = 13, SS = 15

you have to init it with: 
SPIClass SDSPI(HSPI);
SDSPI.begin(14, 27, 13, 15);    

after it you can init the SD card connection with:
SD.begin(15, SDSPI)

e3bble, 2019-04-18:
yes, but it not help me , because i have 4 spi devises. And I modify w5500 and ads1118 library's to work fine on  it, but hen I add SD card it crash somehow. 
But in my case with my library`s it work on one spi for 2 devices ( one of them SD) but I need more. just with different CS pins. 

PS: my modification mainly on spi initialization before transmitting something on spi and deinit after.

colarocker, 2019-04-18:
ok i didn't try this before. did you try different cs-ports for the devices?

e3bble, 2019-04-23:
All the time I init some cs pins for SPI devices.

me-no-dev, 2019-04-23:
this is rather interesting... could you please switch to the ESP32 Dev Board and enable debug in the board menu?

closed 2019-08-15 by stale[bot]