espressif/arduino-esp32 #2667: HTTPClient getString halts if getSize is equal -1

alexodus, 2019-04-13:
the condition inside getString

` if(_size) {`

when _size==-1, is incorrect

Jeroen88, 2019-04-14:
If no _Content-Length_ header is sent, _size seems to be -1 indeed. But more changes are needed than just checking for _size == -1.

me-no-dev, 2019-04-14:
@alexodus very short report mate... what happens, when it happens, why it happens, where it happens (line number or something). Give some info here :P

alexodus, 2019-04-14:
@me-no-dev I'm sorry for the short report! :P
When I call getString in a successfull request to Google IoT Core http bridge, the wemos lolin32 with Arduino ESP32 1.0.1 hangs / halts without reset itself. 

```
 if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
            if  (https.getSize()>0) {        
              String payload = https.getString(); // <- halt
              Serial.println(payload);
            }
          } 

```

I have to check the size of response:

```
 if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
            if  (https.getSize()>0) {        
              String payload = https.getString();
              Serial.println(payload);
            }
          } 

```

In my opinion it's a bad behaviour to permit the halt with _size==-1

me-no-dev, 2019-04-14:
seems like a viable fix :) how about you PR this line?

me-no-dev, 2019-04-14:
1.0.2 will come out (today?) so you might want to hurry :D

alexodus, 2019-04-14:
> seems like a viable fix :) how about you PR this line?

no, it's no a fix, it's a simple control in my code

alexodus, 2019-04-14:
> 1.0.2 will come out (today?) so you might want to hurry :D

I've another  issue, much more important,  about HTTPClient in arduino esp32 version 1.0.2 rc2 (in 1.0.1 is ok). I hope to submit it today! :P

closed 2019-08-15 by stale[bot]