espressif/arduino-esp32 #2670: HTTPClient the POST request fails with "send header failed" error in versione 1.0.2 rc2

alexodus, 2019-04-14:
This code works like a charms in 1.0.1
```
#include <Arduino.h>
#include <WiFi.h>
#include <WiFiMulti.h>
#include <WiFiClientSecure.h>

#include <HTTPClient.h>
#include <base64.h>

int status = WL_IDLE_STATUS;

const char* ssid     = "<your ssid>"; // Your WiFi ssid
const char* password = "<your password>"; //Your Wifi password

/**
 * setup
 */
void setup() {
  // initialize serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // attempt to connect to WiFi network:
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid, password);
    // wait 1 second for connection:
    delay(1000);
  }
  
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

constexpr char* rootCACertificate = \
"-----BEGIN CERTIFICATE-----\n"
"MIIEdTCCA12gAwIBAgIJAKcOSkw0grd/MA0GCSqGSIb3DQEBCwUAMGgxCzAJBgNV\n"
"BAYTAlVTMSUwIwYDVQQKExxTdGFyZmllbGQgVGVjaG5vbG9naWVzLCBJbmMuMTIw\n"
"MAYDVQQLEylTdGFyZmllbGQgQ2xhc3MgMiBDZXJ0aWZpY2F0aW9uIEF1dGhvcml0\n"
"eTAeFw0wOTA5MDIwMDAwMDBaFw0zNDA2MjgxNzM5MTZaMIGYMQswCQYDVQQGEwJV\n"
"UzEQMA4GA1UECBMHQXJpem9uYTETMBEGA1UEBxMKU2NvdHRzZGFsZTElMCMGA1UE\n"
"ChMcU3RhcmZpZWxkIFRlY2hub2xvZ2llcywgSW5jLjE7MDkGA1UEAxMyU3RhcmZp\n"
"ZWxkIFNlcnZpY2VzIFJvb3QgQ2VydGlmaWNhdGUgQXV0aG9yaXR5IC0gRzIwggEi\n"
"MA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDVDDrEKvlO4vW+GZdfjohTsR8/\n"
"y8+fIBNtKTrID30892t2OGPZNmCom15cAICyL1l/9of5JUOG52kbUpqQ4XHj2C0N\n"
"Tm/2yEnZtvMaVq4rtnQU68/7JuMauh2WLmo7WJSJR1b/JaCTcFOD2oR0FMNnngRo\n"
"Ot+OQFodSk7PQ5E751bWAHDLUu57fa4657wx+UX2wmDPE1kCK4DMNEffud6QZW0C\n"
"zyyRpqbn3oUYSXxmTqM6bam17jQuug0DuDPfR+uxa40l2ZvOgdFFRjKWcIfeAg5J\n"
"Q4W2bHO7ZOphQazJ1FTfhy/HIrImzJ9ZVGif/L4qL8RVHHVAYBeFAlU5i38FAgMB\n"
"AAGjgfAwge0wDwYDVR0TAQH/BAUwAwEB/zAOBgNVHQ8BAf8EBAMCAYYwHQYDVR0O\n"
"BBYEFJxfAN+qAdcwKziIorhtSpzyEZGDMB8GA1UdIwQYMBaAFL9ft9HO3R+G9FtV\n"
"rNzXEMIOqYjnME8GCCsGAQUFBwEBBEMwQTAcBggrBgEFBQcwAYYQaHR0cDovL28u\n"
"c3MyLnVzLzAhBggrBgEFBQcwAoYVaHR0cDovL3guc3MyLnVzL3guY2VyMCYGA1Ud\n"
"HwQfMB0wG6AZoBeGFWh0dHA6Ly9zLnNzMi51cy9yLmNybDARBgNVHSAECjAIMAYG\n"
"BFUdIAAwDQYJKoZIhvcNAQELBQADggEBACMd44pXyn3pF3lM8R5V/cxTbj5HD9/G\n"
"VfKyBDbtgB9TxF00KGu+x1X8Z+rLP3+QsjPNG1gQggL4+C/1E2DUBc7xgQjB3ad1\n"
"l08YuW3e95ORCLp+QCztweq7dp4zBncdDQh/U90bZKuCJ/Fp1U1ervShw3WnWEQt\n"
"8jxwmKy6abaVd38PMV4s/KCHOkdp8Hlf9BRUpJVeEXgSYCfOn8J3/yNTd126/+pZ\n"
"59vPr5KW7ySaNRB6nJHGDn2Z9j8Z3/VyVOEVqQdZe4O/Ui5GjLIAZHYcSNPYeehu\n"
"VsyuLAOQ1xk4meTKCRlb/weWsKh/NEnfVqn3sF/tM+2MR7cwA130A4w=\n"
"-----END CERTIFICATE-----\n";

bool sendTelemetryData() {
  bool result = true;
  WiFiClientSecure *client = new WiFiClientSecure;
  if(client) {
    client->setCACert(rootCACertificate);
   {
      // Add a scoping block for HTTPClient https to make sure it is destroyed before WiFiClientSecure *client is 
      HTTPClient https;
      Serial.print("[HTTPS] begin...\n");
      if (https.begin(*client, "https://httpbin.org/post")) {  // HTTPS
#if 0
        String authorization = String("Bearer ")+getJWT();
        Serial.println(String("authorization = ") + authorization);
        https.addHeader("authorization",authorization);
        https.addHeader("content-type","application/json");
        https.addHeader("cache-control","no-cache");
#endif
        Serial.print("[HTTPS] POST...\n");
        // start connection and send HTTP header
        int httpCode = https.POST("{\"binary_data\": \"aGVsbG8K\"}");
  
        // httpCode will be negative on error
        if (httpCode > 0) {
          // HTTP header has been send and Server response header has been handled
          Serial.printf("[HTTPS] POST... code: %d\n", httpCode);
  
          // file found at server
          if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
            Serial.println(String("see payload")+https.getSize());
            if  (https.getSize()>0) {        
              String payload = https.getString();
              Serial.println(payload);
            }
            result = true;
          } else result = false;
        } else {
          Serial.printf("[HTTPS] POST... failed, error: %s\n", https.errorToString(httpCode).c_str());
          String payload = https.getString();
          Serial.println(String("payload:") + payload);
          result = false;
        }
  
        https.end();
      } else {
        Serial.printf("[HTTPS] Unable to connect\n");
        result = false;
      }

      // End extra scoping block
    }
  
    delete client;
  } else {
    Serial.println("Unable to create client");
    result = false;
  }
  return result;
}

/**
 * loop
 */
void loop() {
  sendTelemetryData();
  Serial.println();
  Serial.println("Waiting 10s before the next round...");
  delay(10000);
}
```
In 1.0.2 rc2, also without header, it failed:

```
[HTTPS] begin...
[HTTPS] POST...
[HTTPS] POST... failed, error: send header failed
```

me-no-dev, 2019-04-14:
i see no relevant changes in either lib since 1.0.1, so this might be something internal.
Could you please put some debug and see what write returns here: https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/src/HTTPClient.cpp#L1069

maybe something in tls changed and headers can not fit at once or something...

me-no-dev, 2019-04-14:
Also I suggest you enable debug in the Arduino board menu :) don't know which board or ide you use since you refused to fill the proper issue form ;)

me-no-dev, 2019-04-14:
@tobozo just posted what seems to be the same issue in gitter

alexodus, 2019-04-14:
```
[E][ssl_client.cpp:33] handle_error(): SSL - Bad input parameters to function
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -28928
[E][ssl_client.cpp:33] handle_error(): SSL - Bad input parameters to function
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -28928
```

me-no-dev, 2019-04-14:
yeah we are already discussing and tracing this with @tobozo on gitter. feel free to join :)

alexodus, 2019-04-14:
The same problem with https server of google iot ( https://cloudiotdevice.googleapis.com) with the relative ca certificate

jotbe, 2019-04-14:
I am new to ESP32 and this issue matches mine that I have with a simple `GET` when I try to run the example "BasicHttpsClient". The example "WifiClientSecure" worked btw (when I used the same CA and target server as in the "BasicHttpsClient".

tobozo, 2019-04-15:
found a temporary workaround that can be applied until a real solution is found: 

in `libraries/HTTPClient/src/HTTPClient.cpp` at line 978, change this :

```
 // this fails
 // if(!_client->connect(_host.c_str(), _port, _connectTimeout))
```

by this:
```
  // this works
  if(!_client->connect(_host.c_str(), _port))
```

me-no-dev, 2019-04-15:
I just pushed a commit. Temp workarounds only when I am not active :) I would rather have permanent fixes :P please pull the latest master and confirm all is OK

tobozo, 2019-04-15:
Last commit fixed if for me \o/

jotbe, 2019-04-16:
Same here, thank you!

alexodus, 2019-04-17:
It works like a charm!! 👍

closed 2019-08-15 by stale[bot]

meltdown03, 2019-09-15:
I still have to use the temp workaround to avoid this error, did a recent commit bring this issue back? I have the exact same error as @alexodus when posting to the IFTTT server using a rootCA. Changing the line in HTTPClient.cpp makes it work. Even the BasicHttpsClient.ino example fails to work unless I remove `_connectTimeout`

Nathan-ma, 2019-09-18:
@meltdown03 Same here, The code was working last week and know is simply no and giving me this exact same errors.

I'm using github raw files as well to storage my OTA for test purposes.

Milanoss, 2019-09-24:
Same here. Issue started few days back - can be related to ESP32 update which I did.

me-no-dev, 2019-09-24:
fixed in master. I guess you guys are using the package manager to install the esp32 arduino?

stevennoyce, 2019-09-25:
I just installed via the arduino package manager on a fresh system and I am facing this issue.

Drkstr, 2019-10-02:
I have updated my libraries by redoing the steps described here: 
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/mac.md#installation-instructions-for-mac-os 

I am still getting the "error: send header failed" error message. 

I have double-checked that I am on the master branch. 
I am using the "BasicHTTPSClient.ino" example from the Arduino IDE. 

Does anyone have any recommendations on how I could fix this?

tobozo, 2019-10-02:
use the master and lose the board defines or patch manually until the problem is eventually fixed in release ?

Drkstr, 2019-10-02:
Thanks for the reply.

I am in the master branch. 

What do you mean by "lose the board defines" and "patch manually"?