espressif/arduino-esp32 #2674: Looking for functions used on webserver

EngiGamesGitHub, 2019-04-15:
Hardware:
Board:	?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:	?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:	Arduino IDE
Flash Frequency:	?40Mhz?
PSRAM enabled: ?no? ?yes?
Upload Speed:	?115200?
Computer OS: Windows 10

Description:

I am trying to call GetStill and StartStream from my loop rather than through the buttons on the web interface using the CameraWebServer example. 

I am unable to find the include that has that functionality.

me-no-dev, 2019-04-15:
those are things that happen through the browser. Code is the example in the http c file.

closed 2019-04-15 by me-no-dev

waldorffsmama, 2019-09-26:
Try this: 

http://[your_cams_ipaddress]:81/stream