espressif/arduino-esp32 #2676: WebServer::_currentUri - Different results for domain name vs IP address

TheeCaptain, 2019-04-16:
### Hardware:
Board: ESP32 Pico Kit v4.1
Core Installation version: 1.0.1, 1.0.2-rc2
IDE name: Arduino IDE
Flash Frequency: Not Sure
PSRAM enabled: Not Sure
Upload Speed: 115200
Computer OS: Windows 10

### Short Description:
When a request is made for http://www.example.com/test/t.php to WebServer
WebServer::_currentUri contains the entire URI (http://www.example.com/test/t.php)

When a request is made for <Server IP Address>/test/t.php to WebServer
WebServer::_currentUri contains only the path of the URI (/test/t.php)

This means you need two handlers for these two identical requests. I understand this is not a problem. It becomes a problem when you are using a captive portal and there are now an infinite number of domain names that can sit in front of the request URI.

### Description:
I am trying to add a handler to WebServer::on("/test/t.php") that handles all requests for this string regardless of what precedes it in _currentUri, either an IP address or a domain name.

The reason this is a problem is because when using a captive portal it creates infinite possibilities for handlers. It depends on what the user enters into the browser.

When a request is made for http://www.example.com/test/t.php to WebServer
WebServer::_currentUri contains the entire URI (http://www.example.com/test/t.php)

When a request is made for <Server IP Address>/test/t.php to WebServer
WebServer::_currentUri contains only the path of the URI (/test/t.php)

If you set up a handler for WebServer::on("/test/t.php") I would expect the server would handle this request for any request making it to this server not just requests from an IP address.

The request handler does not trigger when _currentUri has a domain name in front, instead WebServer::onNotFound(handler) is called.

Is this the behavior we expect from an HTTP server?

I have "solved" my problem by removing the substring in front of _currentUri in Parse.cpp. This is not robust nor does it seem to be the proper solution to me. Likewise, I would like to be able to use the core code and not rely on maintaining this as a quirk for my own applications. I feel like the request string should be broken down into several parts like Apache does for you.

Apache and PHP have several environment variables like the following that allow for finer tuning of handling an HTTP request.
["SCRIPT_NAME"]=> "/test/t.php"
["REQUEST_URI"]=>  "/test/t.php?id=5&b=9"
["QUERY_STRING"]=> "id=5&b=9"
["HTTP_HOST"]=> "www.example.com"

### Sketch:  
```cpp
#include <WiFi.h>
#include <WebServer.h>
#include <DNSServer.h>

const byte DNS_PORT = 53;
IPAddress apIP(192, 168, 4, 1);
DNSServer dnsServer;
WebServer server(80);

String responseHTML = "";
String responseHTML_head = "<!DOCTYPE html><html><head><title>CaptivePortal</title></head><body>";
String responseHTML_body = "";
String responseHTML_foot = "</body></html>";

void setup() {
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP("DNSServer CaptivePortal example");

  // if DNSServer is started with "*" for domain name, it will reply with
  // provided IP to all DNS request
  dnsServer.start(DNS_PORT, "*", apIP);

  server.begin();
  server.on("/", handleRoot);
  server.on("/test/t.php", handleTest);
  server.onNotFound(handleNotFound);
}

void handleRoot(){
  responseHTML_body += "<h3>Root Handler</h3>\n";
  responseHTML = responseHTML_head + responseHTML_body + responseHTML_foot;
  server.send(200, "text/html", responseHTML);
  responseHTML_body = "";
}

void handleTest(){
  responseHTML_body += "<h3>Test Handler</h3>\n";
  responseHTML = responseHTML_head + responseHTML_body + responseHTML_foot;
  server.send(200, "text/html", responseHTML);
  responseHTML_body = "";
}

void handleNotFound(){
  responseHTML_body += "<h3>Not Found Handler</h3>\n";
  responseHTML = responseHTML_head + responseHTML_body + responseHTML_foot;
  server.send(200, "text/html", responseHTML);
  responseHTML_body = "";
}

void loop() {
  dnsServer.processNextRequest();
  server.handleClient();
}
```

### Debug Messages: After connecting Android phone to Access Point and captive portal opening.
```
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: -1, ERR: 104
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
```

### Debug Messages: Requesting by IP the root in web browser
```
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128 
```

### Debug Messages: Requesting by http://www.example.com in web browser.
```
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
```

me-no-dev, 2019-04-16:
the handles is properly defined. in reality it should have worked. Please try printing the requested URL in the notFound handler so we see what is that URL

TheeCaptain, 2019-04-16:
Thank you for your lightning fast reply. I added the request uri with server.uri(); to the notFound handler.

```cpp
#include <WiFi.h>
#include <WebServer.h>
#include <DNSServer.h>

const byte DNS_PORT = 53;
IPAddress apIP(192, 168, 4, 1);
DNSServer dnsServer;
WebServer server(80);

String responseHTML = "";
String responseHTML_head = "<!DOCTYPE html><html><head><title>CaptivePortal</title></head><body>";
String responseHTML_body = "";
String responseHTML_foot = "</body></html>";

void setup() {
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP("DNSServer CaptivePortal example");

  // if DNSServer is started with "*" for domain name, it will reply with
  // provided IP to all DNS request
  dnsServer.start(DNS_PORT, "*", apIP);

  server.begin();
  server.on("/", handleRoot);
  server.on("/test/t.php", handleTest);
  server.onNotFound(handleNotFound);
}

void handleRoot(){
  responseHTML_body += "<h3>Root Handler</h3>\n";
  responseHTML_body += "<h3>" + server.uri() + "</h3>\n";
  responseHTML = responseHTML_head + responseHTML_body + responseHTML_foot;
  server.send(200, "text/html", responseHTML);
  responseHTML_body = "";
}

void handleTest(){
  responseHTML_body += "<h3>Test Handler</h3>\n";
  responseHTML_body += "<h3>" + server.uri() + "</h3>\n";
  responseHTML = responseHTML_head + responseHTML_body + responseHTML_foot;
  server.send(200, "text/html", responseHTML);
  responseHTML_body = "";
}

void handleNotFound(){
  responseHTML_body += "<h3>Not Found Handler</h3>\n";
  responseHTML_body += "<h3>" + server.uri() + "</h3>\n";
  responseHTML = responseHTML_head + responseHTML_body + responseHTML_foot;
  server.send(200, "text/html", responseHTML);
  responseHTML_body = "";
}

void loop() {
  dnsServer.processNextRequest();
  server.handleClient();
}
```

TheeCaptain, 2019-04-18:
I added function printDiognostics() to also print WebServer::_currentUri to Serial along with some other info. I call printDiognostics() in all request handles. It shows that _currentUri is http://example.com/test/t.php when requested as example.com, but only /test/t.php when requested from 192.168.4.1.

```cpp
#include <WiFi.h>
#include <WebServer.h>
#include <DNSServer.h>

const byte DNS_PORT = 53;
IPAddress apIP(192, 168, 4, 1);
DNSServer dnsServer;
WebServer server(80);

String responseHTML = "";
String responseHTML_head = "<!DOCTYPE html><html><head><title>CaptivePortal</title></head><body>";
String responseHTML_body = "";
String responseHTML_foot = "</body></html>";

void setup() {
  Serial.begin(115200);
  
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP("DNSServer CaptivePortal example");

  // if DNSServer is started with "*" for domain name, it will reply with
  // provided IP to all DNS request
  dnsServer.start(DNS_PORT, "*", apIP);

  server.begin();
  server.on("/", handleRoot);
  server.on("/test/t.php", handleTest);
  server.onNotFound(handleNotFound);
}

void handleRoot(){
  responseHTML_body += "<h3>Root Handler</h3>\n";
  responseHTML_body += "<h3>" + server.uri() + "</h3>\n";
  responseHTML = responseHTML_head + responseHTML_body + responseHTML_foot;
  server.send(200, "text/html", responseHTML);
  responseHTML_body = "";
  printDiognostics();
}

void handleTest(){
  responseHTML_body += "<h3>Test Handler</h3>\n";
  responseHTML_body += "<h3>" + server.uri() + "</h3>\n";
  responseHTML = responseHTML_head + responseHTML_body + responseHTML_foot;
  server.send(200, "text/html", responseHTML);
  responseHTML_body = "";
  printDiognostics();
}

void handleNotFound(){
  responseHTML_body += "<h3>Not Found Handler</h3>\n";
  responseHTML_body += "<h3>" + server.uri() + "</h3>\n";
  responseHTML = responseHTML_head + responseHTML_body + responseHTML_foot;
  server.send(200, "text/html", responseHTML);
  responseHTML_body = "";
  printDiognostics();
}

void printDiognostics(){
  String message = "";
  message += "_currentUri: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "Not GET";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";

  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }

  message += "\nHeaders: ";
  message += server.headers();
  message += "\n";

  for (uint8_t i = 0; i < server.headers(); i++) {
    message += " " + server.headerName(i) + ": " + server.header(i) + "\n";
  }

  Serial.print(message);  
}

void loop() {
  dnsServer.processNextRequest();
  server.handleClient();
}``

closed 2019-08-15 by stale[bot]