espressif/arduino-esp32 #2680: esp_ble_gap_set_device_name(......); Stack smashing protect failure!

fluffymadness, 2019-04-17:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.2
IDE name:						Arduino IDE 1.8.8
Flash Frequency:					80MHz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
I tried to use the following example
https://gist.github.com/GameDragon2k/90709fbe6b15e66c1b274dde3e1b2439

When I change the bluetooth device name to something longer than
BLEDevice::init("Xbox Wireless Controll"); // Give it a name
e.g to "Xbox Wireless Controller"

The function which is called by :::init() is the following:
esp_ble_gap_set_device_name(deviceName.c_str());

I get the following in the stack trace:
### Debug Messages:
```
[I][BLEDevice.cpp:555] getAdvertising(): create advertising
[D][BLECharacteristic.cpp:466] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:557] getAdvertising(): get advertising
[D][BLEServer.cpp:281] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEAdvertising.cpp:186] start(): >> start: customAdvData: 0, customScanResponseData: 0
[D][BLEAdvertising.cpp:198] start(): - advertising service: 00001812-0000-1000-8000-00805f9b34fb
[D][BLEAdvertStang.cpp:246] start(): << start
mashing protect failure!

abort() was called at PC 0x40151763 on core 0

Backtrace: 0x40091c4c:0x3ffd3730 0x40091e7d:0x3ffd3750 0x40151763:0x3ffd3770 0x400ed705:0x3ffd3790 0x4013a029:0x3ffd37f0 0x40119899:0x3ffd3810 0x4011f156:0x3ffd3830 0x400fa3c6:0x3ffd3850 0x40090461:0x3ffd3880

Rebooting...

```

lbernstone, 2019-04-17:
https://github.com/espressif/arduino-esp32/blob/master/.github/ISSUE_TEMPLATE/bug_report.md#L19

fluffymadness, 2019-04-17:
Decoding stack results
0x40091c4c: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40091e7d: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x4014d753: __stack_chk_fail at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/stack_check.c line 36
0x400eacd5: BTM_BleWriteScanRsp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btm/btm_ble_gap.c line 1628
0x401379f9: bta_dm_ble_set_scan_rsp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/dm/bta_dm_act.c line 4958
0x40117269: bta_dm_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/dm/bta_dm_main.c line 394
0x4011cb26: bta_sys_event at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/sys/bta_sys_main.c line 496
0x400f7996: btu_task_thread_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 233
0x40090461: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

closed 2019-08-15 by stale[bot]