espressif/arduino-esp32 #2684: HTTPClient authorization fail

jrickard, 2019-04-18:
----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						Arduino IDE
Flash Frequency:					80Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					921600
Computer OS:                                         Mac OSX

### Description:
Describe your problem here

I have a device on the local network.  I'm trying to connect to it.  Here's the results:

401
{
  "status": 401,
  "error": "",
  "info": "Authentication required",
  "moreInfo": ""
}

If I enter the line in host on a browser, it works perfectly.  But in HTTPClient not so much.
I've tried setting various authentications as well with zero success.  But the authorization example actually shows this form as the preferred.  And it works great in a browser.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
if (WiFi.status() == WL_CONNECTED)
    { 
     String host = "http://envoy:044892@192.168.1.149/api/v1/production/inverters"; 
     http.begin(host);  
      int httpCode = http.GET();  
         if (httpCode > 0)  
          {
            String payload = http.getString(); 
            Serial.println(httpCode);
            Serial.println(payload);
          }
          else Serial.println("Error on HTTP request"); 
   
    http.end(); 
### Debug Messages:
```
Verbose.  Compiles and runs fine.  Just bad results from the server.
[V][HTTPClient.cpp:224] beginInternal(): url: http://envoy:044892@192.168.1.149/api/v1/production/inverters
[D][HTTPClient.cpp:265] beginInternal(): host: 192.168.1.149 port: 80 url: /api/v1/production/inverters
[D][HTTPClient.cpp:1002] connect():  connected to 192.168.1.149:80
[V][HTTPClient.cpp:1097] handleHeaderResponse(): RX: 'HTTP/1.1 401 Unauthorized'
[V][HTTPClient.cpp:1097] handleHeaderResponse(): RX: 'Date: Thu, 18 Apr 2019 22:33:26 GMT'
[V][HTTPClient.cpp:1097] handleHeaderResponse(): RX: 'Connection: close'
[V][HTTPClient.cpp:1097] handleHeaderResponse(): RX: 'Date: Thu, 18 Apr 2019 22:33:26 GMT'
[V][HTTPClient.cpp:1097] handleHeaderResponse(): RX: 'WWW-Authenticate: Digest qop="auth", realm="enphaseenergy.com", nonce="Nvu4XEWt3hdlEieNfKwy3H34vjU="'
[V][HTTPClient.cpp:1097] handleHeaderResponse(): RX: 'Content-Type: application/json'
[V][HTTPClient.cpp:1097] handleHeaderResponse(): RX: 'Content-Length: 90'
[V][HTTPClient.cpp:1097] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:1127] handleHeaderResponse(): code: 401
[D][HTTPClient.cpp:1130] handleHeaderResponse(): size: 90
[D][WiFiClient.cpp:482] connected(): Disconnected: RES: 0, ERR: 128
[D][HTTPClient.cpp:1264] writeToStreamDataBlock(): connection closed or file end (written: 90).
[D][HTTPClient.cpp:369] disconnect(): tcp is closed
```

jrickard, 2019-04-19:
The problem appears to be indicated in the verbose debug 'WWW-Authenticate: Digest qop="auth", 
and realm="enphaseenergy.com", nonce="Nvu4XEWt3hdlEieNfKwy3H34vjU="'

This would seem to indicate the server uses Digest Access Authentication as described in RFC 2617 rather than the earlier Basic authentication.  I would request this library incorporate Digest Access Authentication.

[](https://tools.ietf.org/html/rfc2617)

jrickard, 2019-04-19:
Are those crickets???

me-no-dev, 2019-04-21:
OK, we could look into it

jrickard, 2019-05-16:
Not an interesting problem I presume.   Oh well.....

closed 2019-08-14 by stale[bot]